ExpressLRS/ExpressLRS #1142: Happymodel ES24TX Pro Series (Slim Pro) outputs twice the power set in Lua Script

goner988, 2021-11-28:
## Current Behavior

When measuring the output power of the TX Module with the IRC RF Power Meter V2 (measuring Peak Power not average, @2400MHz) the TX Module appears to output twice the power being set via the Lua Script. This behaviour was verified using ELRS V1.2 and V2.0RC3 on the TX. 

## Steps to Reproduce

1. Connect IRC RF Power Meter V2 to HM ES24TX Pro Series 

2. Turn Radio on

3. Set output power

4. Measure actual output power

## Details

Is it possible my device is defective? Is there something I can check with a multimeter to verify proper functionality of the TX Module?

## Your Environment

- TX hardware: Happymodel ES24TX Pro Series (Slim Pro)
- RX hardware: HM EP2
- Power Meter: IRC RF Power Meter V2
- Handset model: FRSKY X-Lite Pro
- OpenTX: 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): version 1.2 and 2.0RC3
- Packet Rate: all packet rates
- Telemetry Ratio: Across all telemetry ratios
- user_defines: BINDING_PHRASE, UART_INVERTED, FEATURE_OPENTX_SYNC, AUTO_WIFI_ON_INTERVAL

[image]
[image]
[image]
[image]
[image]
[image]
[image]

JyeSmith, 2021-12-08:
Mine is looking ok 25-100mW.  250 and 500 are a bit high.

https://streamable.com/s85zi1

qubic, 2021-12-10:
Maybe you flash the wrong firmware, flash the es24tx pro firmware instead of es24tx.

brandonrc, 2022-01-07:
@wvarty

closed 2022-06-25 by github-actions[bot]

danflyingrobot, 2024-05-19:
How long does that x lite last with the elrs module?