ExpressLRS/ExpressLRS #1144: R9 Random LQ flashing to 0

Hrisfpv, 2021-12-01:
Hi i have an r9m flashed with elrs.I have random lq flashes to 0 and then instantly recovering to 100.On my post flight log in bf it shows min rssi dbm -130 min link 0% and so on.It’s like a mini failsafe.On my main quad i have an r9mx and r9mm on my other but the same thing happends to both.Running 2.0 rc3 but it’s has been like that even on stable releases like 1.2.

deadbytefpv, 2021-12-01:
At what power level do you have the r9M? Does it have ample cooling?

What about your soldering of the receiver wiring?

Hrisfpv, 2021-12-01:
200hz 250mw no fan mod.
My wiring for the reciever is perfect.

deadbytefpv, 2021-12-01:
There's a possibility the module is overheating and would reboot or going into thermal protection.. 

Try a lower power level if it will not happen again..

Hrisfpv, 2021-12-01:
I don’t think the module is overheating, because it’s cold outside and it should be hot at 250mw.I will try thoe

AlessandroAU, 2021-12-01:
What baudrate? Is the module making beeping noises?

Hrisfpv, 2021-12-01:
Nope its not making beeping noises,i don’t have problems with failsafes it’s just that i see 0lq for a split second in my osd.As far as the baudrate im not sure

AlessandroAU, 2021-12-01:
What RFrate are you using? 200hz or something lower? Did you do resistor mod to R9M? If not you have to lower baudrate in opentx/edgetx to 115200.

Hrisfpv, 2021-12-01:
My r9m is the 2019 version so I don’t need a resistor and im running 200hz

brandonrc, 2022-01-07:
@Hrisfpv any update on this issue? 

Have you upgraded to the latest firmware and does it still do it on that firmware?

closed 2022-01-13 by AlessandroAU