ExpressLRS/ExpressLRS #1162: Power setting in Lua doesn't correspond to power set by TX button (2.0.0-RC4)

endymuhardin, 2021-12-07:
## Current Behavior

When TX power changed via module button, display in Lua script is not adjusted accordingly.

## Steps to Reproduce

1. Open Lua script

2. Set dynamic power to Off

3. Observe max power setting, for example 100mW. LED in module is blue

4. Push the button at TX module to change power. In this case, BetaFPV Nano TX, one press will bump the power to 250mW. LED turns purple, indicating the power level is at 250mW

5. Observe max power displayed in Lua. It still shows 100mW. Expected behavior : changed to 250mW

6. Try exit lua and start lua again. It still shows 100mW

## Your Environment

- TX hardware: BetaFPV Nano TX 915
- RX hardware: irrelevant
- Handset model: Jumper T-Lite
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): 2.0.0-RC4
- Packet Rate: irrelevant
- Telemetry Ratio: irrelevant
- user_defines: irrelevant

pkendall64, 2021-12-07:
The LUA screen shows the configured (startup) value, whereas the button changes the current operating power level, not the configuration.

So this is working as expected.

endymuhardin, 2021-12-08:
So the lua value does not get updated like version 1? 
I think most users will expect it to behave like version 1, the value get updated as we change current power level.
At least when dynamic power is off.
In this current form, there will be discrepancy from what the user sees (in lua) vs what the actual power level (and LED status).

And while we're discussing about power level button, when dynamic power is on, what is the expected behavior for the button? Can it change current power level?

pkendall64, 2021-12-08:
We could change it to set the configured max power output like LUA, but then users would ask why it's different when they go back to the status screen and we'd have to tell them that thats the current power no the configured power.
So I actually think this might be the right thing to do. I'll ask the other devs.

endymuhardin, 2021-12-08:
I actually have no objection about the current implementation. 
I personally use HM Slim Pro which does not have any button nor screen.

If you guys decide to stay with the current implementation, maybe we can clarify the behavior in the documentation or the label in lua screen to differentiate configured vs actual.

brandonrc, 2021-12-09:
@pkendall64 I think the HM slim pro is a different issue. 

The OLED screen does match the dynamic power when it changes. It stays with the power adjustments.

pkendall64, 2021-12-09:
> @pkendall64 I think the BetaFPV nano is a different issue.
> 
> The OLED screen does match the dynamic power when it changes. It stays with the power adjustments.

Yes, I know on the status screen it is the current value. My point was that there are 2 potential values, the configured value and the current value in 2.0, whereas 1.x only has the configured.

CapnBry, 2021-12-16:
> If you guys decide to stay with the current implementation, maybe we can clarify the behavior in the documentation or the label in lua screen to differentiate configured vs actual.

To further clarify, the Lua only shows configuration, it never shows status for anything except the status row up top. The telemetry ratio also dynamically changes but the Lua always shows the configured value. It wouldn't make any sense for the Lua to show the current value because then you'd never be able to tell what the configured value was. Oh it says 10mW, is that just right now, or the highest it will go? It is pretty obvious that it **has to** show the configured value.

That said, the button changing the current value is a mess, especially if the user has set Dynamic Power on because the power will immediately change again. Having power on a module button is not a great idea for anything except those running without Lua at all. However, changing the button to change the config is bad because when you change config the TX will interrupt communication with the RX as it saves the change, which if your model was far away and prompted you to raise the power, could cause a complete failsafe during the blackout.

endymuhardin, 2021-12-18:
Ah, I wasn't aware that saving config interrupt the communication. So changing power via button currently doesn't change (and save) config?

pkendall64, 2021-12-18:
> Ah, I wasn't aware that saving config interrupt the communication. So changing power via button currently doesn't change (and save) config?

Correct, because it's not changing configuration, it's just changing the current power level. Which as CapnBry says, will change immediately if running with dynamic power.

endymuhardin, 2021-12-18:
So, do you suggest I close this issue?

pkendall64, 2021-12-18:
Yeah, I think so. I think we need to come up with a way to handle it better, but with dynamic power it makes it really difficult to know what to do with it.

closed 2021-12-18 by endymuhardin