ExpressLRS/ExpressLRS #1163: telemetry lost (2.0.0-RC4)

HWT662988, 2021-12-08:
## Current Behavior

I update the **BETAFPV 915Mhz Micro TX** and **BETAFP 915Mhz Nano RX** from 1.1.0 to 2.0.0-RC4,the telemetry lost while distance between tx and rx more then 2 meters(telemetry not lost within 1meter).
It wokr properly before update to 2.0.0-RC4(using 1.1.0 firmware),and flashing back to 1.1.0,the telemetry didn't lost.
I tested two 915Mhz modules, and both of them had the above problems.

(The telemetry lost problem just happen on 915Mhz which upgrate to 2.0.0-RC4  , the did not happen on 2.4Ghz.)

## Your Environment

- TX hardware:BETAFPV 915Mhz Micro TX
- RX hardware:BETAFPV 915Mhz Nano RX
- Handset model:MAMBO Ethix
- ExpressLRS version (TX & RX MUST MATCH):2.0.0-RC4
- Packet Rate:100hz
- Telemetry Ratio:1:64
- user_defines:
-DRegulatory_Domain_FCC_915
-DHYBRID_SWITCHES_8
-DFEATURE_OPENTX_SYNC
-DUART_INVERTED
-DAUTO_WIFI_ON_INTERVAL="20"
-DUNLOCK_HIGHER_POWER
-DFAST_SYNC

brandonrc, 2021-12-09:
@HWT662988 what is your range? I have a BETAFPV 915Mhz Micro TX and it basically died on me. It would get about 10 feet and I would fail safe on 2.0.0rc4,3,2

Did not try anything older. What does your range look like, like LQ values for x meters.

HWT662988, 2021-12-09:
The RSSI value drops to 50% about 10 feet.

edernucci, 2021-12-18:
Fixed on #1198

closed 2021-12-19 by deadbytefpv