ExpressLRS/ExpressLRS #1165: Allow failsafe values to be set in the RX

MartinHugh, 2021-12-09:
Background
--------------
Note that this request appears to be talking about Matek functionality.  The Matek functionality is attempting to provide functionality which is not present without a FC, but it is wrong anyway.  A better solution would be if the RX itself provided the functionality. The critique of the Matek implementation is being used to explain why that solution is inadequate and why implementing this in RX itself would be a better solution.

Problem
-------------
I am using Happymodel EP1 recievers with PWM converter boards for using in my fixed wing aircraft.

Some of the aircraft use flaps and the need for a 6th channel (5 is for arm and is only 1 bit anyway) means that I cannot use the integrated Matek PWM board because it only has 5 channels.  This board however I believe allows failsafes from the Express LRS web page running on the board because it is a "PWM firmware".

The other two options are to use the Matek CRSF-.PWM converter boards.  These do have failsafe settings.  However they cannot be modified as far as I can tell.  The Matek Configurator appears to be a read-only tool for failsafes.

Just to make matters worse, the failsafes make assumptions about the what is on what channel, especially requiring throttle to be on channel 3 (I run TAER on all my models so this is a pain).  Yes I could change the mapping in the TX model, but if I want to switch between RX protocols temporarily for any reason, I dont want to have a TX model with specific channels allocation for ELRS and another model for any other protocols - just because the failsafes on the converter require a particular mapping.

Even if the channels in the TX were mapped to suit the board, the fixed values chosen are not particularly useful. Often a slight turn and a slight nose down would be useful in failsafe mode.  Even if one wanted level flight in failsafe, the ability to not be able to set the values means that one cannot accomodate slight variations in the mechanical trim which mean that neutral always exactly on 1500ms.

What would be useful would be if PWM failsafes could be set in the RX, regardless of whether the firmware is a PWM specific one or not.

Thank you for reading.

deadbytefpv, 2021-12-09:
http://www.mateksys.com/?portfolio=crsf-pwm#tab-id-3

From the Failsafe Section on that tab linked above:
Failsafe
- Failsafe value is set automatically after CRSF-PWM board is powered on and receive CRSF signal. LED blinks 8 times quickly.

- Failsafe value = The PWM value of each channel(except CH3) when CRSF signal is received by CRSF-PWM board for the first time

- CH3 is specially arranged for the throttle,  Failsafe value = 988 by default.

- Usually put the Throttle  joystick to the lowest point,  put AIL(Roll), ELE(Pitch), RUD(Yaw) middle before powering on receiver.

MartinHugh, 2021-12-09:
Thank you deadbyte.  If this works, this does get around some of the problem - but it means I cannot set my elevator failsafe without remapping all my TX models.

brandonrc, 2021-12-12:
> ## Background
>
> Note that this request appears to be talking about Matek functionality. The Matek functionality is attempting to provide functionality which is not present without a FC, but it is [...]

So, I have to disagree with this idea that the RX should handle fail safes. A RX should only be used for control link and nothing more. ELRS itself is a very minimal control link which makes it special but also makes it the best at what it is used for, radio control. 

You really should have assisted co-pilot boards if you want fancy fail safe stuff. Kind of like what ZOHD makes. It is import to keep the code very clean and do not contain anything that is `wing` specific or even `drone` specific. 

A RX should get a signal from a TX that contains the channels and deliver that to a FC or `extra hardware` to be used. 

We have to avoid cluttering up ELRS code based on each use case because with the RC world.. there are 1000's of use cases.

brandonrc, 2021-12-12:
@CapnBry you are a wing guy right? Do you think we should add the ability to set failsafe values in the RX?

MartinHugh, 2021-12-12:
@brandonrc, many thanks for your comments.

MartinHugh, 2021-12-12:
@CapnBry yes I fly (flying wings and fixed wings without flight controllers and would prefer to set failsafes in the RX.

brandonrc, 2021-12-12:
> @brandonrc, many thanks for your comments.

This is just my thought, maybe other dev's might thing different. 

What is funny is I use to mess around alot with mavlink and autonomous stuff and wanted the RC link to contain everything in one. But once I started working with the ELRS guys I found that what makes ELRS amazing is the simplicity. 

Crossfire attempts to add a bunch of stuff but seems to fail at a few things due to trying to be a one stop shop. 

I now run my links with no telemetry because I will either record with OSD or have a mavlink radio for only telemetry as a separate connection. 

That way my 2.4ghz control link does only 1 thing and does it very well.

brandonrc, 2021-12-12:
@FOG-Yamato

MartinHugh, 2021-12-13:
I just wanted to emphasise that whilst I understand the comments from @brandonrc above, what I am asking for is not a new or unusual feature for radio's used for fixed/flying wing /PWM applications.  It is the norm for systems such as Spektrum/FrSky/Futaba/Jeti etc to have failsafes stored in the RX.   I certainly would not call it "fancy", it is a fundamental feature and far from farming this out to a external hardware I would argue that the best place for this for reliability would be as close to the point where the system detects that the signal has failed.  i.e. in the RX.  

I completely understand wanting to keep things simple, the problem is that once you start adding luxurious add on features, the stable door is open and there will be no end to it.  What I am trying to say is this is not a luxurious feature, it is an essential safety feature and should be part of the RX.

deadbytefpv, 2021-12-13:
The PWMP receivers DO save the failsafe on the receiver.

MartinHugh, 2021-12-13:
Thanks @deadbytefpv, I did discuss why I found the Matek integrated board is not suitable for many applications in the original question (only 4 effective channels).  But you are right there does seem to be some acknowledgement that this is an RX feature, but I think it should be available for firmwares which are not specific to integrated PWM boards.

CapnBry, 2021-12-14:
The only time failsafe values are needed though are when operating in PWM output mode, and we already have failsafe for PWM output mode as deadbyte has pointed out.

A connection to an FC would never need failsafe values because failsafe determination and failsafe behavior is entirely determined on the FC itself. Adding a failsafe routine on the RX would interfere with that FC behavior and is undesirable. Therefore ExpressLRS should **never** implement failsafe when operating in CRSF output mode.

CRSF to PWM converters are responsible for determining their own failsafe behavior and we will not change our output to add additional failsafe behavior due to the same protocol being used to communicate to a FC. We will not add any code to make it a special mode of CRSF-output-but-connected-to-something-that-doesn't-do-failsafe-the-way-you-want-it because that is the responsibility of the converter itself. The converter software may choose to implement CRSF config protocol passthrough available in 2.0 which would allow the ELRS V2 Lua to configure this _through_ the RX, but that's up to them.

Asking for this to be implemented in the RX itself is like buying a FC without a current sensor and saying, well the RX can probably calculate the current from voltage sag and throttle position so why not?!

MartinHugh, 2021-12-14:
@CapnBry  thank you for your comments.  I note the wontfix, fair enough - though I think your last analogy is a bit of a stretch. Especially if you consider the dozens of brands of RX which do host failsafe, none of them do any current calculation.

closed 2021-12-14 by MartinHugh

psfales, 2024-05-21:
I'm adding this comment in hopes that it will help someone who (like me) came here in hopes of finding out how to set the failsafes on a Matek CRSF-PWM converter.   I don't think the answer is widely know because it took me a lot of searching to find it, even though the answer is buried somewhere on the Matek web site (https://www.mateksys.com/?portfolio=crsf-pwm#tab-id-7)

Specifically, you need to put a temporary jumper between the PWM1 and PWM2 signal pins.   Set all the channels on the transmitter to the values you want for failsafe, and power up the transmitter.   Those values will now be used to set the stored failsafe values.   Then you can power down and reconnect the PWM1 and PWM2 outputs. 

This works for all channels, even channel 3 (so you can use it to set a failsafe for channel 3 to something other than -100%)

This might depend on the firmware installed in the Matek converter,   But if you don't have a fairly recent version (2.6.0 at this writing), I recommend upgrading as there some useful bug fixes and new features.