ExpressLRS/ExpressLRS #1168: Micro TX LED Color Saving

bmfkai, 2021-12-10:
Is there any way to save the LED color so it's persistent through a power cycle? 

Enhancement, maybe?

pkendall64, 2021-12-10:
Is that a betafpv module? If it is then you have to ask them as those are "enhancements" they have made and not part of regular ELRS.

brandonrc, 2021-12-12:
@pkendall64 @AlessandroAU @wvarty 

This is not something we want to add to ELRS? 

I really do not like the ide of using the RGB as some kind of personal color custom thing... it is better to use it for different modes and settings we use for the firmware right?

brandonrc, 2021-12-12:
Vote :+1:  for yes and :-1: for no.

bmfkai, 2021-12-12:
Sorry, this was meant for the BetaFPV micro tX module.

deadbytefpv, 2021-12-12:
@bmfkai mainline ExpressLRS implementation doesn't care which module it is.. 

Module LED status code is explained here:
https://www.expresslrs.org/2.0/quick-start/led-status/#transmitter-module-led-status

brandonrc, 2021-12-12:
> Sorry, this was meant for the BetaFPV micro tX module.

Yea, betaFPV micro TX is not going to have the custom LED stuff betafpv made. They decided to advertise some feature to customers that was not apart of the ELRS code. Hate they did that to users but if you want to use the latest and official ELRS code, there will not be custom LED RGB colors.

pkendall64, 2021-12-12:
The idea of a fixed set of colours and transitions is so it can be used as a debugging and informational aid to the user and the devs. I we allow users to select their own colour then we can't do that, so it's a big NO from me.

bmfkai, 2021-12-12:
For what it's worth, I wasn't referring to any indicator LED. I was referring to the cooling fan LED on the BetaFPV micro tX which is selectable in their menu leading me to believe it's purely aesthetic. When searching for a requests section, I ended up here instead of the BetaFPV repo by accident.

pkendall64, 2021-12-12:
> For what it's worth, I wasn't referring to any indicator LED. I was referring to the cooling fan LED on the BetaFPV micro tX which is selectable in their menu leading me to believe it's purely aesthetic. When searching for a requests section, I ended up here instead of the BetaFPV repo by accident.

There is only one RGB LED, and thats the one that BetaFPV have put behind the fan for mood lighting. It has always been used for status information.

BetaFPV are making changes without asking any questions of the ExpressLRS dev team.

closed 2021-12-12 by JyeSmith

brandonrc, 2021-12-12:
@bmfkai you could continue using the betaFPV fork but I would not recommend it. 

There repository was behind our project by several releases. 

We recommend you use the official ELRS firmware. We are trying our best to get an oled menu out as fast as possible but also reliable. 

`2.1` will have the official oled firm. :) 

If you want to do some beta testing with have a PR that we will merge soon into `master` branch. 

BetaFPV branch is : `This branch is 31 commits ahead, 229 commits behind ExpressLRS:master.`