ExpressLRS/ExpressLRS #1170: BetaFPV RX disconnected when starting LUA script (2.0)

blind-oracle, 2021-12-11:
Radiomaster TX16S / OpenTX 2.3.14
TX: HappyModel ES24TX
RX1: Matek 2.4 Diversity
RX2: BetaFPV 2.4

I've flashed 2.0 to all of the above, update to 2.0 LUA.

Combination with Matek RX works flawlessly. But when using BetaFPV RX it works fine only until I start LUA - then immediately RX is disconnected (telemetry lost and no channel updates on the FC).

RX reboot does not help to restore communication - I have to reboot the radio and then the connection is reestablished.
Tried re-flashing RX, no change.

deadbytefpv, 2021-12-11:
I just tried the same setup (only difference is that I'm using EdgeTX 2.5.0 on the Tx16s and a HM ES24TX Pro)

Powered up module. Then the kwad with the BetaFPV Nano 2.4G. Connection established.
Went into ExpressLRS Lua script.. Connection is still intact.
Change a bunch of settings (Packet rate, Telem Ratio, Tx Power). Connection Stays.

pitts-mo, 2021-12-11:
I had found this behavior repeatable using opentx 2.3.14 and the 2.0.0 RCx biulds w/250hz,1/16 or 1/32 telem, wide switches,  dynamic power... confirmed on issue on betafpv and matek. i had suspected it was related to telemitry somehow but it does occur on both inav and betaflight. i do not remember this occuring on my builds w/ happymodel rx. I will retest all my builds with 2.0.0 release in the next couple days and report results. -p

pitts-mo, 2021-12-11:
Sorry, i jumbled the dwtail. I was thibking of lua freakout. But i do recall betafpv rx having odd behavior after attempting to set model match.... somehow the rx did stabalize and accept model match. I will still report updated findongs. -p

blind-oracle, 2021-12-11:
I've disabled model matching both in TX (set to Off) and on RX (set to 255 on the web GUI). No change.

Strange thing is that the connection is lost when running some other LUA scripts from TOOLS.
E.g. if I start `Betaflight Config` -> bang, telemetry lost, reboot radio.

Will investigate more.

@deadbytefpv Well maybe the fact that you use the Pro TX makes the difference as I think this is some OpenTX + TX module problem...

deadbytefpv, 2021-12-11:
@blind-oracle Pro or non-pro shouldn't really matter much.. same hardware/pin mapping.. pro just have some extra pins in use for the PA..

blind-oracle, 2021-12-11:
@deadbytefpv I agree - it shouldn't.
And Matek/BetaFPV shouldn't matter either. But it does for some reason...

Just flashed a plane with Matek 24D - also works like a charm. I have another BetaFPV RX lying around, maybe will test on it.

pitts-mo, 2021-12-11:
Ok, please disregard most of my other comments here.
I do not ever recall my BetaFPV RX loosing connection on entry into luaV2.

In case this ends up being related:
Today, I updated to 2.0.0 final release.
I cycled model match OFF then ON with a BetaFPV RX a couple times. It takes maybe 5-10 seconds to catch hold but did not seem to become stuck like I had seen it do before. It still recall spending ~15 minutes on my first BetaFPV binding/model match attempts where it appeared to be stuck partially bound. Maybe the partially bound issue I saw was related to a change in binding or on attempts to set model match ON after a change of model ID? 

Additionally the luaV2 "freakout" I was originally focused on appeared as cyclic lua page refresh after only partial data display and is repeatable in 2.0.0 final release. I still suspect the cause is related to telemetry data and it occurs with all of my receivers (BetaFPV, Matek, HappyModel).

pitts-mo, 2021-12-11:
My betaflight setup lua (betaflight config) does not appear to run correctly for me. But testing this and other crossfire  and elrs lua versions do not seem to be killing my RX links on my happymodel/Betafpv betaflight builds.

Something with the build the betafpv RX is connect to?

blind-oracle, 2021-12-11:
Update: tried different settings (same as Matek and others) of Hz/Power/Telemetry ratio etc. Nothing helps.

Here's the video of the process: https://tube.novg.net/w/9vRMGm6FwoR2yKfeMn5nKn

blind-oracle, 2021-12-11:
Update: RX reboots infinitely, see topic for an exception.

blind-oracle, 2021-12-11:
Update2: Works fine with -RC4

blind-oracle, 2021-12-11:
I've rebuilt 2.0 in VSCode from maintenance branch - it started to work fine.
Then I've cleaned `.platformio` folder to reset the environment - it started to work also when building from Configurator...

I guess that's my mistake, sorry for disturbance.
Why it worked on Mateks and with RC4 though remains unknown.

blind-oracle, 2021-12-11:
Oops, I guess I'm too early. After cleaning up the environment and reflashing everything from scratch - I have the same behaviour on the Matek receiver...

brandonrc, 2021-12-12:
@blind-oracle is this only with betaFPV rx's? 

Have you tried different TX's or you just have the one?

blind-oracle, 2021-12-12:
@brandonrc Like I said in the last message - that happened also with Matek 24D.

But it looks like that the problem is connected to the build environment - though I've removed `|/.platformio` on my Windows PC - it still produced buggy/unstable(?) binaries, though I don't know what causes this.

Then I switched to MacOS with VSCode and built `2.0.x-maintenance` branch there for all my devices (Matek/HappyModel/BetaFPV) - it works flawlessly everywhere now.

So I guess the cause is not the code, but some PlatformIO inconsistencies... Hard to say what exactly.

pkendall64, 2021-12-12:
If you can get it to crash again, having the stack trace and the .elf file is the only way we can decode the trace to find out where the problem is.

blind-oracle, 2021-12-12:
@pkendall64 Ok, will try. The stack trace like in the initial message is enough? Or do I need to build with some debug defines?
And ELF - just the RX firmware image is enough?

pkendall64, 2021-12-12:
The .elf file,`firmware.elf`, should be next to the .bin file. That contains all the information about the names of the classes and functions that the trace decoder uses to turn the big glob of hex into a human readable stack trace. If you are using configurator, you may have to search your filesystem for the file.

blind-oracle, 2021-12-12:
@pkendall64 Got it, binary with debug info. Will try to repro.

blind-oracle, 2021-12-13:
@pkendall64 Sad, cannot reproduce it anymore. I guess after cleaning the `.platformio` folder I've altered the environment too much. Gonna close this now, will reopen if I'll have a repro.

closed 2021-12-13 by blind-oracle

pkendall64, 2021-12-13:
Ok, thanks for trying.