ExpressLRS/ExpressLRS #1179: Telemetry lost and recovered on mega baud rates

nightwolfix, 2021-12-13:
Hi team,

Myself and several other people report a continuos telemetry lost and recovered on edgetx 2.5, TX16s and several esp based modules like hiyounger and HM es24tx while the baud rate is above 400k. No issues when baud is at 400k. Could you please check into this? Link to FB group reporting this issue: https://m.facebook.com/groups/636441730280366/permalink/885004905424046/

Thanks

schugabe, 2021-12-13:
Does the elrs lua script report any errors?

nightwolfix, 2021-12-13:
No errors

NemesisXB, 2021-12-31:
I have the exact same issue on megabaud. Works fine on 400k.
RM TX16S
EdgeTX
Happymodel TX
Happymodel EP RX

schugabe, 2022-01-10:
@nightwolfix @NemesisXB We have a possible fix in PR #1268 It would be helpful if you could test the PR (flashing possible via configurator). This PR is based on the master branch but the branch is pretty stable and should be good for a test.

NemesisXB, 2022-01-11:
> @nightwolfix @NemesisXB We have a possible fix in PR #1268 It would be helpful if you could test the PR (flashing possible via configurator). This PR is based on the master branch but the branch is pretty stable and should be good for a test.
> 
> The PR was already merged and the fix is in master (and will be releasted in 2.1): but it stil would be great if you could report back if this solves your issues.

@schugabe, Tested on 3.75M and it seems to be solved. Had no telemetry lost telemetry recovered messages.

sendorm, 2022-01-15:
Same issue on t-lite with happymodel ES24TX. Had sensor lost and telemetry lost messages on 3.75M. No problems on 400k. on elrs 2.0.0 with edgetx.

schugabe, 2022-01-21:
Fix is released with 2.1

closed 2022-01-21 by schugabe