ExpressLRS/ExpressLRS #1181: Happymodel ES24TX BLE range/connect problem

konstantinfpv, 2021-12-13:
## Expected Behavior

Expect Bluetooth joystick working

## Current Behavior

Flashed 2.0.0 HM TS24TX on HM 24 T-Lite, via Lua v2, start running BLE Joystick. On my old laptop (win 7) I'm forced to hold the radio on top of my laptop, and latency feels high and has freezes. On my home PC, I even cant connect (win 10).

## Steps to Reproduce
FLASH HM TS24TX via UART (USB)
download new LUAv2
Run LUAv2
Run BLE Joystick
Connect BT
Start LIFTOFF
Joystick isn't smooth and works only just above the laptop keyboard.

- Operating System and version:
- TX hardware: Happymodel 24 T-Lite module
- RX hardware: Win 7 or Win 10

LoakAudio, 2021-12-28:
Hello,

With T-Lite ELRS module and a CSR 4.0 BT dongle, works fine with Win 10 and Velocidrone IF T-Lite and CSR dongle are quite close, like 75 cm. More, connection is lost, within 75 cm no latency, fully playable.
 
Is there a recommended version for BT or a recommended dongle to work fine in BLE ?

pkendall64, 2022-06-10:
The bluetooth and the WiFi of the module share the same antenna so if the WiFi range is limited, so too will be the range of bluetooth. Some modules have better range than others with the choice of their antenna design and theres nothing we can do about that in the software, unfortunately.

closed 2022-06-10 by pkendall64