ExpressLRS/ExpressLRS #1182: Support SBUS protocal

SonireolxXx, 2021-12-13:
I'm trying to hook up an ES900 RX receiver with the old Eagle tree Vector FC, however the Vector only takes SBUS.
Probably everyone is using iNav or Betaflight nowadays but it will be great to have real SBUS output so ELRS can work with many legacy devices.

deadbytefpv, 2021-12-14:
The ESP-based receivers can have inverted TX pads for connection to an SBUS (inverted RX) pads via the RCVR_INVERT_TX option.

SonireolxXx, 2021-12-14:
@deadbytefpv thanks for reading, but it's not the same thing, RCVR_INVERT_TX just reverse the polarity of TX but the protocol is still CRSF. To make it work with some legacy FC it has to be real SBUS output

FOG-Yamato, 2021-12-14:
There is no current plans to add any sort of non PWM or CRSF output to the receivers. SBUS is as you said a legacy protocol that shouldn't be used anymore.

jmxp69, 2021-12-14:
No plan for SBUS is a huge miss IMHO. There is a large user base out there with SBUS which is far more standard as a single wire protocol than Crossfire when you get away from quads. i.e. SxR, ikon, Eagletree, Hobbyeagle, etc..

rburrow87, 2021-12-14:
While SBUS is a "legacy" protocol, it feels a bit premature to forget about supporting it already? I realize it's not ideal in the quad world, but everywhere else (like fixed wing and single rotor), it's still the standard aside from manufacturer specific stuff (like ExBus, XBus, SBus2, FPort, etc.). There's little to no CRSF support there. That may change eventually, but not any time soon.

Or is ELRS intending to stay focused on quads and maybe simple long range fixed wing aircraft? I would love to see it expand and become an open source standard for all RC aircraft eventually.

JyeSmith, 2021-12-14:
Unfortunately its not as simple as "just add sbus".  There is then all of the tlm and extras to go with it.  The dev team has enough fun dealing with a single protocol, let alone now maintaining another.

ELRS is trying to pushing things forward, so maybe it is worth while going back to Eagle etc and asking them to update to a new protocol?

Personally I think inverters have caused enough issues over the years and we can let them die out.  If ELRS continues to support older protocol then manufacturers will continue to release 'legacy' hardware.

NemesisXB, 2021-12-14:
> When ELRS has a substantial enough userbase to influence manufacturer choices, great. Until then ELRS is omitting a substantial potential user base without SBUS support. I don't think anybody is advocating for poorly implemented options like inverters and Eagletree is out of business but they sold countless top shelf flight computers that are still used by a big group of people. There's no point in muddying the discussion of what could or should be done with manufacturers or some of the poor choices they've made. The simple fact is that omission of SBUS excludes the possibility of a substantial userbase ever even considering ELRS as an option. They will see Crossfire or PWM and immediately swipe left.

Like Jye said. Having to maintain another legacy protocol would be silly. ELRS is done mostly in the spare time of the devs(I know I'm stating the obvious, but I'm stating it anyway). I'm sure if another dev joins and creates a PR with SBUS, the ELRS devs will most likely merge it. But the current devs all have things they are already working on. Another thing to take note is that there is hardware where they already have to do optimisations to fit the code. Adding something like SBUS which will not fit on those targets will most likely fragment ELRS (it can probably be achieved by a compile time switch).
You also keep saying ELRS is missing out on a substantial potential user base as if they will make more money if more people use it. Which is obviously not the case. For somebody like TBS, throwing engineering hours at it would make sense, because for them, more users = more money.

jmxp69, 2021-12-14:
I appreciate the nature of opensource. I've never mentioned money. My point is about adoption. The more adoption, the more users will clamor for things like native integration of ELRS (and whatever supported protocols) into things like transmitters and mainstream receivers. Again, I don't see a point in muddying the conversation with details not germane to the single point that a lack of SBUS omits a substantial userbase.

If that fact is of no interest to the ELRS team then so be it. Nobody said anything about money. Nobody demanded anything be done right this very second. It's a simple ask.

JyeSmith, 2021-12-15:
So support a legacy protocols, for a FC that is no longer manufactured, from an out of business manufacturer?

From last count there are 12 active manufacturers in discussion with ELRS and more who have touched base.  I think we are doing ok influencing manufacturer choices.

SunjunKim, 2021-12-15:
Supporting SBUS in the ELRS project requires a fundamental change in the code base, and is extremely unlikely to happen.

It'd be better to launch a separate project CRSF -> SBUS converter board, like a Matek CRSF->PWM converter. This will be far more simple, and also include TBS boards in the game. However, ELRS repo is not the right place.

jmxp69, 2021-12-15:
So effectively this is tech will be limited to very simple fixed wing and quad applications. Got it. Too bad.

SunjunKim, 2021-12-15:
> So effectively this is tech will be limited to very simple fixed wing and quad applications. Got it. Too bad.

There are FCs for fixed wings that support CRSF protocol. Like Matek F405-WING. Limiting factors are the legacy FCs, not ELRS project itself.

jmxp69, 2021-12-15:
Right.

closed 2021-12-15 by SonireolxXx

SunjunKim, 2021-12-15:
I leave this Arduino libarary just for FYI, for someone who's interested in implementing SBUS into ELRS or making a converter:

https://github.com/bolderflight/sbus

cmanley, 2022-05-13:
Please also keep in mind that there are a lot of closed source stabilizers out there that only support SBUS such as the ZOHD KoPilot Lite, etc. One could of course add a CRSF to SBUS converter into the chain, but in a micro or DLG fixed wing model that would be a bit too much.

pstein29, 2022-07-05:
Does this help anyone? http://www.mateksys.com/?portfolio=crsf-sbus

secretman28, 2022-07-23:
its a shame im flying drone from the start of it almost.. sbus is not the best but when im flying my plane i want to keep the same rc link that inspire me confiance, dor now is crsf , cauz i can set the rx to output sbus i can plug in anything gyro, smooth flight , hobby eagle, bavarian deamon .and it work , nothing aupport cross fire protocol outside of the quad world.... till elrs get out i want to.make the move to change you  guys rock, but if incanot fly my plane , my turbine jet helico etc with it crsf stay on top for me i know your better at it butake it acessible to people that want it and ton of people will considere it cauz it will be a ton better ask a plane 3d pilot if i want more than the best 75-100hz shure they want what make them better at it... should not be that hard to make an option of sbus or crsf output and on top of that...... sbus dont need telemetry....

secretman28, 2022-07-24:
> Does this help anyone? http://www.mateksys.com/?portfolio=crsf-sbus

it add latency to the system

Dropsleutel, 2022-07-24:
What about this fork, anybody using it ?
https://github.com/mstrens/ExpressLRS/tree/Sbus-and-12-channels

secretman28, 2022-07-24:
> What about this fork, anybody using it ? https://github.com/mstrens/ExpressLRS/tree/Sbus-and-12-channels

this is nice find tanks

richardclli, 2022-10-14:
I will do it anyway, why not? Together with all choices: SBUS/SRXL/SRXL2/ibus/ibus2

secretman28, 2022-10-14:
Sbus is the main universal protocol for rc acessories if you wana put every one its still good to me

scooper1, 2022-11-18:
sbus would be useful for helis as well most do not have crsf only sbus or dsm

pkendall64, 2022-11-18:
> sbus would be useful for helis as well most do not have crsf only sbus or dsm

The main problem with S.BUS is that then people want S.Port telemetry. S.Port is a FrSky protocol which apparently has patents and they aggressively go after vendors that put S.Port into their receivers.

jmxp69, 2022-11-18:
> > sbus would be useful for helis as well most do not have crsf only sbus or dsm
> 
> The main problem with S.BUS is that then people want S.Port telemetry. S.Port is a FrSky protocol which apparently has patents and they aggressively go after vendors that put S.Port into their receivers.

I think it's completely fair to avoid anything that could be an IP violation. However, there are *plenty* of other applications for sbus without the need for s.port.

secretman28, 2022-11-18:
no need s.port, sbus plain and simple will unleash a bunch of possibility for plane heli people, 1 guy made a elrs fork output sbus,but still on 2.something... ask him if he can join your team just add an option when we flash crsf or sbus output.  still my opinion no need fport sport etcetc

deadbytefpv, 2022-11-19:
CRSF to SBUS converter, courtesy of Matek

secretman28, 2022-11-19:
add 7-14ms delay but yes...

jmxp69, 2022-11-19:
> CRSF to SBUS converter, courtesy of Matek

Right now that's the only option. It adds a point of failure, adds latency, and only accepts 5v, so in some cases, it requires now putting a BEC in front of the converter to provide the correct voltage which now means two additional points of failure.

padcom, 2022-11-22:
> While SBUS is a "legacy" protocol, it feels a bit premature to forget about supporting it already? I realize it's not ideal in the quad world, but everywhere else (like fixed wing and single rotor), it's still the standard aside from manufacturer specific stuff (like ExBus, XBus, SBus2, FPort, etc.). There's little to no CRSF support there. That may change eventually, but not any time soon.
> 
> Or is ELRS intending to stay focused on quads and maybe simple long range fixed wing aircraft? I would love to see it expand and become the go-to open source RF link for all RC aircraft eventually.

You know, when you talk to the great guys behind ELRS it kinda feels like this is a circle of people focused on one type of user: where the receiver is connected to INAV/Ardupilot/Betaflight. That's it. They impose certain rules (like arming on AUX1) claiming it is for safety reasons. Frankly speaking, I can see no reason for forcing arming to be the most important thing when I am 25km out with my plane and the only things I need it to do more than anything is steering and RTH. I don't give a ... about arming. So even though I am using the hardware/software combination and the name would suggest it's a long-range system. In point of fact, it is not even fully suited for that purpose too and forcefully putting the ARM on another channel can lead to unexpected circumstances..

I think that in all seriousness the point that needs to be made is that this is, after all, an opensource project and you and anyone else can fork it, mess around with it, even change how the packet structure looks like if you so desire, and make that link fit 100% your purpose. My idea is to make sure I strip it down from anything I don't need, and then maybe implement a super-long-range link using the great work done by the ELRS team. Sort of "standing on the shoulders of the giants", if you will..

padcom, 2022-11-22:
> > CRSF to SBUS converter, courtesy of Matek
> 
> Right now that's the only option. It adds a point of failure, adds latency, and only accepts 5v, so in some cases, it requires now putting a BEC in front of the converter to provide the correct voltage which now means two additional points of failure and more complexity than should be required if there were simply an sbus output option on the Rx.

Especially if the hardware is ready for it, and it is only the will of the developers to accept contribution that isn't the core team's goal.

scooper1, 2022-11-23:
this might make me start doing some programming again and go and fork this firmware but it would be nice to keep it in core so that other features come in.

i would be happy if it were only in the 16 channel mode and without arming

padcom, 2022-11-24:
You know, after I wrote those comments I decided to take a step back and to think, what can I do with the latest modes, especially the 100Hz/2 with faster telemetry? Since this is a fully (well, almost, but good enough) 15ch+AUX1 setup. My original setup was to have RTH on AUX1 and LAUNCH/ARM on AUX7. Shuffling stuff around I am now sending both AUX1 (1 bit) and AUX7, and I moved RTH to another free channel. On top of that, I've setup the dynamic power boost to use AUX12 which is the 16th channel that I currently use. I point of fact I am so far good with 4+5+1 channels but, that doesn't mean it every setup will be the same.

Doing a bit of testing at a close range (~2km) the switches react no different than before. It always does take a moment for the OSD to update (if it started blinking with blink off) so it might not even be possible to perceive the difference. I'd say the new modes are really for long range. The sticks are still sent with every packet, so some measure of control can be exercised and eventually either RTH will kick in or the signal is going to fall below noise level and failsafe will kick in.

It still would be cool to be able to select if the output should be SBUS (with telemetry) or CRSF (selected by default). It'd blast FrSky out of the market like in no time.

padcom, 2022-11-24:
Which basically means I surrendered to the pressure of the elders.

pkendall64, 2022-11-24:
> It still would be cool to be able to select if the output should be SBUS (with telemetry) or CRSF (selected by default). It'd blast FrSky out of the market like in no time.

We cannot do telemetry (S.PORT) as that is patented by FrSky. See my comment above https://github.com/ExpressLRS/ExpressLRS/issues/1182#issuecomment-1320424199

If you mean telemetry via sensors i.e. vbat, vario etc then that is do-able.

padcom, 2022-11-24:
That is what I meant, yes. I take it that an SBUS output (device generating sbus) is not a problem, but an input (device consuming SBUS) is? I am a bit fuzzy on that part... Or is generally S.PORT in a receiver patented, and no anybody shall implement it?

secretman28, 2022-11-24:
only std sbus no telemetry needed do the job for all aftermarket gyro, swash, servo extender controller telemetry is not the limiting factor it is the crsf output.

 if anything did futaba pattent the sbus.2 that have telemetry?

pkendall64, 2022-11-24:
> That is what I meant, yes. I take it that an SBUS output (device generating sbus) is not a problem, but an input (device consuming SBUS) is? I am a bit fuzzy on that part... Or is generally S.PORT in a receiver patented, and no anybody shall implement it?

From what we have been told, any manufacturer that has S.PORT input is in violation of the patent. So if ELRS added S.PORT input on the receivers then the manufacturers would not be able to use it. FrSky are very aggressive in chasing patent violations as well, from what we are told. So it's best to just steer clear of S.PORT.

secretman28, 2022-11-24:
> only std sbus no telemetry needed do the job for all aftermarket gyro, swash, servo extender controller telemetry is not the limiting factor it is the crsf output.
> 
> if anything did futaba pattent the sbus.2 that have telemetry?

but realy not needed for most people

richardclli, 2022-11-24:
Well, for US law, if one owns a hardware product.  It is completely legal to tap the wires, study what is going on in the signal and share / publish any findings.  And an opensource project is considered to be a way to share / publish the findings.  So it is almost impossible for Frsky to sue any opensource project based on any patent in protocols.  However, if a manufacturer made some s.port related hardware and sell, then the product can be considered to infringe the patent (depends what the patent claims and if evident can be found legally).

And currently, there are codes in OpenTX/EdgeTX that decodes S.Port based telemetry, and what Frsky can do is to change the protocol fromt time to time to make it becomes incompatible, but not any law actions.  Also, the MPM has a lot of reverse engineered protocols and no manufacturer has taken any law actions for all these years.

Dropsleutel, 2022-11-24:
I would also love to put an ELRS Rx with SBUS in some of my ancient models.  Could the Sbus code from Mstrens above, not be used in the master ?

ROSStargh, 2022-11-25:
It's funny to say that SBUS is too old so it should not be supported. That just makes me buy more SBUS receivers to fly my aircrafts without Betaflight controllers. And manufacturers of such aircrafts would rather not make products that support CRSF because they do not see many people using ELRS receivers to fly their products.

franz4949, 2022-11-25:
Here's another very interesting new protocol, named MiLo, with LoRa modulation in development stage, obviously satifying all wishes of fixed wing pilots and overcoming specific ELRS matters that may prevent fixed wing pilots from a basic switch to RLRS. And that code is intended to run on inexpensive ELRS hardware and will provide full 16 channels with 11 bit resolution, SBus as well as also S.Port  =>  https://github.com/midelic/MILO-RX

secretman28, 2022-11-25:
> It's funny to say that SBUS is too old so it should not be supported. That just makes me buy more SBUS receivers to fly my aircrafts without Betaflight controllers. And manufacturers of such [...]

exacrlg why in stilljpushing this, if we can set sbus out many not drone flyer will switch

pkendall64, 2022-11-26:
Ok, we are looking at SBUS, in fact I am working on a PoC at the moment, but we will not add S.PORT.

There are downsides to SBUS without S.PORT that users must be aware of as well.
1. no flight controller telemetry
2. passthrough flashing will not be possible, as the serial connection is from RX to FC only

To explain and hopefully clarify why I say "we will not add S.PORT".

As an open source project we could quite easily add S.PORT telemetry using other open source projects as a guide to what is expected. The problem that we face is that the manufacturers would then not be able to ship ELRS in their products for fear of legal action because of the patents that are held. We have been reliably informed that the holder of the patent is quite aggressive in taking legal action against any manufacturer that ships products that can receive S.PORT telemetry.

padcom, 2022-11-26:
Can we then officially reopen this GH issue, or at least link it to one that is about implementation of SBUS?

secretman28, 2022-11-26:
> Ok, we are looking at SBUS, in fact I am working on a PoC at the moment, but we will not add S.PORT.
>
> There are downsides to SBUS without S.PORT that users must be aware of as well.
>
> 1. no flight [...]

good for me all plane heli pepole just wnat to make the gyro or servoexpender work

padcom, 2022-11-26:
Would it be pushing if we asked for having it configurable between CRSF and SBUS in the web ui?

pkendall64, 2022-11-26:
> Would it be pushing if we asked for having it configurable between CRSF and SBUS in the web ui?

That is exactly what I'm doing 👍

secretman28, 2022-11-26:
your the man! 
seriously, amazing for expending the project to the next lvl 3d guy for plane and heli will realy understand what 333 hz servo are for and many of them will change to it 
tanks for taking the time to do this you guy listen the user a lot and we realy apreciate it !

RandMental, 2022-11-26:
Just want to put SBUS2 on the table that would allow telemetry options.

While doing abus, can we include SBUS2?

franz4949, 2022-11-27:
> To explain and hopefully clarify why I say "we will not add S.PORT".
>
> As an open source project we could quite easily add S.PORT telemetry using other open source projects as a guide to what is [...]

Here is the FrSky patent which obviously protects its SPort from reverse engineering and integration in any other non-FrSky receiver  => https://patents.justia.com/patent/20220358065

franz4949, 2022-11-27:
Mybe only solution might be a DIY receiver with S.Port ...

secretman28, 2022-11-27:
no need for all what people need. telemetry will work for what important, link rssi db etc, rx batt ,etc what realy needed will still work.
 if you have fpv... will still be on the screen....

franz4949, 2022-11-27:
not all are interested in flying fpv, nevertheless they are interested in getting telemetry for their fixed wing models transmitted to the transmitter and it's a deal-breaker for their decision to start and possibly switch to ELRS

padcom, 2022-11-27:
Not necessarily. Telemetry provided through control link does not have to be as good as a lower frequency dedicated two-way telemetry like the 433MHz modules commonly used with Ardupilot. Maybe I'm wrong, maybe not.. but the air seems to be pretty saturated with stick positions...

padcom, 2022-11-27:
@franz4949 I am not really getting what you mean. So, what you're implying is that if you don't get telemetry on your radio, then it is a deal-breaker? Seriously? 99.9% of the existing SBUS hardware doesn't support any telemetry, and those that do support barely anything beyond what the TX sends regarding link data. Also, flying BVLOS just with telemetry sent back to the transmitter is just plain irresponsible in my very humble opinion.

padcom, 2022-11-27:
> if you have fpv... will still be on the screen....

If you don't have FPV you don't fly BVLOS and you can use an old ACCST system if you specifically need SBUS with S.PORT telemetry. The range will be more than enough for such flights. For everything else just SBUS output will be more than enough.

jmxp69, 2022-11-27:
> Here is the FrSky patent which obviously protects its SPort from reverse engineering and integration in any other non-FrSky receiver => https://patents.justia.com/patent/20220358065

EIRP is regulated by governments yet some very prominent ELRS modules were just released with 1w max power from the manufacturer yet certain firmware allow for certain configuration changes to enable 2w power. Seems if shipping firmware used a non s.port version that could be configured later by the user this issue could be retired. 

In any case, the most important thing is to take the first step and turn on s.bus.

secretman28, 2022-11-27:
to make more people in sbus is the key , heli, plane people (los or 3d etc) dosent need telemetry you stilljget link strenght anyway tanks again to make this change im shure many people will apreciate you guys rock

[24 more replies]