ExpressLRS/ExpressLRS #1184: ExpressLrs 2.0  Flash via Passthrough R9MM/mini   - FAIL-!

al1965al, 2021-12-14:
The problem is observed in a lot of people.
ExpressLrs 2.0  Flash via Passthrough R9MM/mini   - FAIL-!
Photo
https://ibb.co/j87f4pp

The latest version of the configurator.
I tried to flash 2 receivers R9Mini  .via Passthrough  Error.
Version 1.2.0 is flashed via Passthrough without problems.

deadbytefpv, 2021-12-14:
Try Clearing PlatformIO dependency.. 

On the ExpressLRS Configurator -> Support -> Clear PlatformIO Dependencies

al1965al, 2021-12-14:
I'll try. Thanks.

al1965al, 2021-12-14:
Configurator 1.3.2
Right now in the same configurator.
Version 1.2.1 was flashed WITHOUT PROBLEMS!
The next version 2.0.RC1  error with the firmware.
.............................................................................................
Flash error
PackageException: Could not install package 'paolop74/extEEPROM @ ^3.4.1' for 'windows_amd64' system:
  File "C:\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 180:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\.platformio\platforms\ststm32\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 98:
    force=force,
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 104:
    % (spec.humanize(), util.get_systype())
========================== [FAILED] Took 3.80 seconds ==========================
Environment                                     Status    Duration
----------------------------------------------  --------  ------------
Frsky_RX_R9MM_R9MINI_via_BetaflightPassthrough  FAILED    00:00:03.797
==================== 1 failed, 0 succeeded in 00:00:03.797 ====================

al1965al, 2021-12-14:
All versions after 1.2.1 show a bug!

deadbytefpv, 2021-12-14:
This is your issue here:
PackageException: Could not install package 'paolop74/extEEPROM

that's not a bug of the firmware.. your network can't get to the correct package and install the dependencies for compiling the firmware.

al1965al, 2021-12-14:
That is, my network cannot find the package.
Another person from another country - also cannot find the package.
Very interesting.
And all other packages (versions 1.2.1 and below - find)
Don't you think it looks funny?

al1965al, 2021-12-14:
Thanks a lot.!
The question is closed.!
Everything is working!
I flashed 2.0 two planes.
I did- reinstal -clear Firmware Files-reclearing PlatformIO Dependencies

**Thanks for the quick help!**

closed 2021-12-14 by deadbytefpv