ExpressLRS/ExpressLRS #1185: No documentation that ARM must be on AUX1 _and_ be active HIGH

javamatte, 2021-12-14:
## Current Behavior

ELRS docs are clear about ARM needing to be on AUX1.

As of 2.0, however, there seems to be a requirement that the ARM mode needs to be active HIGH.
https://github.com/ExpressLRS/ExpressLRS/issues/1111#issuecomment-971320894

If it is a requirement, the docs need to be updated and it should be highlighted. Arming is serious business between the FC and the TX. Having the receiver impose requirements on that is a bad idea; ARM aux channel and inversion should be configured in the ELRS configurator by the pilot in my opinion.

## Steps to Reproduce

1. Read this link: https://github.com/ExpressLRS/ExpressLRS/issues/1111#issuecomment-971320894
2. Search documentation for "AUX1"
3. Note that there are no mentions of AUX1 being HIGH in the documentation.

deadbytefpv, 2021-12-15:
https://www.expresslrs.org/2.0/quick-start/pre-1stflight/#modes

SunjunKim, 2021-12-15:
> https://www.expresslrs.org/2.0/quick-start/pre-1stflight/#modes
> 
> https://www.expresslrs.org/2.0/software/switch-config/

I added some more highlights on them.

deadbytefpv, 2021-12-15:
@SunjunKim I made a PR too with regards to this..

SunjunKim, 2021-12-15:
> @SunjunKim I made a PR too with regards to this..

oops, coincidence.

deadbytefpv, 2021-12-18:
@javamatte Could this be closed now?

Were you not satisfied with the updates on the Docs linked?

javamatte, 2021-12-18:
Thanks for the quick updates!

The changes to the docs make the ARM/Aux1-High issue much more clear.

closed 2021-12-18 by javamatte

zakkhoyt, 2022-07-28:
This thread proved useful to me. I had been trying to get VTx backpack working (no goggles). I didn't understand why it wasn't working on my multiple setups. This thread is what got me to take a look at my Aux channel which turned out to be the problem. 

What I'm trying to say is that a mention about this in the backpack literature would be useful. Perhaps under the [Backpack Usage](https://www.expresslrs.org/2.0/hardware/backpack/esp-backpack/#backpack-usage) section.