ExpressLRS/ExpressLRS #1186: flywoo EL24E target has bad rssi dbm

wang322588, 2021-12-16:
## Current Behavior
I flashed the new target " flywoo EL24E rx "for my Firefly 1S Nano Baby Quad . When checking the signal health it has bad rssi dbm value.And the remote distance is short.

The rssi dbm is about -50 ~ -60 when the Tx and Rx between 1m far. Tx power is 10mw. I touch the rx with tx and value is still -35.
This is unnormal because flash the 1.2.0 version (DIY-2400-ESP8285-SX1280) works fine ,the dbm is -35. nearly -19 while touching.
2.0.0 add new target for flywoo ,I flash both DIY and FLYWOO but the issue remains.
## Steps to Reproduce
flash  flywoo EL24E rx
jumper t-lite flash DIY 2.4 lora 1280f27
set tx power 10mw dynamic power is off 
connet and check the dbm value 

1.

2.

3.

4.

## Possible Solution (Not obligatory)
I think the new rx firmware maybe the problem. Tx works fine while connecting other rx.

## Details

## Your Environment

- TX hardware:flywoo EL24E
- RX hardware:DIY 2.4G lora1280F27
- Handset model: jumper t-lite
- OpenTX version (including nightly number) 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): both 2.0.0 (5c48083)
- Packet Rate:150
- Telemetry Ratio:1:32
- user_defines:default settings

SunjunKim, 2021-12-17:
> I think the new rx firmware maybe the problem. Tx works fine while connecting other rx.

What are the "other rx"? Could you list them, please?

wang322588, 2021-12-17:
I have a BETAFPV Lite elrs receiver which bahave normally.

wang322588, 2021-12-20:
I tested betafpv 2.4G lite receivers seems to be the same problem. Nano receiver is good.So I think it is not the software issue.Close it.

edernucci, 2021-12-22:
@wang322588 I have a suspect here and I need your help testing it. Can you please change your module firmware to version 1.2.1 and test again? I'm afraid it's a problem similar to the one we had with BetaFPV's TX.

wang322588, 2021-12-23:
@edernucci I will try

wang322588, 2021-12-24:
@edernucci I tested the version 1.2.1.Works fine.So I reopen it I think the 2.0.1 firmware also have this issue.

edernucci, 2021-12-24:
@wang322588 I'm confused. What is your TX? Is a 2.4 DIY?

wang322588, 2021-12-25:
Yes.A ESP32 and SX1280lora

wang322588, 2021-12-31:
On receivers using an SMD antennas, expect lower values than those equipped with wire antennas. If the receiver with the SMD antenna has heat shrink, enclosed in a whoop canopy, surrounded with carbon parts or buried inside a plane/fixed wing, expect some more attenuation of the signal.quote by elrs webinfo

closed 2021-12-31 by wang322588