ExpressLRS/ExpressLRS #1190: SBUS output for helicopters

robhaswell, 2021-12-16:
I would like to move my helicopter fleet over to ELRS. While ELRS's standout features of range and command rate do not necessarily apply to LOS helicopters, there are still plenty of problems with existing control link ecosystems which make it attractive to move over to ELRS, especially for future builds.

Unfortunately most common helicopter flight controllers ("FBL units") only accept legacy protocols such as SBUS or IBUS. This would require that something like SBUS output would need to be added to ELRS. #581 discusses this.

There is an additional issue with helicopters, which is that there are typically 6 channels - AERT1C. For helicopters the "C" channel (Collective) must be full-resolution, however there is no need for full-resolution Throttle, so only 4 full-resolution channels is needed. For Throttle, a 6-bit channel would be fine, so there is already support for all the required channels in ELRS 2.0. Additionally the AUX1 channel is not "Arm", but is mostly only used to indicate if the helicopter is in self-levelling mode or not. This only requires 1-bit so the existing "Arm" channel should be fine. The only change which would be needed would be that the RX would need to map its usual channel output into different channels when using SBUS, so that a full resolution channel can be output on CH6 for Collective.

So, to summarise, the two changes I think needed to make ELRS suitable for helicopter pilots would be:

* SBUS output support
* Channel output remapping

Thanks for your consideration.

*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

CapnBry, 2021-12-16:
I appreciate the detailed information about how helicopters use their channels, but as we've said many many many times, SBUS output is not something we will implement. If SBUS does eventually make it into ELRS, then this can be reopened.

closed 2021-12-16 by CapnBry

poikm302, 2024-06-02:
[SBUS: swapped ch3 and 6](https://github.com/ExpressLRS/ExpressLRS/commit/76a82dd10b7a94f897b2081533c81c5c9be483bf)

(https://github.com/ExpressLRS/ExpressLRS/discussions/2479)

poikm302, 2024-06-02:
https://github.com/ExpressLRS/ExpressLRS/discussions/2479
76a82dd10b7a94f897b2081533c81c5c9be483bf

(original author)https://github.com/mha1