ExpressLRS/ExpressLRS #1195: Not able to build RC3, RC4 and 2.0.0 on Linux (deb) configurator

jaapf4fs, 2021-12-17:
### Discussed in https://github.com/ExpressLRS/ExpressLRS/discussions/1194

<div type='discussions-op-text'>

<sup>Originally posted by **jaapf4fs** December 17, 2021</sup>
Hi I get the following ERROR when I try to build RC3, RC4, 2.0.0 or  2.x-maintenance branch.

Target HappyModel_EP_2400_RX_via_WIFI and HappyModel_ES24TX_2400_TX_via_WIFI

*************************************************
BUILD ERROR
AttributeError: module 'gzip' has no attribute '_COMPRESS_LEVEL_BEST':
  File "/home/disk/.platformio/penv/lib/python3.6/site-packages/platformio/builder/main.py", line 179:
    env.SConscript(item, exports="env")
  File "/home/disk/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/disk/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/disk/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/build_html.py", line 59:
    build_common(env, "rx_index.html")
  File "/home/disk/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/build_html.py", line 47:
    build_html(mainfile, "INDEX_HTML", out, env)
  File "/home/disk/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/build_html.py", line 39:
    out.write(','.join("0x{:02x}".format(c) for c in compress(data.encode('utf-8'))))
  File "/home/disk/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/build_html.py", line 25:
    with gzip.GzipFile(fileobj=buf, mode='wb', compresslevel=gzip._COMPRESS_LEVEL_BEST, mtime=0.0) as f:
========================== [FAILED] Took 0.99 seconds ==========================
Environment                     Status    Duration
------------------------------  --------  ------------
HappyModel_EP_2400_RX_via_WIFI  FAILED    00:00:00.986
==================== 1 failed, 0 succeeded in 00:00:00.986 ====================

*************************************************

What I tried so far

    Deleted the configurator v1.3.2
    checked the dependencies
    Updated to Python 3.8.12
    fresh install configurator v1.3.2
    Cleared the platformio dependencies
    Cleared the firmware files
    tried the 2.x-maintenance branch
    Still the same error on RC3 RC4  2.0.0 and 2.x-maintenance branch  :-(

**RC1 and RC2 are no problem**
    rgds,
    Jaap
</div>

pkendall64, 2021-12-18:
This is fixed in 2.0.x-maintenance branch by PR #1173

jaapf4fs, 2021-12-18:
It's working, 
also on Target 2.0.0 when I change "compresslevel=gzip._COMPRESS_LEVEL_BEST" to "compresslevel=9" in the build_html.py file..

brandonrc, 2021-12-19:
@jaapf4fs we fixed this in the firmware but there is no way we can change that in the tags. The solution is to just run the latest firmware and there should not be a problem.

closed 2021-12-19 by brandonrc

brandonrc, 2021-12-19:
For anyone else that wants to build on a linux system, you have to install python 3.9 which contains the new GZIP functions. 

If you do not want to update to the latest python version we did patch this so in `2.0.1` it should work with old python.