ExpressLRS/ExpressLRS #1203: Security concern with bind phrase

brandonrc, 2021-12-21:
A few people on JB's live chat asked about security threat with bind phrases. What happens if someone uses your phrase and connects to your drone before you do. Or what happens if someone uses the same phrase.. can they take over control? 

https://www.youtube.com/watch?v=z-BgE34tc58

Just like we have lock on packet rate on first connection, should we try a lock on tx with first connection. Not sure what unique identifier we could use on the TX? Is this even something we could do? 

Someone else brought up maybe only using that bind phrase for the first lock then it would not use it anymore. I am not sure I like that idea because it would make it hard to have multiple TX's right? 

*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

brandonrc, 2021-12-21:
After speaking with a few people, looks like this issue is very unlikely to cause issues. 

1. You have to have 6 consecutive packets for BF to disarm. 
2. You have to have the same bindphrase
3. The TX does not start in sync, so 'most' of the time they don't collide

Maybe if companies ship with a default bind phrase, then we might have some issues.

brandonrc, 2021-12-21:
Closing this until a company makes a bad decision with shipping all systems with default bind phrase.

closed 2021-12-21 by brandonrc

avsaase, 2021-12-21:
One thing I like about ExpressLRS is that I can switch radios in-flight. Making this change would make that impossible. 

The documentation is also pretty clear on setting a unique bind phrase. Since it is not a secret you can just use your name and birth date, that should rule out any collision.

brandonrc, 2021-12-21:
> One thing I like about ExpressLRS is that I can switch radios in-flight. Making this change would make that impossible.
> 
> The documentation is also pretty clear on setting a unique bind phrase. Since it is not a secret you can just use your name and birth date, that should rule out any collision.

Agreed. I do not think this is a big enough issue considering @JyeSmith and @AlessandroAU have flown together with the same phrase with little to no issues. 

I would also think lock on first connection could be a default option that one could turn off like our packet rate. 

Still do not think it is worth the dev effort for this edge case, unless a company really messes up.. but then we could just tell users to flash a unique bind phrase and solve that.