ExpressLRS/ExpressLRS #1205: VTX Administrator  unexpected behavior wiht model ID match

bike797, 2021-12-21:
## Current Behavior
 

1. Make Model A and B. (or already been created model) using model match.
   set different model ID A(ID 11) and B(ID 12).
2. Select Model A
    Setting VTX Adminstrator (Band, Channel, Pwr Lvl ... )
    ex) Band R, Channel 8, Pwr 3
3. select change Model B
   Setting VTX Adminstrator
   ex) Band E, Channel 6, Pwr 5
4. selcect change Model A
   check VTX Adminstrator
   each values ​​have changed to Model B instead of Model A

## Steps to Reproduce

1. Model A VTX setting parameter changed to model B.

2. When models A and B use different VTX products, unexpected behavior occurs in pit mode, output, etc.

3. Pwr Lvl 2 value can be 200mW output for A model and Pit mode for B model

## Possible Solution (Not obligatory)

 1. Best solution
 - Put VTX Adminstrator prameter setting into Model Match function.

  2. Second solution
  - Put VTX Adminstrator On/Off key into LUA.
## Details

## Your Environment

- TX hardware: Happymodel 2.4
- RX hardware: Happymodel EP1, EP2, SPI FC
- Handset model: RadioMaster TX16S 
- OpenTX version (including nightly number) EdgeTX Ver2.6.0(d775e75e)
- ExpressLRS version (TX & RX MUST MATCH): ExpressLRS 2.0
- Packet Rate: any
- Telemetry Ratio: any
- user_defines: default

CapnBry, 2021-12-21:
Not all the configured values are per-model, there's not enough storage space to store them all. [As per the docs](https://www.expresslrs.org/2.0/software/model-config-match/#model-configuration), the only per-model configuration items are:
```
Setting | Description
-- | --
Packet Rate | The RC update frequency (500Hz, 250Hz, etc)
Telem Ratio | Telemetry ratio (Off, 1:128, 1:64, etc)
Switch Mode | Method for sending switches to the receiver
Model Match | Enable the model match feature (see below)
Max Power | Transmitter output power level
Dynamic Power | Enable Dynamic Power switching
```

bike797, 2021-12-22:
When the power is turned on, the channel and power level set in TX are automatically transferred to the VTX via RX.
VTX Adminstrator is user friendly and fantastic Function in ExpressLRS 2.0.

But if the VTX administrator behaves like it is now, I don't want to use this function.
This makes the user uncomfortable in some situations, contrary to the initial design intent.

Let's assume the following situation.
The following VTX Table is applied to Model A and B.

Model A VTX 
> vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

Model B VTX
>vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 0 5885 5905 0 0 
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 2
vtxtable powervalues 0 1 
vtxtable powerlabels 25 200

1st flight with VTX A (Model A)
 - VTX Power level set to 4 in ExpressLRS VTX Adminstrator LUA.
change model and flight with VTX B (Model B)
 - VTX Power level is automatically set to 4. 
   However there is no 5 powervalue in model B

In the opposite case, there is the same problem.
The powervalues ​​to set the VTX output 200mW are different for each model.

In this case, you have to manually change the VTX power level whenever you change the drone model in TX(Handset).

If there is an OSD, you can check whether the current VTX power level setting is the desired value.
However, if there is no OSD, you cannot check the current power level except for the LED status of the VTX.
Most users do not remember the power level according to the LED state.

If memory space is a problem, even if Channel, Bandn, Pit Mode is excluded, it is good and I think that Pwr Lvl is included in the model match.
Or at least you can turn it on/off to prevent the VTX from being set to an unintended power level, I think.

SunjunKim, 2021-12-22:
So an easy solution would be not sending VTX settings on connect.

Currently, it seems like VtxSendState is set VTXSS_UNKNOWN when disconnected, and update VTX table when it's firstly connected. I understand the intention of this behavior, but it seems to like causing a problem when dealing with different VTXs on different models.

I'd like to suggest updating VTX only if "SEND VTx" command is issued on the lua script.

SunjunKim, 2021-12-22:
Maybe @StonedDawg and @JyeSmith can discuss about this issue?

JyeSmith, 2021-12-22:
> > vtxtable powerlevels 5
> > vtxtable powervalues 25 100 200 400 600
> > vtxtable powerlabels 25 100 200 400 600
> 
> > vtxtable powerlevels 3
> > vtxtable powervalues 0 0 1
> > vtxtable powerlabels 25 25 200
> 

The easiest solution is to make the VTx index match as above.

> I'd like to suggest updating VTX only if "SEND VTx" command is issued on the lua script.

Absolutely not :)

SunjunKim, 2021-12-22:
> > > vtxtable powerlevels 5
> > > vtxtable powervalues 25 100 200 400 600
> > > vtxtable powerlabels 25 100 200 400 600
> > 
> > 
> > > vtxtable powerlevels 3
> > > vtxtable powervalues 0 0 1
> > > vtxtable powerlabels 25 25 200
> 
> The easiest solution is to make the VTx index match as above.

I can see that's one way of solving this, but isn't it causing lots of confusion? as the author experienced?
The solution seems not very trivial for everybody.

CapnBry, 2021-12-22:
Yeah I am with Jye 100% we have to send the VTX settings on connect, that's the whole point of the feature. It definitely is a problem though as bike797 points out that the power level isn't the same across multiple models so it should be a per-model setting.

I'm not sure what to do about this because storing all the VTX settings per model increases the size of the config by 128 bytes which I think will make it overflow the size available on STM-based TX. Only storing some of the VTX settings per-model is too confusing too, I think it needs to be all or nothing. An option would be to reduce the number of models we support from 64 to 32, which is still a lot but OpenTX supports 64 IDs so I feel like that could be confusing too.

I really don't have a solution for this that doesn't compromise more that it achieves.

Pelado-Mat, 2021-12-22:
I also experience this issue with one of my quads, was confusing at first but I finally did understand what was happening, of course after a crash behind the trees as the vtx was in 25mw...

I think that per model VTX has another downside. Its really convenient to fix the freq once and then keep the same config while changing quads.  Personally I prefer the current behavior, having a "global" VTX Admin setting.

Perhaps this issue is just a consequence of ERLS not been aware of the limitations of the VTX (ergo... vtx tables) It just pushes blindly a power level to the FC and hopes for the best.

StonedDawg, 2021-12-22:
we could have vtx administrator not to send MSP command if modelMatch is On, meaning that the VTX administrator will only work with those RX that't doesn't use modelMatch (universal rx) since the VTX administrator is acting universal (same towards all model), but I still prefer having it to set the same channel and band to any of my model rather than having my vtx administrator suppressed because my modelmatch is on. usually one guy sticks to 1 vtx frequency for the whole day... (at least this is what is happening around me only).

SunjunKim, 2021-12-22:
How about the `No Sync`, or `Keep current` in the option, as the author suggested?

CapnBry, 2021-12-22:
I don't like tying VTX to MM either, since Model Match and VTX Admin are not in any way related. 

> How about the `No Sync`, or `Keep current` in the option, as the author suggested?
"Off" is the same as No Sync and is what we use to key off of sending the whole packet. Betaflight and iNav do not support a partial update, so individual parameters can't be set to No Sync. "keep current" would require us to query the current settings then send them back again which is a lot of work and also I don't think gets us much closer to a solution for all the complexity it would have to add. Also adding more options to channel will overflow their storage haha.

SunjunKim, 2021-12-22:
> Betaflight and iNav do not support a partial update,

Oh I didn’t know that. Then for the simplicity the current off/on seems to be the best practice.

pkendall64, 2021-12-22:
You can also set the power level in LUA to `-`, which means "don't change power level". i.e. the VTx stays at the BF configured power level.

Pelado-Mat, 2021-12-22:
> You can also set the power level in LUA to `-`, which means "don't change power level". i.e. the VTx stays at the BF configured power level.

That should be documented! 
[image]

CapnBry, 2021-12-22:
> That should be documented! [image]

I didn't put in the documents because it's not necessarily true. Setting it to - in iNav I believe prevents you from arming until you change it to something else. Or at least it used to at the time the docs were written. I think in betaflight it's also not intentional that it doesn't change the power, it just doesn't because it's out of range.

Pelado-Mat, 2021-12-23:
> I didn't put in the documents because it's not necessarily true. Setting it to - in iNav I believe prevents you from arming until you change it to something else. Or at least it used to at the time [...]

Its still true just tested with INAV 3.0.2

bike797, 2021-12-23:
From what I've checked, "-" seems to turn off the output or change it to another value in Betaflight. rather than keeping the existing values.
Let's check again if this is true.

pkendall64, 2021-12-23:
> From what I've checked, "-" seems to turn off the output or change it to another value in Betaflight. rather than keeping the existing values. Let's check again if this is true.

Looks like BF sets the power to "None" when we send `-` i.e. 0.

CapnBry, 2021-12-23:
Yeah so I guess let's just get PRs in to BF and iNav to not change anything if the power sent is 0? Both seem to have not-good behavior so if we can get it to just not change, then that would be a step in the right direction.

bike797, 2021-12-23:
I agree that the CapnBry suggested idea is one of the ways to solve the problem.
However, I wonder if inav and BF will agree to these changes.

pkendall64, 2021-12-23:
iNAV and BF already have code that skips setting the Power level if it's not in the MSP packet.
I have a PR for ELRS ready to go.

kimman1, 2021-12-29:
Can we read vtx config over telemetry or MSP and sync it to lua script ?

CapnBry, 2021-12-29:
I definitely don't want to do that. The ELRSV2.lua is just a shell that displays a list of items sent by the TX/RX/etc so I don't want to get into it querying the FC to get more info, especially because it wouldn't even know what to do with it-- it doesn't know what the "VTX Power" is, it just knows to display some arbitrary text with some other arbitrary text next to it.

Even if it did query the FC and knew where to put it, then what? That's like the reverse of VTX Admin, where it will overwrite what you assigned with whatever the FC has set. I think pkendall64's solution is a good compromise in that it will always set the band/channel no matter what model (since you only have one goggles so everything needs to be on the same channel) but only set the power/pit mode if you set that to not "-" which means it won't muck with the power across different models if they have different power settings.

Dropsleutel, 2021-12-31:
Possibly also the safest in race events etc. When bringing multiple quads to the event.   You always turn on your Transmitter first,  can check or set the Vtx Channel to the one you got assigned etc, then power the Quad.    But that brings a question, VTX in Quad should only en-power, after receipt it's ELRS VTX settings.  (and not start upon previous VTX betaflights setting.)

closed 2022-01-13 by bike797