ExpressLRS/ExpressLRS #1206: SPI RX Packet Rate index not follow TX Packet Rate when rx is bind

bike797, 2021-12-21:
## Current Behavior

Not sure if the value set in TX applies to RX.
 - The packet rate should be set equal to Tx when rx and tx are first connected.
 - Packet rate of rx can be changed only in CLI.
 - The packet rate output to the OSD matches the TX value.
    Inconsistent with the packet rate checked in CLI.

Is the currently set packet rate the value of TX? Or is it the value set in RX?

## Steps to Reproduce

1. Connect SPI RX via betaflight Cli 
2. Bind SPI RX and TX module (set packet rate 500hz) 
3. Turn off TX
4. Change SPI RX expresslrs_rate_index = 3 save.
   Double check! paramter has changed. 
5. Turn off RX
6. Turn on TX and set TX Packet Rate 500Hz (check 500hz)
7. Turn on and connect SPI RX via betaflight Cli. 
    Check bind with tx.
    Check RX parameter (expresslrs_rate_index). 
8. TX Packet Rate is 500Hz but RX rate_index still 3(50Hz)
    Not change
9. Check OSD Packet rate 
    OSD show 500Hz

## Your Environment

- TX hardware: HappyModel 2.4 TX
- RX hardware: HappyMdoel ExpressLRS X 2.4 FC SPI RX
- Handset model: RadiomasterTX16S
- OpenTX version (including nightly number) : EdgeTX Ver2.6.0(d775e75e)
- ExpressLRS version (TX & RX MUST MATCH): TX- ExpressLRS 2.0,  RX-betaflight_4.3.0_CRAZYBEEF4SX1280_eb873308c.zip
- Packet Rate: any
- Telemetry Ratio: any
- user_defines: default

pkendall64, 2021-12-21:
The rate is set by the TX. Therefore I think this bug really needs to be raised against the betaflight PR.

phobos-, 2021-12-22:
@pkendall64 I'd close this. Issue has been raised and fixed in the bf PR.

bike797, 2021-12-22:
This issue is clearly fixed after e00a3abc version.
THX

closed 2021-12-22 by bike797