ExpressLRS/ExpressLRS #1208: 1RSS very low on Betafpv 900Mhz micro TX

Pierre-Antoine2707, 2021-12-22:
## Current Behavior

1 RSS <-50dbm at 1m distance (TMPR=100mW)

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: Betapfv 900Mhz micro
- RX hardware: HappyModel ES900Rx
- Handset model: TX16s
- OpenTX version (including nightly number) : EdgeTX 2.5
- ExpressLRS version (TX & RX MUST MATCH): 2.0.0
- Packet Rate: 200Hz
- Telemetry Ratio: 1:32
- user_defines:

wang322588, 2021-12-22:
https://github.com/ExpressLRS/ExpressLRS/issues/1186

edernucci, 2021-12-22:
@Pierre-Antoine2707 @wang322588 please flash with branch 2.0.x-maintenance. There's a issue with 2.0 and BetaFPV TX.

wang322588, 2021-12-23:
@edernucci The 2.0.x-maintenance is not solve the problem.

Pierre-Antoine2707, 2021-12-23:
@edernucci It works but the TX becomes completely instable at 500mW

brandonrc, 2021-12-23:
https://github.com/ExpressLRS/ExpressLRS/releases/tag/2.0.1

https://github.com/ExpressLRS/ExpressLRS/pull/1198

Closing this issue because it was fixed and patch in 2.0.1.

brandonrc, 2021-12-23:
> @edernucci It works but the TX becomes completely instable at 500mW

I hate to say this but I recommend not running it at 500mW. I know betaFPV is releasing a new 1000mW version and I hope they are addressing these issues. 

I am willing to bet your TX is getting hot and that sx127x radio chip does not like heat.

brandonrc, 2021-12-23:
Closing this issue and opened a new issue with the unstable 500mW problem. https://github.com/ExpressLRS/ExpressLRS/issues/1215

closed 2021-12-23 by brandonrc