ExpressLRS/ExpressLRS #1209: Define for `-DMY_BINDING_PHRASE` no longer working - needs to use `-DMY_UID` when building TX firmware otherwise RX cannot bind.

howels, 2021-12-22:
## Current Behavior

Define for `-DMY_BINDING_PHRASE` no longer working - needs to use `-DMY_UID` when building TX firmware otherwise RX cannot bind.

## Steps to Reproduce

1.  Build using the `-DMY_BINDING_PHRASE=blah` define

2. RX does not bind

3. Build using the `-DMY_UID=43242322` define (copying value from one auto-generated in configurator when using the normal non-manual options dialogue)

4. RX does bind

## Possible Solution (Not obligatory)

Either correct the documentation for defines and render -DMY_BINDING_PHRASE obsolete, or ensure that -DMY_BINDING_PHRASE now generates the required -DMY_UID value.

## Details

Tested on https://github.com/ExpressLRS/ExpressLRS/commit/4540c15aca4b6c89c649b535b8fe0b350fc2500c

## Your Environment

- TX hardware: BetaFPV 2400 Micro
- RX hardware: HM EP1
- Handset model: 
- OpenTX version (including nightly number)
- ExpressLRS version TX https://github.com/ExpressLRS/ExpressLRS/commit/4540c15aca4b6c89c649b535b8fe0b350fc2500c RX 2,0
- Packet Rate: 500
- Telemetry Ratio: 1:128
- user_defines:

deadbytefpv, 2021-12-22:
`build flags: ['-D PLATFORM_ESP32=1', '-D CONFIG_TCPIP_LWIP=1', '-D BEARSSL_SSL_BASIC', '-I ${PROJECTSRC_DIR}/hal', '-DTARGET_TX=1 -Wall -Iinclude', '-DRADIO_2400=1', '-include target/DIY_2400_TX_ESP32_SX1280_E28.h', '-D VTABLES_IN_FLASH=1', '-O2', '-DMY_BINDING_PHRASE="nospacesbindingphrase"', '-DMY_UID=199,25,167,100,92,154', '-DFEATURE_OPENTX_SYNC', '-DUSE_TX_BACKPACK', '-DUART_INVERTED', '-DAUTO_WIFI_ON_INTERVAL="60"', '-DWS2812_IS_GRB', '-DFAST_SYNC', '-DLATEST_COMMIT=52,53,52,48,99,49', '-DLATEST_VERSION=72,69,65,68', '-DTARGET_NAME=DIY_2400_TX_ESP32_SX1280_E28']`

deadbytefpv, 2021-12-22:
[image]
[image]

howels, 2021-12-22:
TLDR; quotes matter because the entire key=value is hashed to make the UID, not just the binding phrase itself.

closed 2021-12-22 by howels