ExpressLRS/ExpressLRS #1215: BetaFPV Micro 900mhz unstable at 500mW

brandonrc, 2021-12-23:
## Current Behavior
@Pierre-Antoine2707 reported that at 500mW betaFPV 900mhz TX module is unstable. 

I believe that we are seeing a heat issue with the tx. The sx127x chips do not handle heat well and well become unstable at high temps. 

## Steps to Reproduce
Set BetaFPV Micro 900mhz at 500mW and fly. 

## Possible Solution (Not obligatory)
1. Check the output power and make sure we are not exceeding 500mW. If we can reduce the power, that might help reduce the heat. As long as we are not giving people less than advertised power. 

2. Improve the cooling, more of a BetaFPV thing..

## Details

## Your Environment

- TX hardware: BetaFPV micro 900mhz
- RX hardware: Any 900mhz

brandonrc, 2021-12-23:
@Pierre-Antoine2707 please add any information to this issue.

Pierre-Antoine2707, 2021-12-26:
@brandonrc With 2.0.1 the TX become unstable (telemetry lost every 2'', impossible to enter LUA). Only recovery possible: flash with 2.0.0 where 500mW is stable but with low 1RSS issue
RX : Happymodel ES900RX
Handset: Tx16s + EdgTX 2.5.

pkendall64, 2021-12-26:
I just ran the BetaFPV 915 Micro TX for 10mins at 500mW on 2.0.1 with a BetaFPV 915 RX and there was no problem. The fan was going in the TX. The link was stable at around -70-90 dBm 100%LQ pretty much the entire time.

Pierre-Antoine2707, 2021-12-26:
I just tried with a 2nd BetaFPV 915 Micro TX, I had the same issue at 500mw. The TX becomes immediately instable after switching to 500mw so I don't think it is heating issue.

brandonrc, 2021-12-30:
> I just tried with a 2nd BetaFPV 915 Micro TX, I had the same issue at 500mw. The TX becomes immediately instable after switching to 500mw so I don't think it is heating issue.

are these new TX's or were they purchased during the first batch? 

I had a prototype betaFPV 900mhz that just stopped working completely. 

It started with telemetry lost left and right. 

Think they addressed the issue.

Carlospepunto, 2022-01-03:
I have a lot of problems with Betafpv 900 mhz, with 2.0.0 low RSS and with 2.0.1 impossible to execute lua at 400.000 baud rate and lost of telemetry messages at higher baud rates.

brandonrc, 2022-01-04:
> I have a lot of problems with Betafpv micro 900 mhz, with 2.0.0 low RSS and with 2.0.1 impossible to execute lua at 400.000 baud rate and lost of telemetry messages at higher baud rates. Frsky R9 tx with elrs works fine. Radiomaster TX16S with EdgeTX 2.5 and Betafpv micro 900 MHZ at 868 eu regulation domain, r9mm rx

yea.. we are having a lot of issues with betafpv micro 900 mhz... 

Does this work with 115k baud rate?

brandonrc, 2022-01-04:
@HWT662988

Carlospepunto, 2022-01-04:
> 

> 

With 115k bauds lua works better but very slow

Liber-coder, 2022-01-09:
I ran into the same problem.
When the voltage of tx16s is lower than 8.0V, turn on 500mw, the voltage will jump between 7.8V and 8.0V, and the betafpv 900mhz module cannot work normally, its LED will flicker, it seems that the output power of tx16s is not enough. Caused the module to switch between off and on.
At the same time, I did a test on the x9d remote control. When 500mv was turned on, the voltage of x9d was reduced from 8.0V to 7.3V, but it worked normally.
-----
tx16s: EdgeTx2.4.0 and 2.6.0rc2
betafpv micro 915mhz: 1.0.0 and 2.0.1

brandonrc, 2022-01-09:
> I ran into the same problem. When the voltage of tx16s is lower than 8.0V, turn on 500mw, the voltage will continuously jump between 7.8V and 8.0V, and the betafpv 900mhz module cannot work normally, its LED will flicker, it seems that the output power of tx16s is not enough. Caused the module to switch between off and on. At the same time, I did a test on the x9d remote control. When 500mv was turned on, the voltage of x9d was reduced from 8.0V to 7.3V, but it worked normally.
> 
> My Environment: tx16s: EdgeTx2.4.0 and 2.6.0rc2 betafpv micro 915mhz: 1.0.0 and 2.0.1 My Micro 900 TX is one of the first batch, with the small USB plug hole in the case.
> 
> Related issue: #1248

This makes me feel like it is a hardware issue for sure :(

The new module seems to work fine so for some reason it is pulling to much power which definitely sounds like a hardware issue. 

Thanks for the update! I will close this issue but let BetaFPV know :)

closed 2022-01-09 by brandonrc