ExpressLRS/ExpressLRS #1224: Failure to flash a R9MX with INAV passthrough

OlivierC-FR, 2021-12-28:
- RX hardware: Frsky R9MX
- OpenTX version (including nightly number) : 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): 2.0.1
- FC : Matek F722-SE INAV 3.0.2

R9MX is with its correct ELRS bootloader, but then the flashing fails, going through a Matek F722-SE running INAV 3.0.2
I tried to check the box at the bottom (UART_RECV SPD something don't recall the exact name) to 115200 or 57600, didn't seem to make any difference)

Screenshot 1 and 2 below are if I don't push the receiver button before plugging the USB cable, and screenshot 3 is when the button is pushed prior powering. The R9MX shows blinking red/green as expected.
Before that I managed to successfully flash an identical R9MX, through a Matek F405-WSE, also running INAV 3.0.2, but it took like maybe 20 attempts, maybe had a lucky one, I don't know.

[image]

[image]

[image]

Dropsleutel, 2021-12-31:
I can remember I had the same issue with I nav. Strangely it only worked when set to Auto.....
I made a screenshot at the time, only set as such pass true flash it worked; 

[image]

OlivierC-FR, 2021-12-31:
Thanks, I will try with auto, I have few more receivers to flash

pkendall64, 2022-01-22:
@OlivierC-FR Did you get this working, can we close this issue?

OlivierC-FR, 2022-01-22:
> @OlivierC-FR Did you get this working, can we close this issue?

Can be closed I guess, over 4 R9MX receivers, I managed to flash only 1 using INAV, by sheer luck and after trying like 20 times. 2 others I flashed at the first attempt with Betaflight, and the last one never wanted to be flashed for whatever reason. So I guess it's an issue with INAV passthrough.

closed 2022-01-22 by pkendall64

bill-zb, 2022-01-26:
HI, I have exactly the same issue as OliverC-FR in almost the same environment:
RX hardware: Frsky R9Slim+ OTA
OpenTX version : 2.3.14
ExpressLRS version: 2.1
FC : Matek F722-WING INAV 3.0.1

I'm not even including the screenshot, because they are identical as in the Oliver's issue #1224. 
Is there something specific in Inav, that prevents successful flashing? Yesterday I flashed 2 quads with BF4.3 and R9MM without a hiccup...

bill-zb, 2022-01-27:
Just to not leave this issue unnecessarily open - I fixed it by flashing bootloader 0.5.3, as proposed in the issue #1256. So for me, this one can be closed.