ExpressLRS/ExpressLRS #1227: Fan turns off on high power level

thegavereguy, 2021-12-29:
## Current Behavior

The fan speed decreases as the power output rise. It spins the fastest on 10mw and it turn completely off on 500mw (max power output of the module). I don't know if the values are inverted in the lua script (10mw sets the power to max) and i don't a meter to test it.

## Steps to Reproduce

1. Flash latest stable version (2.0.1)
2. Power on the radio and change power level in the lua script

## Details

- On the BetaFPV branch for the module the fan works fine

## Your Environment

- TX hardware: BetaFPV 2.4Ghz Micro Tx
- Handset model: Radiomaster TX16S
- EdgeTX version (including nightly number): 2.4.0
- ExpressLRS version (TX & RX MUST MATCH): 2.0.1

pkendall64, 2021-12-29:
This is really weird as we only have control over the fan being on or off. I have noticed on my module that when I hit 500mW the fan slows down and I start to get CRC errors, so I think that the module is drawing too much power.
Have you set the fan threshold in LUA to 10mW? That is the only reason that it would come on at 10mW.

brandonrc, 2021-12-30:
> This is really weird as we only have control over the fan being on or off. I have noticed on my module that when I hit 500mW the fan slows down and I start to get CRC errors, so I think that the [...]

maybe the PA is pulling to much power? 🤣 I am not an EE.. I can test this out. 

You have the same module right? (Actually you have the new 1w prototype one) Set the fan on with 10mw.. and increase the power.. see if the fan slows down..

brandonrc, 2021-12-30:
@pkendall64 I cannot reproduce this issue on any TX.. including the betaFPV module.

brandonrc, 2021-12-30:
@thegavereguy does this happen when you bind to a rx as well? 

1. Not connected to the drone, does this happen? 
2. When you are connected to the drone, does this happen?

brandonrc, 2021-12-30:
Tested this on 3 betaFPV modules and no issue with higher power and the fan.

closed 2022-01-04 by brandonrc

brandonrc, 2022-01-04:
@HWT662988

Closed this issue because it looked like a hardware issue. FYI