ExpressLRS/ExpressLRS #1229: DIY Target won't bind in v2.x (JHEMCU specifically in my case)

ipetepete, 2021-12-29:
## Current Behavior

JHEMCU SP24S which uses the DIY_2400_RX_ESP8285_SX1280 target fails to bind with either the Binding phrase or manually.
Binding worked on this RX before upgrading to 2.0. I've since updated to 2.1 and the same issue remains.

I also had an issue with the RX not connecting to the home wifi with 2.0 but seems to be resolved in 2.1

## Steps to Reproduce

Flash a DIY target 2.0 and try to bind

## Possible Solution (Not obligatory)

## Details

I'm not sure this is an issue with my specific RX or with the DIY target in general.

## Your Environment

- TX hardware: HappyModel ES24TX 2400 TX (both the slim T-lite and the JR bay module)
- RX hardware: JHEMCU SP24S
- Handset model: Jumper T-lite & Radiomaster Tx16s
- OpenTX version (including nightly number): Edgetx v2.5.0 (on both transmitters)
- ExpressLRS version (TX & RX MUST MATCH): 2.0 (tried 2.1 on rx also)
- Packet Rate: 250
- Telemetry Ratio: 1:128
- user_defines: -

EdgarAllenPoe, 2021-12-29:
TX hardware: HappyModel ES24TX Pro Micro
RX hardware: JHEMCU SP24S
Handset model: Jumper T16
OpenTX version (including nightly number): 2.3.14-otx (1e09791a)
ExpressLRS version (TX & RX MUST MATCH): 2.01 on both
Packet Rate: 500
Telemetry Ratio: 1:128

pkendall64, 2021-12-30:
Can you provide some very closeup images of both sides of the RX please.

ipetepete, 2021-12-30:
@pkendall64 
[image]
[image]
[image]
[image]

kimman1, 2021-12-30:
Looks like it has 2 additional capacitor beside OSC.

EdgarAllenPoe, 2021-12-31:
Here you go.
[image: PXL_20211231_033502678.jpg][image: PXL_20211231_033330251.jpg]

EdgarAllenPoe, 2021-12-31:
[image]
[image]

Zestforlife808, 2022-01-03:
Apologies if this isn't the right place for this.  I can open a separate issue if needed.  

I am having similar binding issues with the JHEMCU EP24S.  Best I can tell, it looks physically the same as the pics above but it has the ipex connector.  I am using the JHEMCU EP24S 2400 RX target flashed over WIFI.  I am able to bind but if the receiver is power cycled it will take a long time to rebind, and when it does, it will often lose the blue light every so many seconds with constant "telemetry lost, telemetry recovered" messages.  Using the LUA script to change telemetry ratios back and forth will sometimes force it to "lock in" and remain bound but I haven't found a way to consistently duplicate the behavior.  

Also, the receiver doesn't seem to be joining my home network despite having been flashed with the credentials.

I am using the same hardware and settings with a Happy Model 2G4 AIO with zero issues.

TX hardware: Namimno 2.4 OLED
RX hardware: JHEMCU EP24S
Handset model: Radiomaster TX16S
Edge TX Version: 2.5
ExpressLRS version (TX & RX MUST MATCH): 2.01 on both
Packet Rate: 500
Telemetry Ratio: 1:128, 1:32, 1:8, 1:2, 1:16

Zestforlife808, 2022-01-06:
I have done some more testing, this time with the DIY_2400_RX_ESP8285_SX1280 target.  It looks like the receiver isn't honoring the last rate used.  I say that because when I get it to connect successfully at say 500 hz 1:8 hz then power cycle the quad, the radio immediately says "telemetry lost" and then the TX module display changes to 500 hz 1:2 and then the blue light starts to flash.

If I go into the LUA script and start fiddling with the telemetry ratio from 1:8 to 1:16, exit the LUA, re-enter the LUA and change from 1:16 to 1:8 and so on and so forth, the receiver will eventually connect with repeating "telemetry lost.. telemetry recovered" messages.  Usually at this point the screen on the TX module says 1:2 no matter what the LUA is set to.  Then I go back into the LUA, set telemetry ratio back to 1:8 and exit the LUA.  I repeat this step until I see the TX module say 1:8 on the screen.  Once the TX module screen telemetry ratio matches what's shown in the LUA, the warnings stop and the blue light remains solid and I can fly without issue.

I am planning to install a FLYWOO ELRS EL24P shortly and if it doesn't have these issues I likely won't be returning to the JHEMCU unless there's a known fix.  If the Flywoo has the same issues, then I'm thinking it might be related to the Namimno TX module and I'll create a separate issue for that.  Though, again, my Namimno module works fine with the Happy Model 2G4 AIO.

ipetepete, 2022-01-06:
@Zestforlife808 it sounds like your issue might be related, but it's different enough that it should likely have it's own bug report. I think most of the details you've posted here might get overlooked piggybacking on this ticket. I would of course bend to whatever the devs feel is appropriate. But I also think that more unique reports there are the better it would be to track down issues. This is just my unsolicited two-penny observation, do with it what you will. Thank you for contributing.

sergjack, 2022-01-07:
Same
Can't bind HM es24tx with sp24s
Tried all versions (1.2,1 , 2.0 , 2.0rc4, 2.0.1) with no luck

vidzo, 2022-01-08:
I'm also in the same position, DIY Slimmer TX will not bind with this receiver, no matter the firmware version. This is two receivers

brandonrc, 2022-01-09:
> I'm also in the same position, DIY Slimmer TX will not bind with this receiver, no matter the firmware version. This is two receivers

Are you running JHEMCU RXs as well?

brandonrc, 2022-01-09:
> Same Can't bind HM es24tx with sp24s Tried all versions (1.2,1 , 2.0 , 2.0rc4, 2.0.1) with no luck

JHEMCU RX's?

jsaginario, 2022-01-09:
Same issue for me. Been through 2 JHEMCU RX's so far. RDQ Support claims JHEMCU has "poor QC"

TX hardware: HappyModel ES24TX Slim Pro
RX hardware: JHEMCU SP24S
Handset model: FrSky X-Lite
OpenTX version (including nightly number): 2.3.14-otx (1e09791a)
ExpressLRS version (TX & RX MUST MATCH): 2.01 on both
Packet Rate: 250
[image]
[image]
[image]
[image]
[image]
[image]

SunjunKim, 2022-01-09:
> > Same Can't bind HM es24tx with sp24s Tried all versions (1.2,1 , 2.0 , 2.0rc4, 2.0.1) with no luck
> 
> JHEMCU RX's?

Probably. No other than JHEMCU has named it sp24s

brandonrc, 2022-01-09:
> > > Same Can't bind HM es24tx with sp24s Tried all versions (1.2,1 , 2.0 , 2.0rc4, 2.0.1) with no luck
> > 
> > 
> > JHEMCU RX's?
> 
> Probably. No other than JHEMCU has named it sp24s

yea I figured that out after doing research on JHEMCU.. did not even know the name before last week. 🤣 came out of no where with hardware.

vidzo, 2022-01-09:
> 

Yes, two SP24S receivers. Neither of them work.

sprrrry, 2022-01-14:
I have 2 SP24S, both are working fine on 2.0.1 fw.
Radio - TX16S with Happymodel ES24TX on 400k baud
Firmware - EdgeTX Nightly and ELRS 2.0.1

sergjack, 2022-01-14:
I suppose there is defect batch.
I bought mine in december of 2021

ipetepete, 2022-01-14:
I don't think that's the case. Mine works fine on 1.x firmware but won't bind on 2.x

Zestforlife808, 2022-01-14:
I don't know if anyone's noticed, but they've already tagged my issue #1260 regarding inconsistent binding with Flywoo and JHEMCU receivers (same board design) as a "hardware issue."  I take that to mean there's no fix coming.  I have no issues with a HM EP1 and a HM 2G4 AIO.

vidzo, 2022-01-15:
That not great. Well that's the last time I but anything from JHEMCU

ipetepete, 2022-01-15:
@pkendall64 or @brandonrc Can we get confirmation from the developers if this is a hardware issue? My rx worked fine on 1.x firmware.

pkendall64, 2022-01-15:
None of us have the hardware so we have no clue! They didn't engage with the developers either.

vidzo, 2022-01-15:
@pkendall64 Is there a developer on the UK that I can send an Rx to?

pkendall64, 2022-01-15:
> @pkendall64 Is there a developer in the UK that I can send an Rx to?

No, no devs in the UK. Mostly Aus, US, Phillipines and a few in other far flung places.

vidzo, 2022-01-15:
> > @pkendall64 Is there a developer in the UK that I can send an Rx to?
> 
> No, no devs in the UK. Mostly Aus, US, Phillipines and a few in other far flung places.

That's a shame. Hopefully there's a fix for the problem.

Dropsleutel, 2022-01-15:
[image]
 
Did you  check and correct the full pcb first for any errors as such ??

jsaginario, 2022-01-15:
I'm happy to send a dev a JHEMCU board if they'd like one.

vidzo, 2022-01-15:
As somebody who's put together a few flight couriers I should have checked....

[image]

rocraz, 2022-01-16:
Hi there. I've got 2 JHEMCU SP24S and one of them is working properly (the right one in the pictures) and the other one is not (the left one in the pictures).
I haven't used v1, only v2 though. I tried the same ".bin" and fresh ".bin" with no success. I tried both SP24S in the same FC, just to make sure the FC and UART were not the cause.
The faulty one allows me to connect to its wifi and flash the firmware, but it won't bind.
I'm in Brazil and I could send the faulty one to any dev in South America.

[image]
[image]

FOG-Yamato, 2022-01-16:
We as devs would like JHEMCU to come in contact with us directly and provide us with schematics and samples directly as such a direct line of communication would allow for these issues to be solved as well as provide support for their hardware.

vidzo, 2022-01-16:
Interestingly, the antenna on the working one is marked 2 while the other is marked 1. Going through the test of the pics in the thread, all other antennas from SP24s that aren't working are also marked one. Not sure if that's something to look at?

rocraz, 2022-01-16:
> I tried the same ".bin" and fresh ".bin" with no success. I tried both SP24S in the same FC, just to make sure the FC and UART were not the cause.
> My TX is Happymodel ELRS24 SlimPro, on a Tango 2 with FreedomTX 1.31.
> The faulty one allows me to connect to its wifi and flash the firmware, but it won't bind.

Vidzo, I hadn't noticed that until you mentioned. Would it mean "1" for V1 and "2" for V2???? I have another TX on my Radiomaster TX16S. I'll downgrade it to v1 and try to bind v1 with this faulty unit and see what happens.

rocraz, 2022-01-16:
> We as devs would like JHEMCU to come in contact with us directly and provide us with schematics and samples directly as such a direct line of communication would allow for these issues to be solved as well as provide support for their hardware.

Yamato, I'll try to talk to Mr. Michael Liu, who I have contacted about this issue, and see what happens. Unfortunately I don't know him, but maybe they'll be convinced by their clients regarding their units.

vidzo, 2022-01-16:
> Vidzo, I hadn't noticed that until you mentioned. Would it mean "1" for V1 and "2" for V2???? I have another TX on my Radiomaster TX16S. I'll downgrade it to v1 and try to bind v1 with this faulty [...]

My post disappeared. That's just me clutching at straws, it probably has no relevance to the issue. I don't know anything about ceramic antennas but unless they differ vastly in design and JHEMCU didn't pick up on it, I'm not sure that would cause the non binding issues. I do find it interesting that it was the only thing that I found different between your two receivers.

Dropsleutel, 2022-01-17:
On the left Rx there are solder bridges between the legs if SX1280, this seems a QC error rather than ELRS. 
Use some flux an drag your iron and fix hardware first.

SunjunKim, 2022-01-17:
> On the left Rx there are solder bridges between the legs if SX1280, this seems a QC error rather than ELRS. Use some flux an drag your iron and fix hardware first.

Those bridged pins are all GNDs. It's probably not the cause of the problem.

vidzo, 2022-01-18:
> 

This is true. In the middle of putting a DIY together and they're ground. Even evident in the screen schematics.

SunjunKim, 2022-01-21:
I got three JHEMCU units, and one arrived defective with the symptoms discussed in this issue. 
I examined this with other ELRS devs, and we concluded that the 52Mhz crystal oscillator used in SP24S and EP24S are defective. The tolerance of the crystal is out of the limit to establish a stable connection. It's because of poor quality control from the vendor.

You can test your RX units' problem is this, by following this procedure:

1. Cooldown (with ice in a plastic bag) or heat (with a hairdryer or some hot) your RX unit. 
2. Observe if a binding is established.
3. After a connection, remove the ice or heat source, and your LQ will drop gradually and finally be disconnected.

If this is your problem, you should ask the seller for a refund or replacement. It's not an issue with the ELRS software, but it's from defective hardware.

Dropsleutel, 2022-01-21:
Are there any loading capacitors connected to the X-Tal?  
If it is to low in Freq, maybe can try to remove them?  

Another option is it possible to finetune the PLL in multimodule some chipset like cc2500 can be centered tuned, to compensate for small tolerances and get best performance.

vidzo, 2022-01-22:
Confirmed this with my SP24S using a heat gun. Wasted so much time rebuilding a DIY transmitter module with replacement parts as I thought that was the problem.

jsaginario, 2022-01-22:
Confirmed as well, also with a heat gun. Great find!

SunjunKim, 2022-01-24:
Closing the issue.
This is a hardware defect, which cannot be resolved by firmware.

closed 2022-01-24 by SunjunKim

sprrrry, 2022-01-25:
Does this mean all receivers have this problem? Is there a way to check the oscillator because I haven't been facing any such issue with my two receivers and I want to be sure.

SunjunKim, 2022-01-25:
> Does this mean all receivers have this problem? Is there a way to check the oscillator because I haven't been facing any such issue with my two receivers and I want to be sure.

I'd propose an "ice&hairdryer" test.
Connect an RX, put it on an ice bag (or dry ice if you're flying in a freakin frozen nature) for about 4-5 min, rest at room temperature for a while, and heat it up with a hairdryer for about a minute. During the procedure, carefully check whether LQ stays good. Repeat the cycle if needed.

If any failsafe or telemetry lost happens, the unit might be unreliable.

vidzo, 2022-01-26:
Has somebody contacted JHEMCU and informed then about this? I'm concerned about people buying this receiver and having a bad experience with it

ipetepete, 2022-01-26:
I did send them a message, but who knows if they'll actually read it or do anything for that matter. It was addressed on Bardwel's FPV News, and I've contacted Pyrodrone (my vendor). That's about all I can do.

rocraz, 2022-01-26:
> Has somebody contacted JHEMCU and informed them about this issue? I'm concerned about people buying this receiver and having a bad experience

  I also sent them a message 2 weeks ago, but got no replies until now. I have also contacted the supplier, but he said it's a problem with ELRS code, even after I said 3 RXs worked fine and only one is giving me this issue.

  For now, I'm gonna try Happymodel and Matek units.

Kevin590, 2022-02-12:
Just an FYI, I am having the same issue trying to bind a newly purchased Flywoo EL24E board. 

I apply heat with a heat gun and it connects, but losing telemetry and then disconnects as it cools off.  I have a handful of Happy Model and BetaFPV boards in quads and never had this issue.

franz4949, 2022-03-04:
I also have binding problem with JHEMCU ELRS-2G4-P 2G4P 2.4G ELRS 5 ch PWM receiver. Flashing with ELRS V2.2.0 was done successfully, binding impossible. I am using a VantacLite Tx module.

rocraz, 2022-03-04:
> I also have binding problem with JHEMCU ELRS-2G4-P 2G4P 2.4G ELRS 5 ch PWM receiver. Flashing with ELRS V2.2.0 was done successfully, binding impossible. I am using a VantacLite Tx module.

It's not a problem with your TX, but with JHEMCU oscilator quality control: it simply doesn't work because the oscilator can't deal properly with the frequency. Unfortunately ...

Kevin590, 2022-03-04:
I have now converted almost all of my cheaper ELRS receivers to the Matek ones. So far they have been flawless, albeit bigger. I'm going to go with Matek from now on.

franz4949, 2022-03-04:
I also meanwhile have ordered two Matek PWM receivers, I did put my first post just to list here also the JHEMCZ PWM-receiver and warn others.