ExpressLRS/ExpressLRS #1230: WiFi Passwords for Hot Spot Mode Not Editable

TomAlperin, 2021-12-29:
While I can enter my own wifi network credentials when at home for connecting receiver and modules to my home network, the wifi network name and password are hard coded into the receiver and module firmware. This could create a situation where two nearby pilots could both have wifi mode enabled and one or both could possibly flash the wrong receiver. This seems most likely in a race or public group flying event. One of the pilots doesn't even need to be powering on their quad specifically to access the wifi network. They could be using USB to adjust other settings and the wifi network will automatically be started.

The password should at least be configurable from the wifi hosted web page and the configurator if not also the name of the wifi network. It should be settable over USB passthrough in case the user forgets the network name.

I would expect this to become even more likely to occur once BetaFlight Configurator access through Wifi is added.

CapnBry, 2021-12-30:
Were you the one asking about this in the discord as well? I came looking to add to that conversation but couldn't remember if it was an existing issue or discord. 😆

This does make sense just because if you're at an event and there's 19 people with their RXes that have gone into wifi mode, you'll never be able to tell which wifi network is yours. I thought about using the target name but there will probably be a lot of overlap there.

What about setting the wifi AP name and password to the home network SSID if the home network can't be found on scan and it goes into AP mode? That way, you know exactly which wifi network is the one you want when at the field, and also the password is set to non-default. Your phone would just connect right to it because it is the same credentials as the home network.

pkendall64, 2021-12-30:
That is a great idea. Always thinking that man 👍

TomAlperin, 2021-12-30:
I am the one who asked about this on Discord. The solution of just using the same credentials sounds like a good solution that wouldn't add complexity to receiver setup. It might be a good idea to put a short note in the configurator and on the web page to let users know this is how it works.

TomAlperin, 2021-12-30:
One other thing you might want to consider is what happens if someone powers up two devices with the same home network. Do not think this is as likely to happen but I would expect the second device to then join the network of the first device.

jurgelenas, 2021-12-30:
> Were you the one asking about this in the discord as well? I came looking to add to that conversation but couldn't remember if it was an existing issue or discord. 😆
> 
> This does make sense just because if you're at an event and there's 19 people with their RXes that have gone into wifi mode, you'll never be able to tell which wifi network is yours. I thought about using the target name but there will probably be a lot of overlap there.
> 
> What about setting the wifi AP name and password to the home network SSID if the home network can't be found on scan and it goes into AP (hotspot) mode? That way, you know exactly which wifi network is the one you want when at the field, and also the password is set to non-default. Your phone would just connect right to it because it is the same credentials as the home network.

What if RX/TX can’t connect to the home Wi-Fi when you are at home and spawns another AP with same credentials?

TomAlperin, 2021-12-30:
> > Were you the one asking about this in the discord as well? I came looking to add to that conversation but couldn't remember if it was an existing issue or discord. 😆
> > This does make sense just because if you're at an event and there's 19 people with their RXes that have gone into wifi mode, you'll never be able to tell which wifi network is yours. I thought about using the target name but there will probably be a lot of overlap there.
> > What about setting the wifi AP name and password to the home network SSID if the home network can't be found on scan and it goes into AP (hotspot) mode? That way, you know exactly which wifi network is the one you want when at the field, and also the password is set to non-default. Your phone would just connect right to it because it is the same credentials as the home network.
> 
> What if RX/TX can’t connect to the home Wi-Fi when you are at home and spawns another AP with same credentials?
> 
> Sometimes TX/RX devices can’t connect to the home Wi-Fi network and this new behavior could be frustrating and hard to debug for the users.

This is a good point. Maybe the network that is spun up could have a name that was a combination of the home networks and the default network to prevent this from being an issue. The name would be unique but not identical to the home network.

CapnBry, 2021-12-31:
Oh yeah that's a good idea. We can just put "TX " or "RX " at the beginning of the home network SSID. That will take care of the issue of having two wifi networks with the same name while in range of the home network, and mitigate some of the in the field issue if someone starts wifi mode on both the TX and RX. I'm also OK with just using "ELRS " as a prefix so there's always only one alternative network name, but I think splitting it to TX / RX allows a little more flexibility.

The only thing it doesn't cover is if someone is at the field and plugs in like multiple RXes in wifi mode at the same time, which I don't think is a scenario we should support. We could fix this as well by using the ESP's flashid to generate a unique AP name, but then you're going to end up with a ton of remembered networks on your phone. And honestly, what are you doing ya wierdo, trying to configurator update all your RX in the field at once?