ExpressLRS/ExpressLRS #1231: Omnibus F4 DS / BetaFPOV 900 Nano Rx on BetaflightPassthrough mode

pierrotm777, 2021-12-30:
Hello,

I try to update my BetaFpv Nano Rx board with the 2.0 release.
The configurator return me these errors.
[code]
Flash error
In file included from .pio\libdeps\BETAFPV_900_RX_via_BetaflightPassthrough\ESPAsyncWebServer-esphome\src\SPIFFSEditor.cpp:1:
.pio\libdeps\BETAFPV_900_RX_via_BetaflightPassthrough\ESPAsyncWebServer-esphome\src\SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
   16 |     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
      |                                                                                                     ^~~~~~
In file included from .pio\libdeps\BETAFPV_900_RX_via_BetaflightPassthrough\ESPAsyncWebServer-esphome\src/ESPAsyncWebServer.h:27,
                 from .pio\libdeps\BETAFPV_900_RX_via_BetaflightPassthrough\ESPAsyncWebServer-esphome\src\SPIFFSEditor.h:3,
                 from .pio\libdeps\BETAFPV_900_RX_via_BetaflightPassthrough\ESPAsyncWebServer-esphome\src\SPIFFSEditor.cpp:1:
C:\Users\pierrot\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:275:15: note: declared here
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));
      |               ^~~~~~
C:\Users\pierrot\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\mmu_iram.cpp:26:34: note: '#pragma message: ICACHE size 16K'
   26 | #pragma message("ICACHE size 16K")
      |                                  ^
Traceback (most recent call last):
  File "C:\Users\pierrot\AppData\Roaming\ExpressLRS Configurator\firmwares\github\ExpressLRS\src/python/BFinitPassthrough.py", line 167, in <module>
    bf_passthrough_init(args.port, args.baud)
  File "C:\Users\pierrot\AppData\Roaming\ExpressLRS Configurator\firmwares\github\ExpressLRS\src/python/BFinitPassthrough.py", line 50, in bf_passthrough_init
    timeout=1, xonxoff=0, rtscts=0)
  File "c:\users\pierrot\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "c:\users\pierrot\.platformio\penv\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "c:\users\pierrot\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "c:\users\pierrot\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Un p�riph�rique attach� au syst�me ne fonctionne pas correctement.', None, 31)
*** [upload] Error 1
========================= [FAILED] Took 27.03 seconds =========================
Environment                               Status    Duration
----------------------------------------  --------  ------------
BETAFPV_900_RX_via_BetaflightPassthrough  FAILED    00:00:27.034
==================== 1 failed, 0 succeeded in 00:00:27.034 ====================[/code]

My SerialRx is on TX6/RX6

Thanks for your help

Pierre

pierrotm777, 2021-12-30:
I have solved my problem.
I have updated my BetaFPV Nano Rx with a FTDI interface.
Very easy.
Just one question now.
Can i use a HappyModel 900 TX 868Mhz with BetaFPV Nano RX 868Mhz ?

jurgelenas, 2021-12-30:
> I have solved my problem.
> I have updated my BetaFPV Nano Rx with a FTDI interface.
> Very easy.
> Just one question now.
> Can i use a HappyModel 900 TX 868Mhz with BetaFPV Nano RX 868Mhz ?

Yes.

closed 2021-12-30 by jurgelenas

pierrotm777, 2021-12-30:
Sorry another question.
I have a BINDIND_PHRASE.
My Tx and Rx are just powered with usb cables.
But receiver blinking quickly !
I think i don't have bind, exact ?