ExpressLRS/ExpressLRS #1232: V2 LUA Issues (NV14)

Aaronsss, 2021-12-30:
## Current Behavior

The elrsV2.lua doesn't seem to work properly. 

Executing the lua direct from the file structure loads up but then doesn't allow any user interaction and the radio has to be rebooted to regain use of the touch screen. Receiver tab in betaflight still shows stick movements so I am guessing the radio hasn't crashed.

Going to model -> setup -> setup (external RF section) does open and works a bit better. It is possible to set the VTX channel and enable wifi etc using this menu but configuration of the TX settings does not work correctly. When you select a TX option to change it displays some strange characters and then when you reload the menu the option has gone to the bottom option in the drop down, it is not possible to change to an option that is not the bottom one. It also stays Saving... up the top but that never goes away until you reload the menu

## Steps to Reproduce

As above

## Possible Solution (Not obligatory)

ELRS.lua did work before upgrading to 2.01. In the old lua version you had to use the trim controls to move around the menu, that functionality seems to be removed in the V2 lua

## Details

Unable to configure ELRS as a result.

EdgeTX may yeild better results but I haven't got around to flashing the radio yet

## Your Environment

- TX hardware: HappyModel ES24TX 2400 TX
- RX hardware: HappyModel EP 2400 RX
- Handset model: Flysky Nirvana
- OpenTX version (including nightly number) Flysky-NV14-2.1.18
- ExpressLRS version (TX & RX MUST MATCH): 2.01
- Packet Rate: 250
- Telemetry Ratio: 1:64
- user_defines:

skorokithakis, 2021-12-31:
I upgraded to 2.0.1 with EdgeTX on the same RX/TX, and the Lua script is stuck on "loading..." in the TX. I don't know if I botched the update or the Lua script doesn't work, but I'm having issues too. Only a red LED lights up solid in the module.

Aaronsss, 2021-12-31:
I have updated to EdgeTX, It has taken all of today to manually add my models to edgeTX (and I still have some left to do) :( but the lua scipt now works as it should

skorokithakis, 2021-12-31:
@Aaronsss why aren't you just automatically converting them? It only took me a few minutes.

Aaronsss, 2021-12-31:
@skorokithakis how did you do it? The update script didn't work for me and the edgetx release notes say this "Settings and models CANNOT be converted from OpenTX or derivatives, you need to start with a clean slate"

skorokithakis, 2021-12-31:
@Aaronsss I think I used my old SD and it just converted the models. They all work with minimal changes (basically some telemetry sensors needed changing). I followed one of the YouTube guides, for example https://www.youtube.com/watch?v=PhpDf-j_3_Q.

hdcrb, 2022-01-17:
converting does not work with some OTX builds for NV14, as it uses a model format which predates OTX 2.4 since the builds bei qba and Stefan73 are based on OTX 2.3

no way around it than creating them from scratch

closed 2022-07-02 by pkendall64