ExpressLRS/ExpressLRS #1233: Improve dynamic power to be more responsive and maintain higher LQ

Naomarik, 2021-12-31:
The current dynamic power code seems to work pretty nicely for LOS flying, but not freestyle when dipping in and out through multiple layers of obstacles. When going down diving down behind multiple layers of concrete walls behind myself, I constantly see LQ as low as 60-70 for awhile, and most my flight can stay in the low 90s.

I've naively hacked a solution that seems to work a lot better for my freestyle use case. This will increase power when moving average LQ drops below 97 and record the moving average RSSI at that time. When the moving RSSI becomes a bit better from the last power increase (hard coded 8dbm) it will decrease power.

I'm not a C++ programmer and this is a very poor implementation that isn't suitable for a PR, but I thought I'd open this issue to have a starting point. I think putting some thought into this is extremely important because ELRS module is pretty power hungry and can drain my TX pretty fast if I just keep it locked to 250/500mw.

I've tested the following at 1:64 telemetry ratio at 250hz and it's very responsive in maintaining high LQ and dropping it down when required: https://github.com/Naomarik/ExpressLRS/commit/66f27ad39e3d73d66e9bafc0d7689454cf7e722c

SunjunKim, 2021-12-31:
I'm planning to review this soon

avsaase, 2021-12-31:
I think the current logic is pretty solid, and what you are describing just shows it is not responsive enough for all use cases. The responsiveness of dynamic power also highly depends on the telemetry rate so you could try increasing that.

With the changes you are proposing I fear power will be unnecessarily high a lot of the time because 97 LQ is still very high during long range flights. For freestyle flights it could make sense to increase the 50 LQ threshold to boost to max power to something like 80 so it would be nice if this is configurable.

TomAlperin, 2022-01-03:
Would being able to configure the minimum power level help with this?

paveltar, 2022-01-07:
> Would being able to configure the minimum power level help with this?

I think this would be a nice feature!

blockfeed, 2022-01-27:
I have had the same experience with dynamic power as @Naomarik flying freestyle (TX16S, 2.1.0, Thor TX, Matek 24-D, 500Hz, 1:64). I have 1-2 places in my usual freestyle practice spot where I hit LQ issues if I stay in place for too long; I will usually hit one of RSSI at -90dB or LQ < 60% (I have audio alerts on the handset). I actually had one user-generated failsafe the other day testing the link durability in one of the spots. It seemed to take too long to increase the power, so generally I'll fly through these spots as quickly as possible to avoid this particular delayed behavior.

JAL513, 2022-02-23:
I have also seen rssi warnings from dynamic power being slow to increase. A minimum power would be nice, or customizable threshold to increase power. I have telemetry at 1/128, I'll bump that up.

SunjunKim, 2022-09-07:
I believe this issue is resolved from SNR-based DP, which bumps up faster (possibly several steps at once) based on the current SNR.

closed 2022-09-07 by SunjunKim