ExpressLRS/ExpressLRS #1241: Lack of error message caused crash

skorokithakis, 2022-01-01:
I upgraded from pre-2 to 2.0.1. I looked at the ELRS settings and set them to whatever I wanted, then backed out.

I went to fly, and during takeoff I switched to autotakeoff mode but the plane switched to manual instead, leading to a crash and the plane breaking in half. Debugging, I saw that the switch mode that I set to Wide was set back to Hybrid. I forgot I have to have no connection to the RX.

The bug is that this silently reverts to "Hybrid" instead of giving a very loud error message saying "disconnect your RX before changing this". I didn't notice the decreased resolution and that led to the crash.

aMax52, 2022-01-02:
Silently? No, as long as the receiver is connected, it is not possible to change this setting, it simply switches back to the preloaded.

skorokithakis, 2022-01-02:
@aMax52 Yes, it silently switches back for me. I select Wide, it stays at Wide for a second and then switches back to Hybrid. However, in the mean time, I've stopped looking at the text and pressed Return, so I missed the update.

deadbytefpv, 2022-01-09:
As per Docs, you CAN'T change switch modes when you're bound/connected to a receiver.. 

https://www.expresslrs.org/2.0/quick-start/transmitters/lua-howto/#switch-mode

skorokithakis, 2022-01-09:
@deadbytefpv Yes, I know.

danxdz, 2022-01-22:
Well thats the case on jumper t8sg, it takes 1s or 2s to change back if binded to rx.. i think its just the radio thats too slow..

pkendall64, 2022-06-10:
In 3.0 an error message is presented if you try to change settings that are not allowed to change while connected to an RX.

skorokithakis, 2022-06-10:
Excellent, thanks @pkendall64!

closed 2022-07-02 by pkendall64