ExpressLRS/ExpressLRS #1244: Easier Switching Between Multiple Receivers

TomAlperin, 2022-01-02:
### Use Case
User has multiple craft and a few transmitters and wants to be able to switch between transmitters while allowing multiple transmitters to be simultaneously in use.
### Current Behavior
To switch between transmitters that can be simultaneously used, a user must flash the receiver with a new bind phrase. Previous versions allowed the model match number to be changed to allow this but the current version still sets up a connection and just prevents the control signal.
### How This Could Be Implemented
If it is not intended that the model match number be used this way, the bind phrase could instead be stored in a way where it could be modified without flashing firmware. The bind phrase could instead be updated using the wifi page. This would reduce the possibility that a firmware flash would fail in the field disabling a craft out in the field.

qubic, 2022-01-02:
Just don't use bind phrase when flashing receiver, it will be into binding mode after power rx 3 times quickly.

TomAlperin, 2022-01-03:
Really like using a bind phrase but will give this a try the next time I flash.

Pelado-Mat, 2022-01-06:
I wish that the bindphrase could be changed in the WIFI webpage of the RX.

Idea: (don't know if its feasible)
The bindphrase could be changed in the luascript. 
If a RX in bound, it could change the bindphrase in both the TX and RX.