ExpressLRS/ExpressLRS #1245: iNav LUA Script with Betaflight FW

deeph46, 2022-01-02:
## Current Behavior

I noticed FUEL bar in iNav LUA Script will give me 0%. Unless, I plug lipo on my whoop first then turn on my TX. This will give me correct reading on FUEL bar in iNav LUA page.

I have re-Calibrated the Power and Battery in Betaflight.
iNav LUA Script : 2.0.1 (https://github.com/iNavFlight/OpenTX-Telemetry-Widget)

## Steps to Reproduce

1. Turn ON TX.

2. Plug lipo on whoop.

3. Open iNav LUA in TX.

4. Note the Fuel is Zero percent sometimes from some percent creeping down to Zero percent.

## Your Environment

- TX hardware: BetaFPV 2.4GHz Nano Module
- RX hardware: HappyModel 2.4GHz RX
- Handset model: TBS Mambo
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): 2.0.1
- Packet Rate: 250
- Telemetry Ratio: 1:16
- iNav LUA Script : 2.0.1 (https://github.com/iNavFlight/OpenTX-Telemetry-Widget)

schugabe, 2022-01-02:
Can you check if the sensors are updated on the telemetry page (in both cases)?

deeph46, 2022-01-02:
I see sensor working fine, always updated. But the sensor name is ''Bat_'', which the percentage for battery.

schugabe, 2022-01-02:
Elrs simply sends the telemetry sensors as received from the FC and since the values are visible I'd say this is not a elrs bug.

brandonrc, 2022-01-04:
I agree with @schugabe that ELRS is just sending telemetry. If those values are off it has to be an issue with the LUA script or INAV telemetry?

deeph46, 2022-01-06:
It iNav LUA that has the bug. I have installed this LUA [version](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/pull/26#issuecomment-1005909794) and it solved the problem.

Thanks all for the help..

schugabe, 2022-01-06:
Thank you for the feedback!

closed 2022-01-06 by schugabe