ExpressLRS/ExpressLRS #1247: ES24TX Pro lost binding to specific receiver

Xedos9er, 2022-01-03:
## Current Behavior
After accesing LUA script and press on other -> Matek RD24 connection got lost. Now I`m not able to regain connection with this module.
Other RX work without problems with this module. Other TX (old HM ES24tx and ES24TX lite) bind without issue to the Matek RX
I tried everything (reflashing RX, TX, schwitch off binding phrase and manual binding), it doesn´t want to bind anymore?!

## Steps to Reproduce

Go into LUA, press on others, acces RX

## Possible Solution (Not obligatory)

No solution

## Details
Is there any way to reset the TX?!

## Your Environment

- TX hardware: ES24 TX Pro
- RX hardware: Matek RD24
- Handset model: Radiomaster TX16s 
- OpenTX version: Edge TX 2.4.0
- ExpressLRS version (TX & RX MUST MATCH): 2.0.1
- Packet Rate: 250hz
- Telemetry Ratio: 1:8
- user_defines: none

Xedos9er, 2022-01-03:
One more information, I managed now to get randomly connection. But I need to bump up the power to very high levels. I get very slow telemetry and dropouts. I can remember that on first connection the dynamic power bumped the power up to max only on the quad with the matek R24D, wich was curious to me. The other TX reduce power to 10mw when connecting to the Matek RX. The affected TX also reduces to lowest level when connecting to other RX as it should.

There´s definately some mess going on with the TX/RX combination. Maybe some issue with the matek diversity?? Since TX and RX is working with other combinations I can say for sure that hardware is working properly

kimman1, 2022-01-04:
> One more information, I managed now to get randomly connection. But I need to bump up the power to very high levels. I get very slow telemetry and dropouts. I can remember that on first connection [...]

Did you use Dynamic Power ?

Xedos9er, 2022-01-04:
I tried with and without dynamic power, it doesn´t change anything. On dynamic it bumps up to highest level. When connection is established I see very low LQ (~30-50%), while dbm is at an adequate level (-38). I also changed every other setting, only changing to 50hz refresh rate helps to get faster a connection (but still pool LQ).
Here all combinations I tried yesterday:
ES24TX pro + R24D = not working
ES24TX pro + EP1 = working
ES24TX pro + EP2 = working
ES24TX + R24 = working
ES24TX + EP1 = working
ES24TX + EP2 = working
ES24TX lite + R24 = working
ES24TX lite + EP1 = working
ES24TX lite + EP2 = working

Xedos9er, 2022-01-06:
Don´t know if its relevant, here´s the compiling log of Matek firmware:
`Processing MATEK_2400_RX_via_WIFI (platform: espressif8266@3.2.0; board: esp8285-8285; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
UID bytes: 118,185,32,82,85,142

build flags: ['-D PLATFORM_ESP8266=1', '-D VTABLES_IN_FLASH=1', '-D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48', '-D CONFIG_TCPIP_LWIP=1', '-D BEARSSL_SSL_BASIC', '-O2', '-DTARGET_RX=1 -Wall -Iinclude', '-DRADIO_2400=1', '-include target/MATEK_2400_RX.h', '-DMY_BINDING_PHRASE="==\\\\\\\\050\\\\\\\\051=="', '-DMY_UID=118,185,32,82,85,142', '-DLOCK_ON_FIRST_CONNECTION', '-DUSE_DIVERSITY', '-DFAST_SYNC', '-DLATEST_COMMIT=98,101,102,102,48,50', '-DLATEST_VERSION=50,46,48,46,49', '-DTARGET_NAME=MATEK_2400_RX']

Building for ESP8266/ESP8285 Platform
AUTO_WIFI_ON_INTERVAL = OFF
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285-8285.html
PLATFORM: Espressif 8266 (3.2.0) > Generic ESP8285 Module
HARDWARE: ESP8266 160MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 68 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <SX1280Driver>
|   |-- <logging>
|   |-- <SPI> 1.0
|-- <CONFIG>
|   |-- <EEPROM>
|   |   |-- <logging>
|   |   |-- <EEPROM> 1.0
|   |   |-- <Wire> 1.0
|   |-- <logging>
|   |-- <OTA>
|   |   |-- <CRSF>
|   |   |   |-- <CRC>
|   |   |   |-- <CrsfProtocol>
|   |   |   |   |-- <CRC>
|   |   |   |-- <MSP>
|   |   |   |   |-- <logging>
|   |   |   |-- <TelemetryProtocol>
|   |   |   |-- <FIFO>
|   |   |   |   |-- <logging>
|   |   |   |-- <helpers>
|   |   |   |-- <logging>
|   |-- <POWERMGNT>
|   |   |-- <DAC>
|   |   |   |-- <Wire> 1.0
|   |   |   |-- <helpers>
|   |   |   |-- <logging>
|   |   |-- <SX1280Driver>
|   |   |   |-- <logging>
|   |   |   |-- <SPI> 1.0
|-- <CRC>
|-- <CRSF>
|   |-- <CRC>
|   |-- <CrsfProtocol>
|   |   |-- <CRC>
|   |-- <MSP>
|   |   |-- <logging>
|   |-- <TelemetryProtocol>
|   |-- <FIFO>
|   |   |-- <logging>
|   |-- <helpers>
|   |-- <logging>
|-- <BUTTON>
|   |-- <DEVICE>
[... 1102 lines cut ...]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\main.cpp.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\libFrameworkArduino.a
Linking .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.elf
Retrieving maximum program size .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.elf
Checking size .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.9% (used 55332 bytes from 327680 bytes)
Flash: [======    ]  56.7% (used 1115345 bytes from 1966080 bytes)
Building .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.bin
esptool.py v3.1
Merged 1 ELF section
========================= [SUCCESS] Took 51.69 seconds =========================

Environment                                    Status    Duration
---------------------------------------------  --------  ------------
HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI  SUCCESS   00:00:51.687
========================= 1 succeeded in 00:00:51.687 =========================`

I also tried a full flash erase via pyflasher and UART connection on both RX and TX, did not help at all.

Xedos9er, 2022-01-06:
I also tried full erase with pyflaher of TX and RX, did not help.
And also the build log from ES24TX Pro:
`Processing HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI (platform: espressif32@3.3.2; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
UID bytes: 118,185,32,82,85,142

build flags: ['-D PLATFORM_ESP32=1', '-D CONFIG_TCPIP_LWIP=1', '-D BEARSSL_SSL_BASIC', '-DTARGET_TX=1 -Wall -Iinclude', '-DRADIO_2400=1', '-include target/HappyModel_ES24TX_Pro_Series_2400_TX.h', '-D VTABLES_IN_FLASH=1', '-O2', '-DMY_BINDING_PHRASE="==\\\\\\\\050\\\\\\\\051=="', '-DMY_UID=118,185,32,82,85,142', '-DTLM_REPORT_INTERVAL_MS="320LU"', '-DFEATURE_OPENTX_SYNC', '-DUART_INVERTED', '-DFAST_SYNC', '-DLATEST_COMMIT=98,101,102,102,48,50', '-DLATEST_VERSION=50,46,48,46,49', '-DTARGET_NAME=HAPPYMODEL_ES24TX_PRO_SERIES_2400_TX']

Building for ESP32 Platform
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 64 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <NeoPixelBus> 2.6.9
|   |-- <SPI> 1.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESP32-BLE-Gamepad> 0.3.4
|   |-- <NimBLE-Arduino> 1.3.1
|-- <SX1280Driver>
|   |-- <logging>
|   |-- <SPI> 1.0
|-- <CONFIG>
|   |-- <EEPROM>
|   |   |-- <logging>
|   |   |-- <EEPROM> 1.0.3
|   |   |-- <Wire> 1.0.1
|   |-- <logging>
|   |-- <OTA>
|   |   |-- <CRSF>
|   |   |   |-- <CRC>
|   |   |   |-- <CrsfProtocol>
|   |   |   |   |-- <CRC>
|   |   |   |-- <MSP>
|   |   |   |   |-- <logging>
|   |   |   |-- <TelemetryProtocol>
|   |   |   |-- <FIFO>
|   |   |   |   |-- <logging>
|   |   |   |-- <helpers>
|   |   |   |-- <logging>
|   |-- <POWERMGNT>
|   |   |-- <DAC>
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <helpers>
|   |   |   |-- <logging>
|   |   |-- <SX1280Driver>
|   |   |   |-- <logging>
|   |   |   |-- <SPI> 1.0
|-- <CRSF>
|   |-- <CRC>
|   |-- <CrsfProtocol>
|   |   |-- <CRC>
|   |-- <MSP>
|   |   |-- <logging>
|   |-- <TelemetryProtocol>
|   |-- <FIFO>
|   |   |-- <logging>
|   |-- <helpers>
[... 640 lines cut ...]
                                       ^
C:\Users\Waldi\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\main.cpp.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\libFrameworkArduino.a
Linking .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.elf
Retrieving maximum program size .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.elf
Checking size .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.9% (used 55332 bytes from 327680 bytes)
Flash: [======    ]  56.7% (used 1115345 bytes from 1966080 bytes)
Building .pio\build\HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI\firmware.bin
esptool.py v3.1
Merged 1 ELF section
========================= [SUCCESS] Took 51.69 seconds =========================

Environment                                    Status    Duration
---------------------------------------------  --------  ------------
HappyModel_ES24TX_Pro_Series_2400_TX_via_WIFI  SUCCESS   00:00:51.687
========================= 1 succeeded in 00:00:51.687 =========================`

Xedos9er, 2022-01-06:
Issue got fixed thanks to Matek Service. I needed to remove these 2 caps:
[image]

So it´s a hardware issue

brandonrc, 2022-01-06:
@maybenikhil maybe we should add this to `known issues with hardware` on `expresslrs.org`

AndreaCCIE, 2022-01-06:
Mine works without any hardware modification

Xedos9er, 2022-01-06:
> Mine works without any hardware modification

Have you checked your R24D if it still has these 2 caps? Are you also using the Happymodel ES24TX **PRO**?

AndreaCCIE, 2022-01-06:
> Have you checked your R24D if it still has these 2 caps? Are you also using the Happymodel ES24TX **PRO**?

Yes to both questions.
I actually have 2 R24D binding without any problem

andlier, 2022-01-06:
Those looks like load caps for the sx1280 crystal oscillator, just removing those is not fixing anything in a proper way, if they were the wrong value to begin with, it might seem like things improve, until it failsafes in a bad place.

Xedos9er, 2022-01-06:
> Those looks like load caps for the sx1280 crystal oscillator, just removing those is not fixing anything in a proper way, if they were the wrong value to begin with, it might seem like things improve, until it failsafes in a bad place.

I'm not an expert in ELRS circuit design, but that still does not explain why this RX is working with any other TX I have despite the ES24TX Pro. 
I've done a testflight today and did not see any issues without those caps. LQ was solid and good dBm values, nothing obvious. If those parts are essential, it's really annoying that matek says, that it's no issue to use the Rx without those parts.

andlier, 2022-01-07:
Its hard to conclude without more info from Matek on why this is a fix. The radiolink depends on both sides having about the same clock frequency from the oscillator, the value of those caps will pull the frequency up or down, typically bigger cap frequency down. It might be that the hm24 pro is a bit too much off in the other direction on frequency, but the mismatch comes within spec with caps removed. Again, this is a bit of speculation.

pkendall64, 2022-01-07:
It is a well known issue to the devs because the SX1280 has built-in 10pF load capacitance and depending on the oscillator used in the circuit there may or may not need to be some very small extra capacitors added to the circuit.
We have had this discussion with multiple manufacturers and some have still added the capacitors. Not saying Matek is one of them though.
The problem is that some manufacturers have not read that part of the SX1280 datasheet and just added the capacitors as per the oscillator datasheet and the values end up being to high and causing the oscillator to be out of spec.

SunjunKim, 2022-01-07:
> I'm not an expert in ELRS circuit design, but that still does not explain why this RX is working with any other TX I have despite the ES24TX Pro. 

It’s a matter of error margin and tolerance. All parts has some errors, like a resister has typically +-1% or 5% depends on its specifications. Crystals also has some margin of errors. TX and RX should have the exactly same frequencies in theory, however in reality they has some high or low frequencies than the reference. Up to some errors they are able to communicate.

With a reference design (no caps), the real values with errors are distributed normally around the reference frequency, however with caps it shits the mean of distributions to down side. Your RX might have the xtal value at the lower end of the distribution (also shifted down by those caps) and TX might have the value at upper end of the distribution. Those frequency difference was not tolerable to communicate with the TX/RX combination, however some other TX unit may have some lower frequencies which was able to make a connection.

MATEKSYS, 2022-01-07:
Whether to use load capacitors and what spec of caps depends entirely on what crystal oscillator is used. We have tested 3x 52MHz, +-10ppm, 8pF crystal oscillators,  without load caps, A:51.99835, error-30ppm. B:51.999983, 0ppm, C:51.000816, error 16ppm.  Usually we will reserve the pad, then match the load capacitance with the selected crystal oscillator.

MATEKSYS, 2022-01-07:
ES24TX pro + R24D = not working
ES24TX pro + EP1 = working
ES24TX pro + EP2 = working
ES24TX + R24 = working
ES24TX + EP1 = working
ES24TX + EP2 = working
ES24TX lite + R24 = working
ES24TX lite + EP1 = working
ES24TX lite + EP2 = working

after we checking out the above tesing, E24D can work with ES24TX/lite,  EP1/EP2 can work with ES24TX pro. so we assume E24D feq shift down, ES24TX/lite & EP1/2 around center.  ES24TX pro shift up.   the difference between E24D and ES24TX pro is out of tolerance. so removing the load caps on E24D will shift the frequency of E24 upward to narrow the gap with ES24TX pro.
If I don't remember wrong. this is the first report of this issue on R24D.  It may be one example occurred because of large errors on crystal oscillators or load caps.

pkendall64, 2022-01-07:
@MATEKSYS  What size load caps did you add to the R24D?

MATEKSYS, 2022-01-07:
> @MATEKSYS What size load caps did you add to the R24D?
1.5pF

pkendall64, 2022-01-07:
> > @MATEKSYS What size load caps did you add to the R24D?
> > 1.5pF
> > Because there are not many options for 52M crystal oscillators, we can only match it according to what oscillators we can source

1.5pF should have been fine as you obviously took into account the 10pF of the SX1280. I guess it's just the tolerances of the different oscillators that different manufacturers have been able to source.

Xedos9er, 2022-01-07:
Thanks for the explanation, now it makes sense to me. Is the ES24TX Pro using load capacitors? For my understanding I could add/exchange some higher value caps, that should bring down the freq of the oscillator?
I ordered another R24D and will also order a Betafpv nano wich also using load caps to verify if it's the module out of spec or the R24D

closed 2022-01-21 by brandonrc