ExpressLRS/ExpressLRS #1248: BetaFPV 900mhz Micro issues with 400k baud rate or higher

brandonrc, 2022-01-04:
## Current Behavior

[image]

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: BetaFPV 900 mhz micro
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

brandonrc, 2022-01-04:
@Carlospepunto could you add some more information about this issue here. 

Trying to keep track of all the `betaFPV` issues.

brandonrc, 2022-01-04:
@HWT662988

Carlospepunto, 2022-01-04:
TX hardware: BetaFPV 900 mhz micro
RX hardware: Frsky R9mm
Handset model: Radiomaster TX16s and Frsky X9D+
OpenTX version (including nightly number): EdgeTX 2.5 and 2.6 RC1
ExpressLRS version (TX & RX MUST MATCH): 2.0.1
Packet Rate: 200 hz
Telemetry Ratio: 1/64 and 1/32
user_defines:
Dynamic power ON with max output power 250 mw

With TX16s at 400.000 max baud rate LUA script doesn't load or loads very slow. At 921 K and above max baud rate the LUA script loads quickly but I have randomly telemetry lost messages at short distances. With ELRS 2.0.0 LUA script works fine with all baud rates but I have low 1RSS issue.
With X9D+ and onebit sampling mode and 400 k max baud rate LUA script works fine but the TX makes a strong buzz sound.
With this version of ELRS seems that the communication between TX module and the transmitter is defective.

bop4hell, 2022-01-04:
Similar issues 

TX hardware : Happymodel ELRS ES900TX Module 
ExpressLRS version : 2.0.1
Handset model: Radiomaster TX16s
EdgeTX : v2.5.0

I flashed my two TX yesturday , the 2.4 Ghz work perfectly, but using the lua script on the 900TX is loading items very sloly.
It was late and not realy investigate, I will try to find and play with different access baud rate.

brandonrc, 2022-01-04:
> Similar issues
>
> TX hardware : Happymodel ELRS ES900TX Module ExpressLRS version : 2.0.1 Handset model: Radiomaster TX16s EdgeTX : v2.5.0
>
> I flashed my two TX yesturday , the 2.4 Ghz work perfectly, [...]

Could you do a test for me? 

Could you flash the wrong firmware on purpose. 

Use the `betaFPV 900mhz tx` not the `betaFPV 900mhz tx micro`

The only difference should be the pinout and the OLED code. 

See if you can use the LUA script. 

@Carlospepunto same with you. See what happens. I had interesting results with my `bad` tx module they sent me.

brandonrc, 2022-01-04:
If you do not feel comfortable doing this then that is ok but you cannot break these modules by flashing firmware. You can always flash it back.

Carlospepunto, 2022-01-04:
With 900tx nano firmware and 400k baud LUA script seems to work fine with the Radiomaster TX16s, at 921k lua works fine but with some packets lost (3/197, 2/198 in the upper right corner, exactly like with the other firmware), at 1,87m and 3,75m everything seems normal but I' can't test with a quad for now.
With the Frsky x9d+ at 400k the transmitter buzzes, nothing with the r9m tx. The fan is always on, and makes noise too but this sound comes from the speaker of the taranis.

bop4hell, 2022-01-04:
> 

> 

Thanks brandonrc for the input, I have flash "betaFPV 900mhz tx" without beter result on the lua script side, and the Wifi part of the board don't boot with this firmware.

I reload the happymodel firmware and now the lua script don't load anything now whatever the baud rate... it was so slow that it was absolutely not usable befor, but it's like that now ^^

Thanks anyway !!

palladin82, 2022-01-05:
Similar problem.
Happymodel ES900TX, 2.0.1 / RM TX16S
EdgeTX 2.5.0/2.6.0 RC2/nightly have a lag in lua at any baudrates.
On EdgeTX 2.4 and OpenTX 2.3.14 all is fine...

palladin82, 2022-01-05:
[2.6 rc2](https://youtu.be/EOrpPxDPX7U)
[2.4.0](https://youtu.be/pbm-345XCms)
same config...

volusiafpv, 2022-01-05:
Hi I have similar issues as posted. No lua anyway If I can help with an input lmk.

brandonrc, 2022-01-06:
> Thanks brandonrc for the input, I have flash "betaFPV 900mhz tx" without beter result on the lua script side, and the Wifi part of the board don't boot with this firmware.
>
> I reload the happymodel [...]

I would ask whoever you got the TX from for a refund.. then buy something else personally. 

You could try and reach out to BetaFPV for a replacement but yea... 

Sorry you are having these issues.

closed 2022-01-06 by brandonrc

bop4hell, 2022-01-07:
Thanks brandonrc, that was my fear from the beginning.
But I wanted to know if I have did some thing wrong when flashing the module.

Thanks to the all team ! and keep up the good work !!!
Happy Fly.

Carlospepunto, 2022-01-07:
Thank you for your time. I reach out Betafpv for a replacement.

brandonrc, 2022-01-09:
> So thank you for your work on this. So with that being said what do you recommend for x9d or 16s?
> […](#)

I recommend Radiomaster TX16s or the Zorro with built in ELRS :)

volusiafpv, 2022-01-09:
Ok what I mean is I have the betafpv 915 tx oled not work ing going to send
back what would be your go to micro module for elrs thank you in advance..

crmitchelmore, 2022-02-06:
Seeing the same thing with Beta fav 2.4ghz. Only 400k max bauds gets telemetry, all other values fail.