ExpressLRS/ExpressLRS #1256: Regression R9slim bootloader fail (v0.5.4)

aquaforum, 2022-01-05:
regression
I reflashed my r9slim+ota receiver on the express rls. The first step was to load it with a [bootloader](https://github.com/ExpressLRS/ExpressLRS/blob/master/src/bootloader/r9slim_plus_ota_elrs_bl.frk) using my transmitter. then connected to the FC and tried to flash it on the express lrs. I failed. see log.

```
Attempting to reboot into bootloader...
[1] retry...
    Bootloader version found : 'v0.5.4'
    Bootloader type found : 'R9SLIM_PLUS'
    Got into bootloader after: 1 attempts
Wait sync... 
```

then I tried to use the previous [bootloader](https://github.com/ExpressLRS/ExpressLRS/blob/4033e29a9e051b1d73865b3b96a17bedbe3b040f/src/bootloader/r9slim_plus_ota_elrs_bl.frk) and everything worked fine for me. 

```
Attempting to reboot into bootloader...
[1] retry...
    Bootloader version found : 'v0.5.3'
    Bootloader type found : 'R9SLIM_PLUS'
    Got into bootloader after: 1 attempts
Wait sync... sync OK

uploading 46596 bytes...
3%
5%
8%
11%
14%
16%
19%
22%
25%
27%
30%
33%
36%
38%
41%
44%
47%
49%
52%
55%
58%
60%
63%
66%
69%
71%
74%
77%
80%
82%
85%
88%
91%
93%
96%
99%
Success!!!!

```

pkendall64, 2022-01-07:
I feel that this was an intermittent issue, just because the only change to the bootloader was the check that it performed to determine if a valid firmware was loaded before trying to run it!
Can you try again a few times to upload, just to verify it was not an intermittent failure please.

aquaforum, 2022-01-07:
resumed bootloader, does not work: 
`=================== FIRMWARE UPLOAD ===================
  Bin file '.pio/build/Frsky_RX_R9SLIMPLUS_OTA_via_BetaflightPassthrough/firmware.bin'
  Port /dev/ttyACM0 @ 420000
  Using CRSF (full duplex)!
======== PASSTHROUGH INIT ========
  Trying to initialize /dev/ttyACM0 @ 420000

Attempting to detect FC UART configuration...
    ** Serial RX config detected: 'serial 1 64 115200 115200 0 115200'
Enabling serial passthrough...
  CMD: 'serialpassthrough 1 420000'
======== PASSTHROUGH DONE ========

Attempting to reboot into bootloader...
[1] retry...
Verified RX target 'FRSKY_RX_R9SLIMPLUS_OTA'    Bootloader version found : 'v0.5.4'
    Bootloader type found : 'R9SLIM_PLUS'
    Got into bootloader after: 1 attempts
Wait sync...[FAILED] Unable to communicate with bootloader...
[FAILED] Unable to communicate with bootloader...

*** [upload] Error -1
========================= [FAILED] Took 25.03 seconds =========================
Environment                                        Status    Duration
-------------------------------------------------  --------  ------------
Frsky_RX_R9SLIMPLUS_OTA_via_BetaflightPassthrough  FAILED    00:00:25.029
==================== 1 failed, 0 succeeded in 00:00:25.029 ====================
`
after downgrading  the bootloader, everything worked again.

AlfitasBoss, 2022-01-20:
For me it's the same like aquaforum. V0.5.4 on the R9 slim + OTA doesn't works exactly like says him. When I  try with v0.5.3 it's works.

pkendall64, 2022-01-23:
I don't really know why this would fail! The only changes were a compilation fix and a check when about to run the ELRS firmware. But you are failing way before that stage. Perhaps we should just revert this for the Slim bootloader.

bill-zb, 2022-01-27:
I can confirm that flashing BL v0.5.3 on my two R9Slim+ OTA helped as well and ELRS v2.1 was successfully flashed afterwards. The bootloader that pkendall64 uploaded in previous comment (file fw.zip, suppose they are also v0.5.4, but recompiled) didn't help, same behaviour as "stock" 0.5.4 BL.

closed 2022-03-25 by JyeSmith

whinds, 2022-08-02:
I am getting this flashing bootloader error in ELRS version 3.0.0 RC1 with one slim+ota receiver on one quad, but oddly did not get it with a different slim+ota on a different quad. It's going to be a paint to unsolder it and reflash via the transmitter to an older v0.5.3 bootloader. Am I understanding correctly that is the solution? I am trying to update a slim+ota from elrs2.5 to elrs3.0 and hitting this issue.

pkendall64, 2022-08-02:
So it seems the slim+ has the same problem and we'll revert the bootloader to the previous version.

whinds, 2022-08-02:
OK, Just confirmed the one that worked was older bootloader v0.5.3 and the one that's not working is v0.5.4
What is the best bootloader plan going forward if I intend to convert more R9slim+ota towards ELRS? Should I stick with v0.5.4 or v0.5.3?

whinds, 2022-08-02:
It's still a little confusing because the person in the beginning did not mention ST-Link flashing, which I thought was required for slim+ota initially. Thats how I did my two slim+ receivers, but they were a couple months apart, so I think the later one got the v0.5.4 bootloader flashed from the taranis. So now I go to update both from ELRS2.5 to 3.0 via the passthrough and we have the issue above where one worked and one didn't. Please make clear what the solution is going to be here because I have more slim+ota to flash and need to know which file to use. I'm also confused because of the first poster failed to mention that they ST-Link flashed the receiver, so can I just downgrade the bootlader and then go back to passthrough method?

pkendall64, 2022-08-02:
Go with 0.5.3, as the bootloader. When 0.5.3 was released whoever did it didn't commit the code changes. So when 0.5.4 was built it does not have the 0.5.3 changes!

pensiljames07-web, 2025-09-20:
buenas tengo un problema no he logrado compilar e instalar en mi receptor frsky R9M SLIM + el lrs, adjunto el registro

  ** Searching flight controllers **
      > FC found from 'COM3'

Detected the following serial ports on this system:
  COM3

=================== FIRMWARE UPLOAD ===================

  Bin file 'firmware.bin'

  Port COM3 @ 420000

  Using CRSF (full duplex)!

======== PASSTHROUGH INIT ========
  Trying to initialize COM3 @ 420000

Attempting to reboot into bootloader...

[1] retry...

Traceback (most recent call last):
  File "runpy.py", line 196, in _run_module_as_main

  File "runpy.py", line 86, in _run_code

  File "C:\Users\PC\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\ff41f67aec3cb8c0be531ff66a9abba3f215285a\firmware\flasher.pyz\__main__.py", line 3, in <module>

    if __name__ == '__main__':
  File "C:\Users\PC\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\ff41f67aec3cb8c0be531ff66a9abba3f215285a\firmware\flasher.pyz\_bootstrap\__init__.py", line 253, in bootstrap

  File "C:\Users\PC\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\ff41f67aec3cb8c0be531ff66a9abba3f215285a\firmware\flasher.pyz\_bootstrap\__init__.py", line 38, in run

  File "C:\Users\PC\.shiv\flasher.pyz_f04292629f842279c0d5a5f202c481fe22dce42182db6b93e33ae77af210a29d\site-packages\binary_configurator.py", line 437, in main

    return binary_flash.upload(options, args)
  File "C:\Users\PC\.shiv\flasher.pyz_f04292629f842279c0d5a5f202c481fe22dce42182db6b93e33ae77af210a29d\site-packages\binary_flash.py", line 182, in upload

    return upload_stm32_uart(args, options)
  File "C:\Users\PC\.shiv\flasher.pyz_f04292629f842279c0d5a5f202c481fe22dce42182db6b93e33ae77af210a29d\site-packages\binary_flash.py", line 52, in upload_stm32_uart

    return UARTupload.uart_upload(args.port, args.file.name, args.baud, target=options.firmware.upper(), accept=args.accept, ignore_incorrect_target=args.force)
  File "C:\Users\PC\.shiv\flasher.pyz_f04292629f842279c0d5a5f202c481fe22dce42182db6b93e33ae77af210a29d\site-packages\UARTupload.py", line 108, in uart_upload
    line = rl.read_line().strip()
  File "C:\Users\PC\.shiv\flasher.pyz_f04292629f842279c0d5a5f202c481fe22dce42182db6b93e33ae77af210a29d\site-packages\SerialHelper.py", line 43, in read_line

    i = max(0, min(2048, self.serial.in_waiting))
  File "C:\Users\PC\.shiv\flasher.pyz_f04292629f842279c0d5a5f202c481fe22dce42182db6b93e33ae77af210a29d\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (OSError(22, 'La operaci�n de E/S se anul� por una salida de subproceso o por una solicitud de aplicaci�n.', None, 995))

PR #1498 (2022-03-25): Revert SLIM+ bootloaders
The bootloader code has had quite some changes after the initial 0.5.3 version was created, but they were never tested against SLIM+ modules as the binaries were not build/uploaded to git!

When the change was made for 0.5.4 to allow a larger range of memory addresses for the application start addresses these other changes made their way into the binaries, which broke the SLIM+. Other subsequent changes to the main firmware meant that we didn't end up needing the larger range of addresses, so reverting the binaries for the SLIM+ will at least get users working out-of-t [... cut]