ExpressLRS/ExpressLRS #1259: ELRS Model switching no connection to module

SilkjamFPV, 2022-01-06:
## Current Behavior
I have several models with ELRS Rx on my TX16S. Model match setting is off. When I switch models I can't connect to the module. I have to power the radio off and back on and then it's working.

## Steps to Reproduce

1.Turn radio on .

2.radio connects to module

3. Change to another model with ELRS Rx and no connection to module

4.power radio off and on and it connects to module
Model match in settings is turned off.
I have the newest Edge TX version on my Radiomaster TX16S.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

CarlovdMeer, 2022-01-07:
Same issue here. See my post on Discord. (Search on Carlo soldering iron)
It seems to happen on my Happymodel TX24. During the switching model in OpenTX, the HM TX24 module will shortly switch off and on, and the HM TX24 brownouts (hangs/crashes…). You will notice a short switch-off-and-on the red LED in the HM TX24. Fix, power cycles the TX or switches to another model, which powers off the external module and switches back to the ELRS model.

Lua script/UART connection fails too. Reproduced on my Jumper T16 and my old QX7. OpenTX 2.3.14.

Two friends tested with their ELRS modules: HMTX24 failed (RM16S) ; DIY 24 works normal (QX7)
Reproducible on ELRS 1.x and 2.x. Features on 2.x, like the model match, stresses the brownout issue.
I never noticed brownouts on switching models on my (external) MPM and Crossfire modules.

I would expect a watchdog timer that resets the module after 100-500 ms. Is there a watchdog timer active anyway?

SilkjamFPV, 2022-01-08:
Thank you Carlo. I will try to find you on discord.

nmare99, 2022-03-13:
Same issue. Model match setting is off. After switching model in EdgeTX, lua script  connection fails to TX module. It's necessary to turn off and turn on the radio to make the module work again
TX hardware: Happymodel ES24 TX 2.4g
RX hardware: Happymodel EP 1 and EP 2 2.4 ghz
Handset model: Taranis QX7 (with cross mod)
Edge TX 2.6.0
TX & RX ExpressLRS version: 2.0.1

SilkjamFPV, 2022-03-13:
My issue was never fixed lol. I'm waiting a little longer to upgrade and
see what happens. For the time being I keep selecting the model i want,
shut the radio off and power it back on and it's fine. Luckily I can use
one model for most of my quads but it does suck!
Lemme know if you ever find a solution.
-Silke

driedger, 2022-03-17:
I have been having this issue since May 2021 with my Happymodel ES24TX 2.4g and a QX7. I recently upgraded to a Radiomaster TX16S, and am now experencing the same problem with this radio as well. QX7 was on OpenTX 2.3.12, TX16S is on EdgeTX 2.5. Had this probem with ES24TX from pre 1.0, through to latest version, and always just powered radio off and on when switching models.
I now have quite a few models on ELRS and this is becoming more cumbersome, so Im also following along here to see if someone finds a solution.

pkendall64, 2022-03-23:
As I mentioned earlier, when you switch models in OpenTX/EdgeTX the handset power cycles the module. If the power off is not long enough it looks like it's causing the module to crash which is why the LUA is failing and you have to power cycle the radio. I have an ES24TX Pro Series and this problem does not happen on that module.
I think this would be best raised as a ticket against EdgeTX as they are very responsive.

closed 2022-04-05 by github-actions[bot]

SilkjamFPV, 2022-10-11:
Got you! I'm still doing the same thing. No solution....yet