ExpressLRS/ExpressLRS #1260: Namimno OLED TX module inconsistent binding

Zestforlife808, 2022-01-07:
## Current Behavior

Namimno OLED TX module does not bind consistently with JHEMCU EP24S or Flywoo EL24P.  I am able to get it to bind by messing with the LUA script but I don't feel that's intended behavior.  TX module seems to work fine with HM 2G4 AIO.

## Steps to Reproduce
I made a video showing the issue.  See timestamps for context.
[https://youtu.be/sxolYfHlH8M](url)
:12 - Model was previously bound at 500 hz 1:8.  Turning on TX shows module in 500 hz 1:8 25mw.
:20 - Plug in quad.  Immediate "telemetry lost".  Module goes into full power 500 hz 1:2 1000mw.  RX light blinking  (it will not bind if left in this state)
1:05 - Enter LUA script.  Currently displaying 500 hz 1:8 but the module says 500 hz 1:2.  Change telemetry ratio to 1:2 via LUA script and exit.  
1:32 - "Telemetry lost.  Telemetry recovered." repeatedly.  Module screen displaying 500 hz 1:2 25 mw.
1:44 - Re-enter LUA script.  Change back to 1:8 telemetry ratio.
1:59 - TX module now says 500 hz 1:8 25 mw.  No more telemetry lost messages.  Solid blue light on RX.

## Possible Solution (Not obligatory)
I don't know why this is happening but there seems to be a janky workaround as seen above.

## Details
The behavior is nearly the same for both JHEMCU and Flywoo receivers.  I detailed my experience with the JHEMCU in #1229.  I tried both the DIY target and JHEMCU EP24S target with the same symptoms.  For the Flywoo, I bricked it via WIFI flash using the EL24P target and had to revive it with bootloader + passthrough using the DIY 2400 RX ESP8285 SX1280 target.  You are seeing the DIY target in the video.

Flywoo RX flash output.  There is a warning about the crystal in there.  Not sure if relevant.  I also saw "Cannot detect RX target, blindly flashing!" on the initial WIFI flash and the subsequent passthrough flashes. 

```
ExpressLrs Configurator
ExpressLRS Configurator v1.3.2
Build
Verifying build system
Downloading firmware
Building user_defines.txt
Flashing device
Logs
Processing DIY_2400_RX_ESP8285_SX1280_via_BetaflightPassthrough (platform: espressif8266@3.2.0; board: esp8285-8285; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
UID bytes: 147,162,92,163,87,43

build flags: ['-D PLATFORM_ESP8266=1', '-D VTABLES_IN_FLASH=1', '-D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48', '-D CONFIG_TCPIP_LWIP=1', '-D BEARSSL_SSL_BASIC', '-O2', '-DTARGET_RX=1 -Wall -Iinclude', '-DRADIO_2400=1', '-include target/DIY_2400_RX_ESP8285_SX1280.h', '-DMY_BINDING_PHRASE="flyeast2"', '-DMY_UID=147,162,92,163,87,43', '-DLOCK_ON_FIRST_CONNECTION', '-DAUTO_WIFI_ON_INTERVAL="20"', '-DHOME_WIFI_SSID="seawest"', '-DHOME_WIFI_PASSWORD="flyeast2"', '-DFAST_SYNC', '-DLATEST_COMMIT=98,101,102,102,48,50', '-DLATEST_VERSION=50,46,48,46,49', '-DTARGET_NAME=DIY_2400_RX_ESP8285_SX1280']

Building for ESP8266/ESP8285 Platform
AUTO_WIFI_ON_INTERVAL = ON
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285-8285.html
PLATFORM: Espressif 8266 (3.2.0) > Generic ESP8285 Module
HARDWARE: ESP8266 160MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 68 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <SX1280Driver>
|   |-- <logging>
|   |-- <SPI> 1.0
|-- <CONFIG>
|   |-- <EEPROM>
|   |   |-- <logging>
|   |   |-- <EEPROM> 1.0
|   |   |-- <Wire> 1.0
|   |-- <logging>
|   |-- <OTA>
|   |   |-- <CRSF>
[... 373 lines cut ...]
Hash of data verified.

Leaving...
Soft resetting...
========================= [SUCCESS] Took 32.44 seconds =========================

Environment                                           Status    Duration
----------------------------------------------------  --------  ------------
DIY_2400_RX_ESP8285_SX1280_via_BetaflightPassthrough  SUCCESS   00:00:32.437
========================= 1 succeeded in 00:00:32.437 =========================
Result
Success!
Actions
```

## Your Environment

- TX hardware: Namimno OLED TX 
- RX hardware:  JHEMCU EP24S or Flywoo EL24P
- Handset model: Radiomaster TX16s MAX
- EdgeTX version : 2.5
- ExpressLRS version (TX & RX MUST MATCH): 2.01 for both
- Packet Rate: 500 hz
- Telemetry Ratio: 1:2, 1:8, 1:16
- user_defines: N/A

Xedos9er, 2022-01-07:
Check out my thread, to me it sounds like the same issue:
https://github.com/ExpressLRS/ExpressLRS/issues/1247

Zestforlife808, 2022-01-07:
@Xedos9er I've been watching your issue with great interest.  I hope this is something that can be resolved with software because, at least in my case, there's clearly a way to establish a bind.  I have a hot air station and I can remove caps to test, but I think even that's a bit much to ask with these component sizes.

SunjunKim, 2022-01-10:
Maybe related to #1229?
Did you try with other RXes?

Zestforlife808, 2022-01-10:
JHEMCU EP24S and Flywoo EL24P behave the same.  I don't have anything else besides a Happy Model 2G4 AIO and that's working fine.  I have a Happy Model EP1 on the way and will update if that's any better.

SunjunKim, 2022-01-10:
> 

From the image searched from internet, JHEMCU EP24S and Flywoo EL24P have just rebranded the same board apparently.
That board seems particularly problematic regarding the recent issue reports.

maybenikhil, 2022-01-13:
We are working on a ExpressLRS Configurator & HW Errata List

pkendall64, 2022-01-22:
Can we close this as this is not a Namimno issue and there is an issue for JHEMCU RX.

AlessandroAU, 2022-01-22:
Agreed

closed 2022-01-22 by AlessandroAU