ExpressLRS/ExpressLRS #1262: Bad packets reported by Thor TX, perhaps related to max bauds/serial speed?

blockfeed, 2022-01-07:
## Current Behavior

The Thor TX module reports inconsistent bad:good packet values in the lua status bar at boot. The two numbers always add up to the packet rate Hz value, but it never seems to settle at 0/500, even after a few minutes. I have both a HM ES24TX and DIY E28 2.4G module that both lock into 0/500 quickly and the value stays locked, regardless of radio max bauds. I am concerned that this error rate may be a contributing factor to or a symptom of another issue I encountered with the Thor TX.

## Steps to Reproduce

1. Power on TX

2. Check status bar of lua script for bad:good packet values. The good/bad packet values in the lua header will fluctuate where error packets is >0 consistently and the value never settles back to zero.

## Possible Solution (Not obligatory)

A fix that seems to resolve the inconsistent values is to re-set the "max bauds" value to 3.5M (even if already supposedly running at that baud rate). Once the value is set manually after power-on, the lua reports 0/500 as expected.

This fix does not appear to work for any baud rate lower than 3.5M. Any any lower rate, the good/bad packet header continues to fluctuate, reporting errors repeatedly.

## Details

While the TX module does bind and flight appears to be mostly normal, it is concerning that the bad:good value never settles to a sane value like on other modules, and does in fact settle when the serial rate on the radio is re-configured manually (if even to the same 3.5M value).

While my initial suspicion was that this was related to EdgeTX, the fact that it only affects the Thor TX Pro suggests it is specific to the module.

## Your Environment

- TX hardware: Thor TX Pro running master (1/7/2021)
- RX hardware: Matek ELRS 24-D, also occurs with HM EP1
- Handset model: TX16S 
- OpenTX version (including nightly number): EdgeTX 2.5, also occurs on 2.6RC2
- ExpressLRS version (TX & RX MUST MATCH): v.2.0.1
- Packet Rate: 500Hz
- Telemetry Ratio: 1:64
- user_defines: NA

Dropsleutel, 2022-01-07:
it sounds like similar issue as the R9M  (were the rootcause never really became clear) other than problem appears when getting hot.

brandonrc, 2022-01-09:
Yea, I am seeing this issue on my module as well... :/

brandonrc, 2022-01-09:
This issue also happens on the betaFPV modules as well.. so not only axis

brandonrc, 2022-01-09:
@AlessandroAU 
@CapnBry

brandonrc, 2022-01-10:
@pkendall64 Awesome work!

closed 2022-01-10 by brandonrc

blockfeed, 2022-01-10:
Confirming this resolved the issue, thank you all! @pkendall64 @brandonrc

pkendall64, 2022-01-10:
> Confirming this resolved the issue, thank you all! @pkendall64 @brandonrc

You are very welcome 😄

AlessandroAU, 2022-01-11:
Works on my Axis Thor