ExpressLRS/ExpressLRS #1263: Save update rate & telemetry ratio per model ID

tenten8401, 2022-01-09:
As the title says, it would be cool to save update rate & telemetry ratio behind a model ID (same one used for model match). This would allow for automatic switch to 50hz (or whatever) for long range builds and then 500hz with low tlm for racing.

Alternatively, it could be saved on the RX itself but I could see that being much harder to implement.

brandonrc, 2022-01-09:
https://github.com/ExpressLRS/ExpressLRS/issues/1205

Wonder if these issues need to be merged into one. 

### Should we save the following  per model ID.
- power
- fan
- rate
- telem
- VTX Administrator

tenten8401, 2022-01-09:
I think rate & telem alone make the most sense, might be nice to have vtx sync too. Configurable maybe?

I don't really see the use for power or fan per-model once dynamic power gets a bit more stable.

brandonrc, 2022-01-09:
But what about power.. that is a long range/race difference right?

brandonrc, 2022-01-09:
You cannot use dynamic power at race events.. only low 10mw or whatever.

tenten8401, 2022-01-09:
Once dynamic power is stabilized it won't matter, appropriate power can be used automatically for everything.

If you're at a race you should have it locked at 10mw regardless of model.

brandonrc, 2022-01-09:
Or your race rigs could have a model for locked 10mw power. With no dynamic power.

tenten8401, 2022-01-09:
I mean I'm not necessarily opposed to it and I think more choice is always better, I'm just not sure I'd ever use it and don't think many others would either

brandonrc, 2022-01-09:
> Once dynamic power is stabilized it won't matter, appropriate power can be used automatically for everything.
> 
> If you're at a race you should have it locked at 10mw regardless of model.

so they would have to change the settings of the power regardless of model... so why would they just not change the settings for packet rate regardless of model. 50hz for lr and 500hz for race like you stated originally?

tenten8401, 2022-01-09:
They would leave power on max for the module and then dynamic on everywhere except restricted 10mw, that way it only uses what power is actually needed for the situation. This being said, I haven't been real impressed with dynamic power but I'm sure the algorithm needs a bit of tuning.

The telemetry rate needed and hz will vary greatly between aircraft, of which you can fly many in one trip. This can be planes, freestyle quads, long range quads and anything else. This is why I think it should be saved per-model and switch automatically. If you're at a race, you're going to be stuck on 10mw regardless of what model you choose, and accidentally switching models to a different quad (I know many run with one model for all) could potentially be a safety issue if the power were to switch. This is why I think it should be independent of models.

pkendall64, 2022-01-09:
These are already saved per model!
https://www.expresslrs.org/2.0/software/model-config-match/

Setting | Description
-- | --
Packet Rate | The RC update frequency (500Hz, 250Hz, etc)
Telem Ratio | Telemetry ratio (Off, 1:128, 1:64, etc)
Switch Mode | Method for sending switches to the receiver
Model Match | Enable the model match feature (see below)
Max Power | Transmitter output power level
Dynamic Power | Enable Dynamic Power switching

tenten8401, 2022-01-09:
derp, thanks :)

even read the page like 4x before posting this.. just blind apparently

closed 2022-01-09 by tenten8401