ExpressLRS/ExpressLRS #1272: GoPro Camera Control

LKAgrifter, 2022-01-12:
Low Energy Bluetooth Control For GoPros. Ideally done using BLE on the RX. Use case examples: Allow long range pilots to save the Gopro's battery and recording capacity until they reach a point of interest, in which case they can start recording with a switch. Another example is to start recording on Arm for Freestyle Pilots and stop n seconds after disarm.

For freestyle pilots, implementation in the TX module would also be feasible as I'd assume most people don't take off from farther than BLE range of themselves.

Functions Could be as simple as on/off and start/stop recording, or more advanced with different modes and even protune settings configurable in the LUA script. Would need to be easily configurable to a switch.

Support for all GoPros would be ideal, but support for the Hero 5 and later should suffice.

It would presumably require paring to the GoPro, this could be done through the LUA. Perhaps if implementation is successful, a later feature could be added to select between different cameras in the LUA, maybe even configuring it to correlate with the model ID in EdgeTX. An example use case is if I fly a session on my freestyle rig and a Hero 10 on a cinewhoop, ELRS would intelligently switch to the correct camera and not power up the wrong one.

Another potential feature is custom telemetry with camera status, so when you flip the record switch, the TX could have verbal confirmation that the camera did in fact start recording.

As a side note, no idea if an implementation of this could be carried over to cameras such as the DJI Osmo Action 2, but I suppose including non-GoPro branded cameras as well would be a neat feature.

*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

newfla, 2022-01-16:
maybe the [OpenGopro repo](https://github.com/gopro/OpenGoPro) can be helpful :)

harry1453, 2022-01-17:
I have no idea if this has been tested, but would having the BLE radio on the RX transmitting right next to the RX antenna potentially swamp the SX1280? Obviously this isn't a problem using WiFi on the ground but I could see this hurting range when you're actually flying.

pkendall64, 2022-01-17:
The current RXs do not have BLE