ExpressLRS/ExpressLRS #1282: Upload via STLink to "FrSky R9 Slim +" RX always ends with "addr too high" error

fiechr, 2022-01-15:
I'm trying to compile and upload firmware 2.0.1 to a FrSky R9 Slim Plus receiver via STLink programmer (the programmer works and has the latest firmware installed).

I always get the "addr too high" error. It doesn't matter which target I chose (not sure exactly if I need to choose `Frsky_RX_R9SLIMPLUS_via_STLINK` or `Frsky_RX_R9SLIMPLUS_OTA_via_STLINK`. But even R9 without "Plus" results in the same error.

On the PCB of the receiver it says "R9 Slim_V3". On the paper wrap around the receiver it says "R9 Slim +".

I've tried with the Configurator and using VSCode with PlatformIO - same error. The file size seems fine to me.  No idea why it doesn't work...

````
Processing Frsky_RX_R9SLIMPLUS_via_STLINK (platform: ststm32@8.0.0; board: bluepill_f103c8; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
UID bytes: 89,103,195,252,99,205
***Building for STM32 Platform
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 (8.0.0) > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 55 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <extEEPROM> 3.4.1
|   |-- <Wire> 1.0
|-- <SX127xDriver>
|   |-- <logging>
|   |-- <SPI> 1.0
|-- <CRSF>
|   |-- <CRC>
|   |-- <CrsfProtocol>
|   |   |-- <CRC>
|   |-- <MSP>
|   |   |-- <logging>
|   |-- <TelemetryProtocol>
|   |-- <FIFO>
|   |   |-- <logging>
|   |-- <helpers>
|   |-- <logging>
|-- <FHSS>
|   |-- <SX127xDriver>
|   |   |-- <logging>
|   |   |-- <SPI> 1.0
|   |-- <logging>
|-- <LQCALC>
|-- <OTA>
|   |-- <CRSF>
|   |   |-- <CRC>
|   |   |-- <CrsfProtocol>
|   |   |   |-- <CRC>
|   |   |-- <MSP>
|   |   |   |-- <logging>
|   |   |-- <TelemetryProtocol>
|   |   |-- <FIFO>
|   |   |   |-- <logging>
|   |   |-- <helpers>
|   |   |-- <logging>
|-- <PFD>
|-- <POWERMGNT>
|   |-- <DAC>
|   |   |-- <Wire> 1.0
|   |   |-- <helpers>
|   |   |-- <logging>
[... 197 lines cut ...]
  0/6 pages written
  1/6 pages written
  2/6 pages written
  3/6 pages written
  4/6 pages written
2022-01-15T17:18:18 INFO src/common.c: Starting verification of write complete
  5/6 pages written
2022-01-15T17:18:18 INFO src/common.c: Flash written and verified! jolly good!
  6/6 pages written
Cmd: /home/cf/.platformio/packages/tool-stm32duino/stlink/st-flash --reset write .pio/build/Frsky_RX_R9SLIMPLUS_via_STLINK/firmware.bin 0x8008000
/home/cf/.platformio/packages/tool-stm32duino/stlink/st-flash --reset write .pio/build/Frsky_RX_R9SLIMPLUS_via_STLINK/firmware.bin 0x8008000
2022-01-15T17:18:18 INFO src/common.c: Loading device parameters....
2022-01-15T17:18:18 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2022-01-15T17:18:18 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2022-01-15T17:18:18 INFO src/common.c: Attempting to write 44788 (0xaef4) bytes to stm32 address: 134250496 (0x8008000)
2022-01-15T17:18:18 ERROR src/common.c: addr too high
stlink_fwrite_flash() == -1
*** Error 255
*** [upload] Error 255
========================== [FAILED] Took 3.10 seconds ==========================
Environment                     Status    Duration
------------------------------  --------  ------------
Frsky_RX_R9SLIMPLUS_via_STLINK  FAILED    00:00:03.105
==================== 1 failed, 0 succeeded in 00:00:03.105 ====================
````

fiechr, 2022-01-17:
Still not sure about that error message, but I got it working now.

After flashing via STLink (with error) I connected the R9 Slim+ to a FC and flashed again via serial pass through. This worked without error.

cruwaller, 2022-01-23:
Hmm, there might be an issue in flash size. I will check this...

cruwaller, 2022-01-23:
oh, that MCU has only 64kB of FLASH and ELRS would require 44448 bytes (almost 44.5kB). 
Bootloader takes 32kB (can be decreased but effects to many existing users) so there are only 32kB available which is not enough!

Unfortunately there seems to be multiple versions of the same same MCU with the different flash sizes (128k vs 64k).

fiechr, 2022-01-23:
Are there different types of these receivers? It seems to work fine now. I've just flashed a new build using BF pass-through. The RX is connecting to the TX (R9M 2018/ACCST) and controls work fine.

[image]
[image]
[image]

closed 2022-06-25 by github-actions[bot]