ExpressLRS/ExpressLRS #1283: jitter at high power with R9M

dataema92, 2022-01-15:
from 250mw and up there is more and more noise that generates a lot of jitter on gimbal channel.

- TX hardware: R9M 2019
- RX hardware: R9MM
- ExpressLRS version (TX & RX MUST MATCH): 2.0.1 (tried also other versions with the same result)

Sekilsgs2, 2022-01-20:
> from 250mw and up there is more and more noise that generates a lot of jitter on gimbal channel.
> 
> * TX hardware: R9M 2019
> * RX hardware: R9MM
> * ExpressLRS version (TX & RX MUST MATCH): 2.0.1 (tried also other versions with the same result)

Why you think that this is ELRS problem?? Noise generate your transmitter (taranis, jumper , radiomaster ...) - high power affects the readings of the gimbal sensors - because of cheap equipment - what transmitter you using?

dataema92, 2022-01-20:
Thank you. My transmitter is a taranis qx7. If I revert back to the original R9 firmware all is correct. As soon as I flash ELRS the noise start on all channels in specific in channel 4. Can you help me?

Sekilsgs2, 2022-01-20:
> Thank you. My transmitter is a taranis qx7. If I revert back to the original R9 firmware all is correct. 

This is strange - i'm using r9m 2019 and tbs mambo - dont have this problems...

brandonrc, 2022-01-21:
@dataema92 could you try and swap antennas? See if the jitter changes?

brandonrc, 2022-01-21:
I am removing bug label for now, if you can test with a different antenna please let me know the results. 

Thanks.

AlessandroAU, 2022-01-21:
I'm 99% sure this is RF interference from a poorly balance antenna. Open up the radio, unplug the gimbal wires and put a few twists the cable assembly and reassemble.

dataema92, 2022-01-21:
unfortunately i don't have a second antenna for the transmitter to try. if i revert back to stock R9 firmware the problem vanish. if it is a problem with the antenna, why on the stock firmware the problem is not present?
Thank you very much to all.

WinD52, 2022-02-03:
I can confirm this issue.  I have the same problem on my R9M and Taranis QX7 Access.  The higher the R9M power, the greater the amplitude of the control bounces.  At a power of 1W, the bounce is + - 30us.  According to your advice, instead of the super8 antenna from frsky, I tried the antenna from happymodel es900, but nothing has changed dramatically.  I discovered this problem a long time ago, it was present on version 1.0 of ELRS.  Sorry for my English.

WinD52, 2022-02-03:
The shielding of the high-frequency module board helped me a little.  I glued the inside of the module with aluminum tape and connected it to ground.  This helped reduce the bounce to 15-20us at 1W, and 5-10us at 250mW.  In Inav, I raised the insensitive threshold to these values.

dataema92, 2022-02-03:
I also tried to put a few more twist on cables without succes.

closed 2022-06-25 by github-actions[bot]