ExpressLRS/ExpressLRS #1306: No connection on 2.1

nightwolfix, 2022-01-22:
I am having issues with HMES24TX and EP receivers when using 2.1 on all of them:
-2.1 TX does not connect with 2.1 RX. RX light blinks slowly but does not go into wifi mode when TX is on. Goes into wifi when TX is off
-2.1 TX binds with 2.0 RX no issues

AlessandroAU, 2022-01-22:
Are you saying they work on 2.0 and not 2.1?

pitts-mo, 2022-01-22:
Hi @nightwolfix 
My first though is the partially bound / model match behavior. Have you tried cycling the model match ON OFF state from LUA? I think the LUA may also provides some status indication that this is occurring.

kongzheng333, 2022-01-22:
Uncheck HOME_WIFI options then have a try.

nightwolfix, 2022-01-22:
> Uncheck HOME_WIFI options then have a try.

Home wifi has always been off

nightwolfix, 2022-01-22:
> Hi @nightwolfix My first thought is the partially bound / model match behavior. Have you tried cycling the model match ON OFF state from LUA? I think the LUA may also provide some status indication that this is occurring.

Tried to cycle on or off and also deactivating model match, still blinking slowly

Witchdoctor2015, 2022-01-23:
Had a similar issue with the happy model ep2 rx.  In 2.1.0 I could flash the rx and it would enter bind mode when a tx was powered on but it would never bind.  going to 2.0.1 allowed the rx to bind.

closed 2022-06-23 by github-actions[bot]