ExpressLRS/ExpressLRS #1307: Home wifi is broken when the SSID contains characters that need escaping in JSON

wucke13, 2022-01-22:
## Current Behavior

I do happen to have a SSID at home which contains an `\` in it. After setting this network as home net, the update webpage on my EP1 RX doesn't work anymore, as the browser fails to parse the response from `mode.json`. The response looks like this:

```
{"ssid":"My SSID with a \","mode":"AP","modelid":255}
``` 

Unfortunately the hole UI element is not displayed, and even re-flashing the firmware does not remove the set WiFi (which is nice in general, just not so nice in this case).

While it is debatable if my particular network name is nice (it did uncover quite some bugs in some WPA clients :), I believe the underlying problem here is more general. According to [this SO](https://stackoverflow.com/questions/4919889/is-there-a-standard-that-defines-what-is-a-valid-ssid-and-password) an SSID can be 0 - 32 octets (as in bytes). Thus I'd argue it's semantically wrong to model it just as a string internally. I would suggest that we treat this as an array of bytes in the JSON. Of course it makes sense to use a text input for the SSID, after all in my setup connecting to the WiFi with the `\` in its name does work, it's just so that I can never change to another home network since the UI is not rendered due tot he malformed JSON response. 

## Steps to Reproduce

1. Set a home network with a `\` in the SSID

## Possible Solution (Not obligatory)

Escape data correctly before it is put in JSON.

## Details

HappyModel EP1 receiver, version 2.1.0

piwi3910, 2022-01-23:
your wifi ssid is not RFC compliant, 

[Link](https://www.cisco.com/assets/sol/sb/WAP321_Emulators/WAP321_Emulator_v1.0.0.3/help/Wireless05.html#:~:text=The%20SSID%20can%20be%20any,%5C%2C%20%5D%2C%20and%20%2B.)

as such i don't see why it would need to be supported as the \ is not allowed to be in a ssid

piwi3910, 2022-01-23:
i need to correct my above statement, 
the latest RFC does allow the character and even unprintable chars
[link](http://standards.ieee.org/getieee802/download/802.11-2012.pdf)

According to last standard 802.11-2012 (Section 6.3.11.2.2), it can be 0-32 octets with an unspecified or UTF8 encoding.

commanderguy3001, 2022-01-23:
a possible solution to this problem could be to encode the ssid/password as a base64 string and storing that in the json instead of the raw data

commanderguy3001, 2022-01-23:
btw, @wucke13 you should be able to flash the receiver with pure blank data, that should also remove any stored password, then you can flash it again with elrs to have a fully reset and working receiver again, might wanna hop on the discord if you need help with that.

wucke13, 2022-01-23:
@commanderguy3001 How would I flash it back? Since it's an EP I thought it's only flash-able via WiFi?

piwi3910, 2022-01-23:
> [EXTERNAL EMAIL]
>
> @commanderguy3001 [github.com]<https://urldefense.com/v3/__https://github.com/commanderguy3001__;!!LpKI!1O0-tRkCZOSqUQwudtV5QSLF_Dd-vH-Gb0rzovurKvvgez4Brvh7lCor72jBMVctyg$> How would I flash it back? Since it's an EP I thought it's only flash-able via WiFi?
>
>
> Internal Use - Confidential

Betaflight pass through?

commanderguy3001, 2022-01-23:
@wucke13 you can flash it either via betaflight passthrough or via a ftdi adapter like this one from amazon:
https://www.amazon.com/HiLetgo-FT232RL-Converter-Adapter-Breakout/dp/B00IJXZQ7C/ref=sr_1_3

closed 2022-04-05 by github-actions[bot]