ExpressLRS/ExpressLRS #1309: Power switches above 24mW while Disarmed and using dynamic power setting

TomAlperin, 2022-01-22:
## Current Behavior

When using the dynamic power setting, the module will sometimes switch to higher power levels when the receiver connects even though the arm channel has not been set to high. 

## Steps to Reproduce

https://drive.google.com/file/d/14xzfI4qduGrUxfsU8PjUFtpLjsj7nECT/view?usp=drivesdk

1.Power on updated transmitter (tested on Namimno Flash OLED).

2. Power on receiver while observing power level. 

3. Remove power from receiver. 

4. Repeat this a few time and observe power level changes. 

## Details

The power level was also observed to not return to the lowest value if the receiver disconnected at a higher power level and the arming aux channel was returned to a low value. 

While this does not appear to cause serious issues, it does behave in a way that is not consistent and what would be expected behavior for the disarmed state. 

## Your Environment

- TX hardware: Namimno Flash (OLED)
- RX hardware: Namimno Flash, HappyModel, FlyWoo.
- Handset model: Horus X10S with EdgeTX. 
- EdgeTX version (including nightly number): v2.6.0-rc3
 fac0eed
- ExpressLRS version (TX & RX MUST MATCH): 2.1.0
- Packet Rate: 160
- Telemetry Ratio: 1/16

pitts-mo, 2022-01-22:
Interesting that the LUA shows this.
I had the impression dynamic power transitioned to on after ARM or LINK and stayed on until the TX power was cycled and any power level changes I detected on new RX connect were just re-settling of the telemetry data after the power went to max at the time I unplugged the prior RX.

TomAlperin, 2022-01-22:
On the previous version, when you disarmed, the power is limited to 25mW and does not increase when the signal is lost. This is done to keep the module from running at full power in a situation where it is not needed.

pitts-mo, 2022-01-22:
I also observe something funny with this.

I just tried 2.1.0 and also find my telemetry reports cycling through power levels on an initial RX link of TX session even without prior use of ARM switch.

On 2.0.0 and the prior dynamic power builds utilized I only observed this on additional RX links per TX session. I had never noticed my power level to be more than 10mw at the initial RX link per TX power session.

CapnBry, 2022-01-22:
The power level MUST NOT return to low power on disconnect or disarm. If the arm switch is bumped in flight, setting low power would almost certainly cause a failsafe, and even the power boosting to max when you switch arm back on may not reconnect in time to save a crash.

* The dynamic power _at handset power on_ is minimum.
* Dynamic power will boost to max if Armed and disconnected from RX (or no RX ever connected)

Those are the only relevant rules. Often you will observe power is at a minimum between packs, but only because you were close when you disarmed and the power was already low.

Note: Dynamic power will boost to max if you reboot the handset in the presence of a receiver, because the receiver's LQ goes from 0 to 100 in this situation and the low LQ triggers a max power transition. This shouldn't happen plugging in a new RX because the LQ starts at 100 in that case.

I believe this is all working exactly as intended and how it should be unless you can give me a specific set of events that does not behave in this manner.

pitts-mo, 2022-01-22:
@CapnBry I am not observing what you describe.

On 2.1.0 I am experiencing my telemetry report power values cycling up and then back down through power modes on every initial RX link. This appears to match what @TomAlperin describes. I consistently observed my telemetry to maintain 10mw for initial RX link on 2.0.0 and all prior dynamic builds I have utilized.

Today's 2.1.0 testing was performed on my DIY TX and HappyModel EP2 receivers.

CapnBry, 2022-01-22:
I can't reproduce that behavior at all, unless armed when I reboot the RX. Every time I reboot the RX while disarmed, it starts at whatever power it was at when the RX was disconnected. The only way it should boost to full on connect is if you were armed when you unplugged the RX, you armed when the RX was off, or the initial LQ received from the RX was low.

CapnBry, 2022-01-22:
Aha! Working with Taranisaurus Wrecks on JB's discord, I see that on first handset power up, the power is minimum as expected. However, when the first telemetry packet comes in, the power goes go up only on this first connection. This is due to an initialization oversight where the "average LQ" is now low and can trigger either a blip to max power, or an increase in power. We'll fix this but it is pretty minor (only increases power on first connection for a short time) and does not fit the description I'm seeing from some here so any further information such as the exact order everyone is doing things in, and the initial reported LQ and power might help in that regard.

pitts-mo, 2022-01-22:
Glad you see it.  Can you restore the bug label?

Additional note:
In 1 of ~40ish instances where I powered up TX and then RX I observed telemetry mw to transition directly to 250mw and it appeared to become stuck... after maybe ~6 seconds I then armed and disarmed which seemed to allow the power to backed down to 10mw.

pitts-mo, 2022-01-22:
Also it appears that I can see RQly begin at 0 and quickly transition up to 100.

CapnBry, 2022-01-23:
> Also it appears that I can see RQly begin at 0 and quickly transition up to 100.

If your RQly starts at 0, then there's no problem with the dynamic power system. An LQ of 0 at any time is going to force the power to max. Usually RXes will start their LQ at 100 on first boot, but any subsequent connection will start at 0. Sounds more like your situation is that the RXes are not connecting properly and are actually double-connecting before their first telemetry packet, which makes the LQ start at 0?

The problem with dynamic power is that there's A LOT of conditions that cause things to do the things they were designed to do, and it seems most are looking at just the power number and saying "I don't think that's right", but it is doing what is right given the inputs.

pitts-mo, 2022-01-23:
I agree @CapnBry ,
I retested 2.0.0 builds are constantly starting initial connection with RQly at 100. Can you suggest a method to isolate why 2.1.0 consistently double connects my RX or otherwise starts RQly at 0? 
Thank you,
-pitts.mo

pitts-mo, 2022-01-23:
I have sampled OpenTX logs for two initial connections and one additional connection per session all while disarmed. From this perspective it does not appear that my RX is double connecting. However, I do see that both 2.0.1 and 2.1.0 do actually begin initial telemetry session with RQly=0. This brings me back to the belief that 2.0.1 and older builds were just not enforcing dynamic power on an initial RX when disarmed but 2.1.0 now does enforce dynamic power at the start of initial connection.

TomAlperin, 2022-01-26:
Not sure but I think the old system was working in a different way. Counting up bad packets instead so when it initially connected, the bad packet count would be at 0. From what I have read in the issues it looks like the method was changed so it would be more reliable in situations where signal dropped suddenly. I think it's looking at more than one value now to decide whether the power should go up.

CapnBry, 2022-01-26:
@pitts-mo if you turn on DEBUG_LOG you can add a line to DynamicPower_Update in tx_main to spit out the RSSI/LQ

pitts-mo, 2022-01-27:
I just put an analog cd-audio cable and a long pin header aside. This is a second time I have wanted to look at the serial console of my DIY TX. I plan to compare a  DEBUG_LOG builds of 2.1.0 and master when I get a chance. So glad you are part of the RC hobbies @CapnBry and thankful for teaching me to fish.
-p

CapnBry, 2022-01-27:
We just merged a fix for that first startup boost, and I included the debug line I was using for testing but it is commented out. Just copy that line to 2.1.0 and you should be able to compare the two. You might also want to turn the `DBGVLN` on the power decrease to `DBGLN` so it shows up (it is verbose-only by default because it can be printed quite a lot).

pitts-mo, 2022-01-28:
This issue appears to be resolved in 2.2.0. @TomAlperin please confirm when you get a chance.

I have attached some of my testing logs. I also perform some additional tests I did not log for abrupt TX and RX power off and arm state. All of these did appear to perform desirably... Just remember to re-arm if you need to trigger an initial power boost after recovering from a TX side problem. :-)

I found it especially interesting to watch the TX log when it determines it is safe to attempt a power decrement :-)

[2.1.0-dynamicpower-LQ-debug-disarmed-initial-connect.txt](https://github.com/ExpressLRS/ExpressLRS/files/7959181/2.1.0-dynamicpower-LQ-debug-disarmed-initial-connect.txt)
[2.2.0-dynamicpower-LQ-debug-disarmed-initial-and-reconnects.txt](https://github.com/ExpressLRS/ExpressLRS/files/7959182/2.2.0-dynamicpower-LQ-debug-disarmed-initial-and-reconnects.txt)

CapnBry, 2022-01-28:
> This issue appears to be resolved in 2.2.0. @TomAlperin please confirm when you get a chance.

Excellent! And checking your logs, it looks like it was exactly the issue I was seeing in 2.1.0, where the LQA takes time to come up to where it is supposed to be and you end up getting 3x power bumps on connect.

CapnBry, 2022-02-01:
Great, thanks for reporting back, glad I could find a fix!

closed 2022-02-01 by CapnBry