ExpressLRS/ExpressLRS #1310: BLE Joystick regression in 2.1.0 vs 2.0.1

rasmusbonnedal, 2022-01-22:
## Current Behavior

When I enable BLE Joystick in the LUA menu on v2.1.0 the message "Joystick running..." shows for a couple of seconds and I am then returned to the ELRS LUA menu. The joystick does not work. Downgrading to v2.0.1 makes the joystick work again.

## Steps to Reproduce

1. Enable BLE Joystick

## Possible Solution (Not obligatory)

I narrowed it down to https://github.com/ExpressLRS/ExpressLRS/commit/7e16ed4bc4e70531703edd5065ce3e07a4871d9c. Reverting this commit on v2.1.0 makes the joystick work again.

## Your Environment

- TX hardware: Happymodel ES24TX
- Handset model: Taranis Q X7S
- OpenTX version: 2.3.14
- ExpressLRS version (TX & RX MUST MATCH): 2.1.0

CapnBry, 2022-01-22:
Is this on the ES24TX Pro or non-Pro? I just tested the BLE Joystick yesterday and it worked as expected on the ES24TX Pro. Is the module rebooting when going to BLE mode?

rasmusbonnedal, 2022-01-23:
Non-pro.
How can I see if it reboots? On the display it looks like it just exits to the elrs lua script

schugabe, 2022-01-23:
I can reproduce this on ES24TX. Same error message as #1246 

`ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40091e50
file: ".pio/libdeps/HappyModel_ES24TX_2400_TX_via_UART/NimBLE-Arduino/src/NimBLEDevice.cpp" line 769
func: static void NimBLEDevice::init(const string&)
expression: esp_bt_controller_init(&bt_cfg)`

schugabe, 2022-01-23:
The issue seems to be that https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lib/BLE/devBLE.cpp#L62 is called multiple times with "Starting BLE Joystick!" twice in the logs. Init BT should only be called once ( https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv422esp_bt_controller_initP26esp_bt_controller_config_t ) and adding esp_bt_controller_get_status() to the output shows 0= idle and then 2=enabled.

```
UART STATS Bad:Good = 0:500
UART CRC failure
UART STATS Bad:Good = 1:489
Set Lua [BLE Joystick]=1
Set Lua [BLE Joystick]=6
UART STATS Bad:Good = 0:500
Set Lua [BLE Joystick]=4
hwTimer stop
hwTimer interval: 5000
Set Lua [BLE Joystick]=6
Adjusted max packet size 64-128
SetPower: 0
Starting BLE Joystick!
status 0
hwTimer stop
hwTimer interval: 5000
UART CRC failure
Adjusted max packet size 64-128
Starting BLE Joystick!
status 2
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x400927b0
file: ".pio/libdeps/Jumper_AION_NANO_2400_TX_via_UART/NimBLE-Arduino/src/NimBLEDevice.cpp" line 769
```

maybenikhil, 2022-01-24:
Fixed in #1322, thank you!

closed 2022-01-24 by maybenikhil