ExpressLRS/ExpressLRS #1314: Wifi broken after update to V2.1.0

TylerDD4, 2022-01-23:
## Current Behavior

 After updating my BETAFPV micro TX and 3 BETAFPV nano RXs all 3 receivers wifi is having the same issue where it broadcasts the signal, but as soon as you try and connect it resets and won't allow you to connect. I'm assuming this is not fixable and basically the wifi is useless now. I had this issue with a few other betafpv receivers and they never worked. 

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: BETAFPV Micro TX 2.4
- RX hardware: BETAFPV nano RX X3
- Handset model: TX16S
- OpenTX version (including nightly number) EDGETX 2.5
- ExpressLRS version (TX & RX MUST MATCH): V2.1.0
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2022-01-23:
Have you asked for help on discord in the #help-and-support channel?
I have updated my BetaFPV Micro TX 1W version to 2.1.0 and wifi continues to work as expected.
I also updated my Nano RX and wifi continues to work as expected there too.

TylerDD4, 2022-01-23:
> Have you asked for help on discord in the #help-and-support channel? I have updated my BetaFPV Micro TX 1W version to 2.1.0 and wifi continues to work as expected. I also updated my Nano RX and wifi continues to work as expected there too.

NO, I have not yet.. Just weird. I didn't have any issues going from 1.x to 2.0, or 2.0 to 2.01. Even going back and forth from betafpv fork to master ELRS branch. But now going to 2.1 all my receivers wifi isn't working anymore. But worked fine to upgrade to 2.1. 😕

deadbytefpv, 2022-01-23:
Weird to me = automatic bug?

TylerDD4, 2022-01-23:
> Weird to me = automatic bug?

It's not a coincidence.

deadbytefpv, 2022-01-24:
Were the receivers getting enough juice? WiFi mode usually takes more power than the normal operations.

closed 2022-04-05 by github-actions[bot]