ExpressLRS/ExpressLRS #1315: 2.1.0

lowefpv, 2022-01-23:
## Current Behavior
So flashed and blinking green light tried on both quads now same result blinking green    Happymodel ep2 just wasted more money buying 2 more 3rd set now 

Flashed and now blinking green slow no WiFi don’t connect tried pass through and bridge no result uart flash may work no time to try ordered 2 more rxs reverting to 2.0.1 not had issues there 
## Steps to Reproduce
All fine 2.0.1  tx took update and running rxs both flashing slow green no response bridged boot and pass through did not work no schematic for uart wiring will not attempt ordered new rx reverted tx  to 2.0.1 and will flash new rx to 2.0.1 when receive as I know works and had no issue 

1.

2.

3.

4.

## Possible Solution (Not obligatory)
None new rxs ordered rx bricked usb board or uart flash may fix but too much hassle 

## Details

3rd set this release should be pulled bricks Happymodel ep2 rx 
 fly and now have to wait few more days and more money wasted 

## Your Environment

- TX hardware:Happymodel es24 rxs
- RX hardware:ep2 rx
- Handset model: x lite 
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):2.1.0 
- Packet Rate:500
- Telemetry Ratio:1.64
- user_defines:

pkendall64, 2022-01-23:
Have you asked for help on discord? A bug report like this does not help anyone diagnose the problem that you are having.
Many devs and other users have upgrade to 2.1 on ES24 RXs without problems. Please if you are having problems, ask for help in the #help-and-support channel on discord.

AlessandroAU, 2022-01-23:
Sorry your update failed, there isn't enough information to determine why. The good news is that it's impossible to fully brick any ELRS receiver. You can recover ALL of your RXs with a little patience. https://www.expresslrs.org/2.0/quick-start/unbricking/

lowefpv, 2022-01-24:
I’ve tried everything on this list beside the usb board but I bought 2 more will tackle it later the board cost more than replacements

fiechr, 2022-01-27:
I had some issue, too. After connecting the RX to the FC, bridging boot (only for factory new RXs) and powering both on at the same time -- it didn't work.
I tried then to disconnect only VCC of the RX and reconnect it again (while keeping TX/RX and GND connected).

For some reason, it worked every time I did that with multiple RX and never when powering all up at the same time.

closed 2022-04-05 by github-actions[bot]