ExpressLRS/ExpressLRS #1316: NamimnoRC FLASH 2400 OLED TX module wifi does not appear when using the OLED or LUA when enabling wifi update

Witchdoctor2015, 2022-01-23:
I was attempting to change the bind phrase by flashing the module via the modules wi-fi but I could not get it to show up at all and eventually just used UART connection to flash.

## Your Environment

- TX hardware: NamimnoRC FLASH 2400 OLED TX
- RX hardware: NA
- Handset model: TBS MAMBO
- FreedomTX version TBS-1.3.0
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate: 250
- Telemetry Ratio: 1:128
- user_defines:

pkendall64, 2022-01-23:
Did you as on discord in the #help-and-support channel. Some devs have Namimno modules and these are tested prior to release.
You don't say if you were connecting to the hotspot or home network either, could it be that it connected to your home network and you expected it to create a hotspot or vice-versa?

piwi3910, 2022-01-23:
have the same module, no issues, 
i do have the same problem with the backpack.
When enabling it, it goes back to the main menu without activating it.
However the TX module just worked, as i just flashed 2.1 on it via wifi.

closed 2022-04-05 by github-actions[bot]