ExpressLRS/ExpressLRS #1317: ExpressLRS Home Wifi does not support wifi passwords over 32 characters

amargeson, 2022-01-23:
## Current Behavior

If your home wifi network password is over 32 characters, ELRS RX and TX modules will not connect and will go back to WAP mode.

## Steps to Reproduce

1.  Put RX or TX into wifi mode

2.  Attempt to connect to a wifi network with a wifi password over 32 characters

3.

4.

## Possible Solution (Not obligatory)

Assuming the ELRS code can only store a 32 character password and no more.  Increase the max characters allowed for wifi passwords.

## Details

Home wifi password best practices for security is to use a pseudorandom password with great length to prolong the duration it would take to manually crack the password.

## Your Environment

- TX hardware:  DIY 2.4GHz module
- RX hardware: Happymodel EP2
- ExpressLRS version (TX & RX MUST MATCH): 2.0, 2.01, 2.1
- Wifi network: WPA2 personal pre-shared key, AES

CapnBry, 2022-01-23:
Yup, this is because internally we only allocate 33 bytes for the wifi password storage. This is an oversight due to us just reusing the max SSID length of 32. We'll look at fixing this next time we change the internal config structure to add something.

Workaround for now: use AP mode

wucke13, 2022-02-21:
Probably this restructuring would also be a good time to fix https://github.com/ExpressLRS/ExpressLRS/issues/1307

pkendall64, 2022-06-10:
This will be in the 3.0 release.

closed 2022-07-02 by pkendall64