ExpressLRS/ExpressLRS #1318: R9MX shows linked but no telemetry signal

Manakoy, 2022-01-23:
## Current Behavior
TX : Frsky TX R9M LITE PRO
RX : R9MX
ELRS version : 2.1.0

The LED of R9MX shows normal link (red and yellow LEDs are always on), but the remote control does not show that it is linked. When the power of the remote control is turned off, the yellow light of the R9MX turns off and the red light flashes.
[image]
[image]

## Steps to Reproduce

1.Disable read protection through [STM32 ST-LINK Utility] software

2.Flashing Bootloader through [STM32 ST-LINK Utility] software
[image]

3.Flash the bin file produced by ExpressLRS Configurator into 0x08008000 of R9MX through the [STM32 ST-LINK Utility] software
[image]
[image]

## Your Environment

- TX hardware: Frsky TX R9M LITE PRO
- RX hardware: R9MX
- Handset model: Frsky x-lite
- OpenTX version (including nightly number) : 2.3.9
- ExpressLRS version (TX & RX MUST MATCH): 2.1
- Packet Rate: 200
- Telemetry Ratio: 1:64

deadbytefpv, 2022-01-23:
1. OpenTX is 2.3.9 when the earliest recommended is 2.3.12
2. you don't seem to have the latest ExpressLRS Configurator because "Enable Telemetry" is no longer needed for 2.x (as it was always enabled on the firmware)

Manakoy, 2022-01-23:
@deadbytefpv 
1. I'm not sure if it's because of the opentx version, I'll try to upgrade to the latest version and test again
2. ExpressLRS Configurator version is 1.3.3, which is indeed the latest

deadbytefpv, 2022-01-23:
[image]
If you have the latest ExpressLRS Configurator and is indeed flashing 2.1, why is it our options are very different for these devices?

Manakoy, 2022-01-23:
Probably due to I am building from a local file. 
[image]

deadbytefpv, 2022-01-23:
Use Manual Mode on Device Options
```
-DRegulatory_Domain_FCC_915
-DMY_BINDING_PHRASE="201361"
-DLOCK_ON_FIRST_CONNECTION
```

Also make sure TLM Ratio is not OFF via Lua Script.

Manakoy, 2022-01-23:
Tried, nothing changed. Very strange.
[image]
[image]

deadbytefpv, 2022-01-23:
See how the packet counter is saying you have 0/250 bad-good packets, when your max packet should only be 200 (for 900 system)?

CRSFShot isn't working...

Do you have any other R9 receivers to test?

Manakoy, 2022-01-23:
I use R9MM for testing and it links fine and receives telemetry data. But it still shows 0/250.
[image]
[image]
[image]

Manakoy, 2022-01-23:
I used another 2.4G TX to test the problem still exists, I think this is the reason for the low version of opentx.
[image]
[image]

deadbytefpv, 2022-01-23:
With all these "working" setups, you're still using the 2.3.9 OpenTX version.. hence CRSFShot isn't working.. 

Refer to this page again for setting up your radio.
https://www.expresslrs.org/2.0/quick-start/transmitters/tx-prep/

deadbytefpv, 2022-01-23:
This is my r9mx that I haven't used for a while. Flashed it via STLink to 2.1 just today for this.
[image]
[image]
[image]
as you can see from the `C` on the lua script, it was connected.. 

and getting telemetry data:
[image]

Manakoy, 2022-01-24:
I just updated to opentx2.3.14, and the phenomenon has changed.
The lua script can display 0/200 normally. After I try to bind, the remote control is connected to the R9MX, but it is quickly disconnected, and then it is connected again and then quickly disconnected, and it keeps looping.
In the telemetry data interface, i can see that several frames of data are received at intervals

deadbytefpv, 2022-01-24:
> I just updated to opentx2.3.14, and the phenomenon has changed. The lua script can display 0/200 normally. After I try to bind, the remote control is connected to the R9MX, but it is quickly [...]

Is this r9mx new-ish? 
When you say "disconnected and connected again", is it the "Telemetry Lost/Recovered" loops?
Does the Lua script top-right corner loses the "C" during these loops?

And at this point, we're shifting into a different issue altogether, I recon? no longer about Telemetry?

Manakoy, 2022-01-24:
It's not a new R9MX, but I'm pretty sure it worked before ELRS was flashed.
Yes, is the "Telemetry Lost/Recovered" loops.
"C" is never displayed in the upper right corner of the lua script, but "-"

bladan83, 2022-04-02:
> It's not a new R9MX, but I'm pretty sure it worked before ELRS was flashed. Yes, is the "Telemetry Lost/Recovered" loops. Yes, "c" sometimes appears

Same problem of you! have you solved?
I made the same procedure of you, but different radio.
My HW: R9MX, R9M 2017 (with resistor mod), Nirvana NV14 edgeTX 2.6
Express LRS version flashed 2.3 (i have tried older version with same results)
if telemetry is set 1/4 or 1/2 "Telemetry Lost/Recovered" loops, other rates no telemetry at all
baud rate set to also to 115k but nothing change. 
I think that the problem for some reason is on the R9MX maybe different revision, mine is rev0.2.

NOTE: I bought R9M 2nd hand, It was flashed with a firmaware called FLEX!

closed 2022-07-09 by github-actions[bot]